espressif/esp-idf #12837: esp-tls fails to allocate even though sufficient memory is available

thetek42, 2023-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

-

### Power Supply used.

USB

### What is the expected behavior?

When enough memory is available, a memory allocation should not fail.

### What is the actual behavior?

Setting up TLS with esp-tls causes a memory allocation to fail. According to the statistics (see the application logs below, printed from the `heap_caps_register_failed_alloc_callback` hook), the allocation should not fail, since only 4 bytes are being allocated, while 1803 bytes are available for the given capabilities, of which the largest block provides 128 bytes.

### Steps to reproduce.

Occurs after establishing a secure connection to a server using esp-tls and not quite enough memory is available.

### Debug Logs.

```plain
E (56929) esp_firmware::init: memory allocation failed
E (56934) esp_firmware::init:  - requested_size: 4 bytes
E (56939) esp_firmware::init:  - free_caps:      1803 bytes
E (56939) esp_firmware::init:  - free_internal:  38199 bytes
E (56944) esp_firmware::init:  - free_spiram:    4118348 bytes
E (56954) esp_firmware::init:  - max_free_block: 128 bytes

Backtrace:

heap_caps_alloc_failed_hook()
heap_caps_alloc_failed()
heap_caps_calloc()
esp_mbedtls_mem_calloc()
mbedtls_calloc()
mbedtls_mpi_grow()
ecp_mod_p521()
mbedtls_mpi_mul_mod()
ecp_add_mixed()
ecp_mul_restartable_internal$isra$0()
mbedtls_ecp_mul_restartable()
mbedtls_ecdh_compute_shared()
mbedtls_ecdh_calc_secret()
mbedtls_ssl_handshake_client_step()
mbedtls_ssl_handshake_step()
mbedtls_ssl_handshake()
esp_mbedtls_handshake()
esp_tls_low_level_conn()
esp_tls_conn_new_async()
... [application code]
```

### More Information.

- External RAM is enabled, but esp-tls is set to use internal memory only.
- Setting esp-tls to use external RAM fixes the issue, since enough memory is available there.

AdityaHPatwardhan, 2024-03-26:
Hello @thetek42,

Sorry for the delay.

In this case the requested size in this case is the size provided for the calloc  
```
E (56934) esp_firmware::init:  - requested_size: 4 bytes
```
But it does not provide the number of chunks to be allocated ( n provided in the calloc function).
so due to this maybe the memory request could be for more size than 4 bytes.

Is it possible for you to add debug prints in mbedtls calloc function?
That way we would know the exact amount required for the memory allocation which is failing.

AdityaHPatwardhan, 2024-04-22:
Hi @thetek42  The issue in the memory allocation information has been fixed with https://github.com/espressif/esp-idf/commit/497c840172fe2a340eb28cfa5331eadee4834641

I am closing this issue, please feel free to re-open if required

closed 2024-04-22 by AdityaHPatwardhan