espressif/esp-idf #12838: alloc(16749 bytes) failed occasionally occurs when you upgrade using http

LHYYSY, 2023-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32 C3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-13

### Power Supply used.

USB

### What is the expected behavior?

Read the firmware normally using http.

### What is the actual behavior?

E (1185926) Dynamic Impl: alloc(16749 bytes) failed
E (1185926) esp-tls-mbedtls: read error :-0x7F00:
E (1185926) transport_base: esp_tls_conn_read error, errno=Success
E (1185933) HTTP_CLIENT: transport_read: error - -1 | ESP_FAIL
I (1185941) native_esp_ota: current_dl_len:1657394,progress_tmp:90
E (1185966) Dynamic Impl: alloc(41913 bytes) failed
E (1185967) esp-tls-mbedtls: read error :-0x7F00:
E (1185967) transport_base: esp_tls_conn_read error, errno=Success
E (1185973) HTTP_CLIENT: transport_read: error - -1 | ESP_FAIL
E (1185980) native_esp_ota: Error: SSL data read error -1

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
Free heap space: 46400 byte
I (1185477) native_esp_ota: current_dl_len:1646592,progress_tmp:90
I (1185541) native_esp_ota: current_dl_len:1647616,progress_tmp:90
I (1185565) native_esp_ota: current_dl_len:1648640,progress_tmp:90
I (1185588) native_esp_ota: current_dl_len:1649664,progress_tmp:90
I (1185655) native_esp_ota: current_dl_len:1650688,progress_tmp:90
I (1185717) native_esp_ota: current_dl_len:1651712,progress_tmp:90
I (1185741) native_esp_ota: current_dl_len:1652736,progress_tmp:90
I (1185764) native_esp_ota: current_dl_len:1653760,progress_tmp:90
I (1185823) native_esp_ota: current_dl_len:1654784,progress_tmp:90
I (1185881) native_esp_ota: current_dl_len:1655808,progress_tmp:90
I (1185905) native_esp_ota: current_dl_len:1656832,progress_tmp:90
E (1185926) Dynamic Impl: alloc(16749 bytes) failed
E (1185926) esp-tls-mbedtls: read error :-0x7F00:
E (1185926) transport_base: esp_tls_conn_read error, errno=Success
E (1185933) HTTP_CLIENT: transport_read: error - -1 | ESP_FAIL
I (1185941) native_esp_ota: current_dl_len:1657394,progress_tmp:90
E (1185966) Dynamic Impl: alloc(41913 bytes) failed
E (1185967) esp-tls-mbedtls: read error :-0x7F00:
E (1185967) transport_base: esp_tls_conn_read error, errno=Success
E (1185973) HTTP_CLIENT: transport_read: error - -1 | ESP_FAIL
E (1185980) native_esp_ota: Error: SSL data read error -1
```

### More Information.

Occasionally triggering this error,

torkleyy, 2023-12-21:
Can you add a log printing the max free block size, please?

LHYYSY, 2023-12-22:
> 你能添加一个日志打印最大可用块大小吗？

Free heap space: 46400 byte

LHYYSY, 2023-12-22:
> Can you add a log printing the max free block size, please?

Ask how much heap memory is needed to avoid this problem

hmalpani, 2023-12-26:
Hello @LHYYSY 
Can you share the sample application and the sdkconfig file which can be used to reproduce this issue?

Thanks!

LHYYSY, 2023-12-26:
> Hello @LHYYSY Can you share the sample application and the sdkconfig file which can be used to reproduce this issue?
> 
> Thanks!
Please take a look. thank you very much.

#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.1.1 Project Configuration
#
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DEDICATED_GPIO_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_GDMA_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED=y
CONFIG_SOC_USB_SERIAL_JTAG_SUPPORTED=y
CONFIG_SOC_TEMP_SENSOR_SUPPORTED=y
CONFIG_SOC_XT_WDT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SUPPORTS_SECURE_DL_MODE=y
CONFIG_SOC_EFUSE_KEY_PURPOSE_FIELD=y
CONFIG_SOC_EFUSE_HAS_EFUSE_RST_BUG=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SYSTIMER_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_HMAC_SUPPORTED=y
CONFIG_SOC_DIG_SIGN_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_MEMPROT_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_AES_SUPPORT_DMA=y
CONFIG_SOC_AES_GDMA=y
CONFIG_SOC_AES_SUPPORT_AES_128=y
CONFIG_SOC_AES_SUPPORT_AES_256=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_ARBITER_SUPPORTED=y
CONFIG_SOC_ADC_DIG_IIR_FILTER_SUPPORTED=y
CONFIG_SOC_ADC_MONITOR_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=5
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=1
CONFIG_SOC_ADC_PATT_LEN_MAX=8
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=4
CONFIG_SOC_ADC_DIGI_DATA_BYTES_PER_CONV=4
CONFIG_SOC_ADC_DIGI_IIR_FILTER_NUM=2
CONFIG_SOC_ADC_DIGI_MONITOR_NUM=2
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_HIGH=83333
CONFIG_SOC_ADC_SAMPLE_FREQ_THRES_LOW=611
CONFIG_SOC_ADC_RTC_MIN_BITWIDTH=12
CONFIG_SOC_ADC_RTC_MAX_BITWIDTH=12
CONFIG_SOC_ADC_CALIBRATION_V1_SUPPORTED=y
CONFIG_SOC_ADC_SELF_HW_CALI_SUPPORTED=y
[... 2162 lines cut ...]
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32C3_TIME_SYSCALL_USE_RTC_SYSTIMER=y
# CONFIG_ESP32C3_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32C3_TIME_SYSCALL_USE_SYSTIMER is not set
# CONFIG_ESP32C3_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

LHYYSY, 2023-12-28:
> Hello @LHYYSY Can you share the sample application and the sdkconfig file which can be used to reproduce this issue?
> 
> Thanks!

Take a look. thank you very much
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/13782548/sdkconfig.txt)

hmalpani, 2023-12-28:
Hello @LHYYSY 
Can you also share the application code which can be used to reproduce this issue?

LHYYSY, 2024-01-05:
> Hello @LHYYSY Can you also share the application code which can be used to reproduce this issue?

I'm using routines to reproduce the problem, and I can't reproduce the problem,but I have both mqtt and http running at the same time, does it matter?

hmalpani, 2024-01-09:
Hello @LHYYSY 

Can you please log the largest free block in heap, you can use this API: https://github.com/espressif/esp-idf/blob/master/components/heap/heap_caps.c#L570

LHYYSY, 2024-01-10:
> 你好
> 
> 你能不能记录堆中最大的可用块，你可以使用这个API：https://github.com/espressif/esp-idf/blob/master/components/heap/heap_caps.c#L570
I used a task during the ota upgrade process to print "printf(" Free heap space: %ld byte free block %d\n", esp_get_free_heap_size(),heap_caps_get_largest_free_block(MALLOC_CAP_32BIT));” Log:
 Free heap space: 66852 byte free block  19456
I (92017) native_esp_ota: current_dl_len:1286144,progress_tmp:80
I (92080) native_esp_ota: current_dl_len:1287168,progress_tmp:80
I (92104) native_esp_ota: current_dl_len:1288192,progress_tmp:80
I (92127) native_esp_ota: current_dl_len:1289216,progress_tmp:80
I (92187) native_esp_ota: current_dl_len:1290240,progress_tmp:80
I (92248) native_esp_ota: current_dl_len:1291264,progress_tmp:80
I (92272) native_esp_ota: current_dl_len:1292288,progress_tmp:80
I (92295) native_esp_ota: current_dl_len:1293312,progress_tmp:80
I (92358) native_esp_ota: current_dl_len:1294336,progress_tmp:80
I (92422) native_esp_ota: current_dl_len:1295360,progress_tmp:80
I (92446) native_esp_ota: current_dl_len:1296384,progress_tmp:80
I (92469) native_esp_ota: current_dl_len:1297408,progress_tmp:80
E (92490) Dynamic Impl: alloc(16749 bytes) failed
E (92490) esp-tls-mbedtls: read error :-0x7F00:
E (92490) transport_base: esp_tls_conn_read error, errno=Success
E (92496) HTTP_CLIENT: transport_read: error - -1 | ESP_FAIL
I (92505) native_esp_ota: current_dl_len:1298131,progress_tmp:80
E (92529) Dynamic Impl: alloc(51491 bytes) failed
E (92530) esp-tls-mbedtls: read error :-0x7F00:
E (92530) transport_base: esp_tls_conn_read error, errno=Success
E (92536) HTTP_CLIENT: transport_read: error - -1 | ESP_FAIL
E (92542) native_esp_ota: Error: SSL data read error -1

hmalpani, 2024-01-16:
Hello @LHYYSY 

Can you update the code code to print the largest free block size such a way that we can get it just before the crash occurs. Also it will be helpful if you can share some application which can be used to reproduce this issue.

closed 2024-01-16 by LHYYSY

chegewara, 2024-10-02:
@rockys-spindance the logs are clear
`largest free block: 16384`
and
`alloc(16749 bytes) failed`