espressif/esp-idf #12839: ESP 32 Devkit v1 hangs

heineken63, 2023-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have flashed my esp board and evertyhing looks wel, but at startup it hangs.
Anyone knows what is wrong? Here the logs from my serial log:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7080
load:0x40078000,len:15584
load:0x40080400,len:4
0x40080400: _init at ??:?

ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c
I (31) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (31) boot: compile time Dec 17 2023 15:14:50
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.1
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=09d38h ( 40248) map
I (119) esp_image: segment 1: paddr=00019d60 vaddr=3ffb0000 size=02158h (  8536) load
I (122) esp_image: segment 2: paddr=0001bec0 vaddr=40080000 size=04158h ( 16728) load
I (132) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=14e60h ( 85600) map
I (164) esp_image: segment 4: paddr=00034e88 vaddr=40084158 size=0825ch ( 33372) load
I (184) boot: Loaded app from partition at offset 0x10000
I (184) boot: Disabling RNG early entropy source...
I (196) cpu_start: Multicore app
I (196) cpu_start: Pro cpu up.
I (196) cpu_start: Starting app cpu, entry point is 0x40081100
0x40081100: call_start_cpu1 at C:/Users/Greeve/esp/esp-idf/components/esp_system/port/cpu_start.c:157

I (0) cpu_start: App cpu up.
I (214) cpu_start: Pro cpu start user code
I (214) cpu_start: cpu freq: 240000000 Hz
I (214) cpu_start: Application information:
I (219) cpu_start: Project name:     esp32_rf_receiver
I (225) cpu_start: App version:      29bead8-dirty
I (230) cpu_start: Compile time:     Dec 17 2023 15:14:15
I (236) cpu_start: ELF file SHA256:  33c8ef3f229b24bf...
I (242) cpu_start: ESP-IDF:          v5.1.2
I (247) cpu_start: Min chip rev:     v1.0
I (252) cpu_start: Max chip rev:     v3.99 
I (257) cpu_start: Chip rev:         v3.1
I (262) heap_init: Initializing. RAM available for dynamic allocation:
I (269) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (275) heap_init: At 3FFB2AF8 len 0002D508 (181 KiB): DRAM
I (281) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (287) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (294) heap_init: At 4008C3B4 len 00013C4C (79 KiB): IRAM
I (301) spi_flash: detected chip: generic
I (305) spi_flash: flash io: dio
W (308) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary 
image header.
I (322) app_start: Starting scheduler on CPU0
I (327) app_start: Starting scheduler on CPU1
I (327) main_task: Started on CPU0
I (337) main_task: Ca�

SoucheSouche, 2023-12-20:
Hi @heineken63, it seems that your program gets stuck just before calling `app_main()`, or at the very beginning of it. Are you at liberty to share the content of your main function or at least what it does?

heineken63, 2023-12-20:
Hi @SoucheSouche 
Here is the app_main is this enough?

void app_main()
{
  if (s_esp_RF433_queue == NULL) {
      s_esp_RF433_queue = xQueueCreate(1, sizeof(uint8_t));
      if (s_esp_RF433_queue != NULL) {
        // Configure the data input
        gpio_config_t data_pin_config = {
          .intr_type = GPIO_INTR_ANYEDGE,
          .mode = GPIO_MODE_INPUT,
          .pin_bit_mask = GPIO_NUM_22, // GPIO_NUM_22 (SEL) DATA PIN!
          .pull_up_en = GPIO_PULLUP_DISABLE,
          .pull_down_en = GPIO_PULLDOWN_DISABLE
        };

        gpio_config(&data_pin_config);

        // Attach the interrupt handler
        gpio_install_isr_service(ESP_INTR_FLAG_EDGE);
        gpio_isr_handler_add(GPIO_NUM_22, data_interrupt_handler, NULL);  // GPIO_NUM_22 DATA PIN!
        xTaskCreate(&receiver_rf433, "receiver_rf433", 2048, NULL, 3, NULL);
    }
  }
}

SoucheSouche, 2023-12-21:
@heineken63, yes thank you very much. I could reproduce your issue. I will investigate and come back to you when I know more.

SoucheSouche, 2023-12-21:
@heineken63, you are passing the pin number `GPIO_NUM_22` in the `gpio_config_t` field `.pin_bit_mask` instead of passing the bit mask.
```
.pin_bit_mask = BIT64(GPIO_NUM_22), // 1 << GPIO_NUM_22
```

It is very easy to overlook this mistake. I couldn't figure out why using the pin 22 as input was not possible without realizing that the configuration provided was erroneous :)
Well, anyway, this should solve your issue.

heineken63, 2023-12-21:
@SoucheSouche 
Thanks the app is starting now.
But the following is happening as not expected, I expected it to keep on listening but this is the only thing showing: 
I (340) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:3 
I (350) main_task: Returned from app_main()

SoucheSouche, 2023-12-21:
I am not sure what you mean. I tested you code and implemented `data_interrupt_handler`, and every time I change the state on the pin 22, the function `data_interrupt_handler` is called.

heineken63, 2023-12-21:
That is strange, I have connected my rf_receiver to pin 22 but nothing happens. 
You say you implemented the data_interrupt_handler, with something of your own?

SoucheSouche, 2024-01-08:
Hi @heineken63,
In your code,
`gpio_isr_handler_add(GPIO_NUM_22, data_interrupt_handler, NULL);`, `data_interrupt_handler` should be a function defined in your code.

```
void data_interrupt_handler(void *arg)
{
    // add the code you want executed every time the interrupt triggers on the gpio 22
}
```

SoucheSouche, 2024-01-15:
@heineken63,
I will close the issue now since the initial problem was resolved. I also provided the necessary information for you to successfully implement the interrupt handler function.

closed 2024-01-15 by SoucheSouche