espressif/esp-idf #12842: MQTT SSL EXAMPLE

edg2411, 2023-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0

### Espressif SoC revision.

esp32-wroom-32e

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

custom board | esp32devkitc

### Power Supply used.

External 5V

### What is the expected behavior?

The expected behavior is the one expected on the ssl example

### What is the actual behavior?

It returns error 
E (15785) esp-tls: [sock=54] select() timeout
E (15785) esp-tls: Failed to open new connection
E (15785) transport_base: Failed to open a new connection
E (15785) mqtt_client: Error transport connect

### Steps to reproduce.

1. Step
2. Step
3. Step
...
use ssl example over examples/protocols/mqtt/ssl

### More Information.

Over tcp works well, cert is accepted tried different brokers and accounts
I did see some issues with the same description but solution didnt work for me

ESP-YJM, 2023-12-22:
@edg2411 It seems the TCP SYN packet no response and TCP connection failed. Could you please check the port whether is open in server side?

edg2411, 2023-12-22:
it is at least with my broker i can connect and talk to with mqtt ssl

ESP-YJM, 2023-12-25:
@edg2411 Could you share the broker url and port? And also share the mqtt/ssl example that mqtt client configuration.

edg2411, 2023-12-26:
i made the example work.. now im trying to get it work with my broker.. im using an account from cloudmqtt i think im not allow to connect through ssl with that account..

8888alex, 2023-12-26:
我使用範例成功的更換成我的mqtt主機及自簽憑證。
但移植到我的專案時卻得到了下面的錯誤。
我不知道我漏了甚麼。

```
I (6041) wifi:AP's beacon interval = 102400 us, DTIM period = 1
W (7045) mesh_mqtt: mqtt_app_start()
_mqttHost ... mqtts://mqtt.aiotsys.com
I (7056) mesh_mqtt: [APP] Free memory: 2029212 bytes
W (7058) mqtt_client: Publish: Losing qos0 data when client not connected
W (7061) Mesh-Lite-Message: send_message_to_parent
I (7072) esp_netif_handlers: sta ip: 192.168.88.50, mask: 255.255.255.0, gw: 192.168.88.1
I (7076) bridge_wifi: Connected with IP Address:192.168.88.50
I (7082) vendor_ie: RTC store: ssid:MikroTik-8A6FB4; bssid:00:00:00:00:00:00 crc:2874121999
I (7101) mesh_mqtt: Other event id:7
E (7133) esp-tls-mbedtls: mbedtls_x509_crt_parse returned -0x1180
E (7133) esp-tls-mbedtls: Failed to set client configurations, returned [0x8015] (ESP_ERR_MBEDTLS_X509_CRT_PARSE_FAILED)
E (7140) esp-tls: create_ssl_handle failed
E (7145) esp-tls: Failed to open new connection
E (7150) transport_base: Failed to open a new connection
E (7158) mqtt_client: Error transport connect
I (7162) mesh_mqtt: MQTT_EVENT_ERROR
```

### 環境
esp32-s2-devkitm-1-v1
IDF v5.0.2

idf_component.yml
dependencies:
idf: ">=5.0"
mesh_lite:
version: ""
espressif/esp_modem:
version: 0.
espressif/esp_modem_usb_dte:
version: 1.*
public: true
rules:

if: "target in [esp32s2, esp32s3]"
usb_device:
path: components/usb/usb_device
git: https://github.com/espressif/esp-iot-bridge.git
rules:
if: "target in [esp32s2, esp32s3]"
espressif/led_strip: "^2.0.0"
lib
version
libesp_mesh_lite_esp32.a :473e10d
libesp_mesh_lite_esp32c2.a :473e10d
libesp_mesh_lite_esp32c3.a :473e10d
libesp_mesh_lite_esp32c6.a :473e10d
libesp_mesh_lite_esp32s2.a :473e10d
libesp_mesh_lite_esp32s3.a :473e10d

### sdkconfig
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/13773244/sdkconfig.zip)

### 我的 mqtt_main
[mqtt_main.zip](https://github.com/espressif/esp-idf/files/13773247/mqtt_main.zip)

ESP-YJM, 2023-12-27:
> 我使用範例成功的更換成我的mqtt主機及自簽憑證。 但移植到我的專案時卻得到了下面的錯誤。 我已經盡可能地去比對與範例的差異，但我找不到漏了甚麼。 或許最大的差異是我的專案是透過mseh-lite建立連線。 因為我將範例的app_main.c複製到我的專案中也得到相同的錯誤。
> 
> ```
> I (6041) wifi:AP's beacon interval = 102400 us, DTIM period = 1
> W (7045) mesh_mqtt: mqtt_app_start()
> _mqttHost ... mqtts://mqtt.aiotsys.com
> I (7056) mesh_mqtt: [APP] Free memory: 2029212 bytes
> W (7058) mqtt_client: Publish: Losing qos0 data when client not connected
> W (7061) Mesh-Lite-Message: send_message_to_parent
> I (7072) esp_netif_handlers: sta ip: 192.168.88.50, mask: 255.255.255.0, gw: 192.168.88.1
> I (7076) bridge_wifi: Connected with IP Address:192.168.88.50
> I (7082) vendor_ie: RTC store: ssid:MikroTik-8A6FB4; bssid:00:00:00:00:00:00 crc:2874121999
> I (7101) mesh_mqtt: Other event id:7
> E (7133) esp-tls-mbedtls: mbedtls_x509_crt_parse returned -0x1180
> E (7133) esp-tls-mbedtls: Failed to set client configurations, returned [0x8015] (ESP_ERR_MBEDTLS_X509_CRT_PARSE_FAILED)
> E (7140) esp-tls: create_ssl_handle failed
> E (7145) esp-tls: Failed to open new connection
> E (7150) transport_base: Failed to open a new connection
> E (7158) mqtt_client: Error transport connect
> I (7162) mesh_mqtt: MQTT_EVENT_ERROR
> ```
> 
> ### 環境
> 
> esp32-s2-devkitm-1-v1 IDF v5.0.2
> 
> idf_component.yml dependencies: idf: ">=5.0" mesh_lite: version: "" espressif/esp_modem: version: 0. espressif/esp_modem_usb_dte: version: 1.* public: true rules:
> 
> if: "target in [esp32s2, esp32s3]" usb_device: path: components/usb/usb_device git: https://github.com/espressif/esp-iot-bridge.git rules: if: "target in [esp32s2, esp32s3]" espressif/led_strip: "^2.0.0" lib version libesp_mesh_lite_esp32.a :473e10d libesp_mesh_lite_esp32c2.a :473e10d libesp_mesh_lite_esp32c3.a :473e10d libesp_mesh_lite_esp32c6.a :473e10d libesp_mesh_lite_esp32s2.a :473e10d libesp_mesh_lite_esp32s3.a :473e10d
> ### sdkconfig
> 
> [sdkconfig.zip](https://github.com/espressif/esp-idf/files/13773244/sdkconfig.zip)
> ### 我的 mqtt_main
> 
> [mqtt_main.zip](https://github.com/espressif/esp-idf/files/13773247/mqtt_main.zip)

0x1180 means `MBEDTLS_ERR_PEM_ALLOC_FAILED`, it seems the internal memory is not enough to malloc buffer for pem. Could you please print the internal memory szie and you can add some debug.

ESP-YJM, 2023-12-27:
> i made the example work.. now im trying to get it work with my broker.. im using an account from cloudmqtt i think im not allow to connect through ssl with that account..

Do the account is for mqtt? Such as mqtt username and password?

edg2411, 2023-12-27:
yes and i add it to the code.. i needed to do that even with tcp and it worked.. the cert is the same as the one in the example i dont understand much about that topic

ESP-YJM, 2024-01-08:
Sorry that i not get your point. Colud you tell me about the problems you are currently encountering and what topic you want to know?

ESP-YJM, 2024-03-06:
@edg2411 Any update for this issue? I am closing this issue. If you have, feel free to open again. Thanks.

closed 2024-03-06 by ESP-YJM