espressif/esp-idf #12843: ADC_ATTEN_DB_12 undeclared

nopnop2002, 2023-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

esp32

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32

### Power Supply used.

USB

### What is the expected behavior?

Build completes successfully

### What is the actual behavior?

[This](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/adc_oneshot.html) document states:
```
enumerator ADC_ATTEN_DB_12
The input voltage of ADC will be attenuated extending the range of measurement by about 12 dB.

enumerator ADC_ATTEN_DB_11
This is deprecated, it behaves the same as ADC_ATTEN_DB_12
```

However, ADC_ATTEN_DB_12 results in an error.
Not implemented according to the documentation.

### Steps to reproduce.

1. esp-idf set-target esp32
2. idf.py build

### Debug Logs.

```plain
/home/nop/rtos/adc_attren_db/main/main.c: In function 'app_main':
/home/nop/rtos/adc_attren_db/main/main.c:23:40: error: 'ADC_ATTEN_DB_12' undeclared (first use in this function); did you mean 'ADC_ATTEN_DB_11'?
   23 |         printf("ADC_ATTEN_DB_12=%d\n", ADC_ATTEN_DB_12);
      |                                        ^~~~~~~~~~~~~~~
      |                                        ADC_ATTEN_DB_11
```
```

### More Information.

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_adc/adc_oneshot.h"
#include "esp_adc/adc_cali.h"
#include "esp_adc/adc_cali_scheme.h"

void app_main(void)
{
    printf("ADC_ATTEN_DB_0=%d\n", ADC_ATTEN_DB_0);
    printf("ADC_ATTEN_DB_2_5=%d\n", ADC_ATTEN_DB_2_5);
    printf("ADC_ATTEN_DB_6=%d\n", ADC_ATTEN_DB_6);
    printf("ADC_ATTEN_DB_11=%d\n", ADC_ATTEN_DB_11);
    printf("ADC_ATTEN_DB_12=%d\n", ADC_ATTEN_DB_12);
}
```

suda-morris, 2023-12-20:
because 6de9757a4b036dde8db943c8686acd1eeca1fce6 is not in the tag v5.1.2. While you're reading the programming guide for a master branch.

AxelLin, 2024-03-05:
This should be closed, @suda-morris

AxelLin, 2024-05-19:
This should be closed, @Alvin1Zhang

closed 2024-05-19 by nopnop2002