espressif/esp-idf #12844: BLE Provisioning getting stuck at WIFI_PROV_CRED_RECV - V5.1

monkeytronics, 2023-12-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.1

### Espressif SoC revision.

ESP32-WROOM32E - 8MB

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

none. Custom hardware

### Power Supply used.

USB

### What is the expected behavior?

Expect provisioning to complete and deinti gracefully once correct AP & password are provided.

### What is the actual behavior?

 Instead, no further events are seen after WIFI_PROV_CRED_RECV & WIFI_EVENT_STA_CONNECTED. The credentials are clearly correct as the Wifi will successfully conenct on subsequent power up. So the provisioning prcess stored the credentials correctly. It does not stop gracefully. Which will be an absolutel killer for user experience in the app. 

### Steps to reproduce.

nvs_init_default(); // usual 
monkey_button_intr_init(); // create interrupt handler for GPIO
monkey_button_pairing_mode(PROVISION_BUTTON_GPIO); // attach a button to ISR which is used to cause a restart and start up in provisioning mode. 

monkey_wifi_init_common(); // esp_netif_init & esp_event_loop_create_default & register events & esp_wifi_init

wifi_prov_mgr_config_t cfg_provisioning = { /* BLE Provisioning */
        .scheme                     = wifi_prov_scheme_ble,
        .scheme_event_handler       = WIFI_PROV_SCHEME_BLE_EVENT_HANDLER_FREE_BTDM
    };
wifi_prov_mgr_init(cfg_provisioning)
wifi_prov_scheme_ble_set_service_uuid(custom_service_uuid)
wifi_prov_mgr_start_provisioning(security, (const void *)sec_params, service_name, service_key)

/* 3. Wait for WiFi Scan -> TIMEOUT_SCAN_DONE */
    xEventGroupClearBits(wifi_event_group, WIFI_SCAN_DONE_BIT);
    EventBits_t bits = xEventGroupWaitBits(
        wifi_event_group,
        WIFI_SCAN_DONE_BIT,
        pdFALSE, // clear on exit = false
        pdFALSE, // wait for all = false
        (TIMEOUT_SCAN_DONE * 1000 / portTICK_PERIOD_MS)
    );

/* 4. Wait for Provisioning complete -> TIMEOUT_PROV_DONE */
    xEventGroupClearBits(wifi_event_group, PROV_GOOD_BIT | PROV_FAIL_BIT);

    /* Wait until success or Failure */
    bits = xEventGroupWaitBits(
        wifi_event_group,
        PROV_GOOD_BIT | PROV_FAIL_BIT,
        pdFALSE, // clear on exit = false
        pdFALSE, // wait for all = false
        (TIMEOUT_PROV_DONE * 1000 / portTICK_PERIOD_MS)
    );

 /* Success */
    if (bits & PROV_GOOD_BIT)
    {
        ESP_LOGE(TAG, "Provisioning Succeeded");
        ret = ESP_OK;
    }
    /* Failure */
    else if (bits & PROV_FAIL_BIT)
    {
        ESP_LOGE(TAG, "Provisioning Failed");
        // https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/nvs_flash.html
        const char *partition_label = "nvs";
        nvs_flash_erase_partition(partition_label); // Ditch failing credentials
        ret = ESP_ERR_NOT_FINISHED;
    } 
    /* Timeout */
    else 
    {
        ESP_LOGE(TAG, "Provisioning Timed out");
        ret = ESP_ERR_TIMEOUT;
    }

Obviously, all the while, the event handler is running alongside all of this. 

### Debug Logs.

```plain
D (36036) NimBLE:

D (36036) wifi:Send sta connected event
D (36036) protocomm_nimble: Write attempt for uuid = 1234ff52-90ab-cdef-fedc-ba0987654321, attr_handle = 12, data_len = 2
D (36036) wifi:D (36056) protocomm_nimble: UUID did not match... 52
connect status 1 -> 5D (36056) protocomm_nimble: UUID did not match... 52

D (36066) protocomm_nimble: UUID did not match... 52
D (36066) wifi:D (36066) protocomm_nimble: UUID (0x52) matched with proto-name = prov-config
obss scan is disabled
D (36076) WiFiProvConfig: In wifi_prov_config_command_dispatcher Cmd=0
D (36076) wifi:D (36086) WiFiProvConfig: Enter cmd_get_status_handler
start obss scan: obss scan is stopped
D (36096) wifi_prov_handlers: Got state : connecting
I (36096) wifi:AP's beacon interval = 102400 us, DTIM period = 1
D (36106) NimBLE: host tx hci data; handle=0 length=5

D (36046) event: running post WIFI_EVENT:4 with handler 0x400d7be4 and context 0x3ffcd890 on loop 0x3ffcc91c
0x400d7be4: monkey_wifi_event_handler at D:/MonkeySource/4-FW/2-esp32-dev-v5.1/airsmart-fw/common/monkey_wifi/monkey_wifi.c:602

D (36116) NimBLE: ble_hs_hci_acl_tx():
I (36126) monkey_wifi.c: * WIFI_EVENT : SYSTEM_EVENT_STA_CONNECTED
D (36126) NimBLE: 0x00
D (36136) event: running post WIFI_EVENT:4 with handler 0x400d8a0c and context 0x3ffdd21c on loop 0x3ffcc91c
0x400d8a0c: wifi_prov_mgr_event_handler_internal at C:/Espressif/v5.1/components/wifi_provisioning/src/manager.c:890

D (36136) NimBLE: 0x00
D (36146) NimBLE: 0x05
D (36146) NimBLE: 0x00
D (36156) NimBLE: 0x01
D (36156) NimBLE: 0x00
D (36156) NimBLE: 0x04
D (36156) NimBLE: 0x00
D (36166) NimBLE: 0x13
D (36166) NimBLE:

D (36236) NimBLE: ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=7 data=
D (36236) NimBLE: 0x03
D (36236) NimBLE: 0x00
D (36236) NimBLE: 0x04
D (36236) NimBLE: 0x00
D (36236) NimBLE: 0x0a
D (36246) NimBLE: 0x0c
D (36246) NimBLE: 0x00
D (36246) NimBLE:

D (36246) protocomm_nimble: Read attempted for characteristic UUID = 1234ff52-90ab-cdef-fedc-ba0987654321, attr_handle = 12
D (36266) NimBLE: host tx hci data; handle=0 length=11

D (36266) NimBLE: ble_hs_hci_acl_tx():
D (36276) NimBLE: 0x00
D (36276) NimBLE: 0x00
D (36276) NimBLE: 0x0b
D (36276) NimBLE: 0x00
D (36286) NimBLE: 0x07
D (36286) NimBLE: 0x00
D (36286) NimBLE: 0x04
D (36286) NimBLE: 0x00
D (36296) NimBLE: 0x0b
D (36296) NimBLE: 0x75
D (36296) NimBLE: 0x43
D (36296) NimBLE: 0x89
D (36306) NimBLE: 0xdd
D (36306) NimBLE: 0x5c
D (36306) NimBLE: 0xd1
D (36306) NimBLE:

D (37166) NimBLE: ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0035 len=5

D (37166) NimBLE: 0x35
D (37166) NimBLE: 0x0c
D (37166) NimBLE: 0x05
D (37176) NimBLE: 0x01
D (37176) NimBLE: 0x00
D (37176) NimBLE: 0x00
D (37176) NimBLE: 0x02
D (37186) NimBLE: 0x00
D (37186) NimBLE:

D (38376) NimBLE: ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=
D (38376) NimBLE: 0x05
D (38376) NimBLE: 0x00
D (38376) NimBLE: 0x04
D (38386) NimBLE: 0x00
D (38386) NimBLE: 0x12
D (38386) NimBLE: 0x0c
D (38386) NimBLE: 0x00
D (38396) NimBLE: 0x50
D (38396) NimBLE: 0x51
D (38396) NimBLE:

D (38396) protocomm_nimble: Write attempt for uuid = 1234ff52-90ab-cdef-fedc-ba0987654321, attr_handle = 12, data_len = 2
D (38416) protocomm_nimble: UUID did not match... 52
D (38416) protocomm_nimble: UUID did not match... 52
D (38426) protocomm_nimble: UUID did not match... 52
D (38426) protocomm_nimble: UUID (0x52) matched with proto-name = prov-config
D (38436) WiFiProvConfig: In wifi_prov_config_command_dispatcher Cmd=0
D (38436) WiFiProvConfig: Enter cmd_get_status_handler
D (38446) wifi_prov_handlers: Got state : connecting
D (38456) NimBLE: host tx hci data; handle=0 length=5
```

### More Information.

Looking at the logs, it appears to be unhappy with the UUID of one of the Bluetooth services... Have these changed? Because it worked fine with ESP-IDF 4.3.

monkeytronics, 2023-12-20:
ACtually on the logs, I think that's just the way it goes through to find the right one. It starts at the bottom and checks each UUID until it finds a match. The issue is that it appears to be stuck in connecting state. The event wait is waiting for a prov success or prov fail event. Whnen it doesn't arrive after 120 seconds, it times out and runs prov_deinit() directly. Although the code properly fails at this point, the first question is why does it get stick in this state? As far as I can tell, it should have everything it needs to resolve everything and gracefully complete the provisioning process. 

D (36036) NimBLE:

D (36036) wifi:Send sta connected event
D (36036) protocomm_nimble: Write attempt for uuid = 1234ff52-90ab-cdef-fedc-ba0987654321, attr_handle = 12, data_len = 2
D (36036) wifi:D (36056) protocomm_nimble: UUID did not match... 52
connect status 1 -> 5D (36056) protocomm_nimble: UUID did not match... 52

D (36066) protocomm_nimble: UUID did not match... 52
D (36066) wifi:D (36066) protocomm_nimble: UUID (0x52) matched with proto-name = prov-config
obss scan is disabled
D (36076) WiFiProvConfig: In wifi_prov_config_command_dispatcher Cmd=0
D (36076) wifi:D (36086) WiFiProvConfig: Enter cmd_get_status_handler
start obss scan: obss scan is stopped
D (36096) wifi_prov_handlers: Got state : connecting
I (36096) wifi:AP's beacon interval = 102400 us, DTIM period = 1
D (36106) NimBLE: host tx hci data; handle=0 length=5

D (36046) event: running post WIFI_EVENT:4 with handler 0x400d7be4 and context 0x3ffcd890 on loop 0x3ffcc91c
0x400d7be4: monkey_wifi_event_handler at D:/MonkeySource/4-FW/2-esp32-dev-v5.1/airsmart-fw/common/monkey_wifi/monkey_wifi.c:602

D (36116) NimBLE: ble_hs_hci_acl_tx():
I (36126) monkey_wifi.c: * WIFI_EVENT : SYSTEM_EVENT_STA_CONNECTED
D (36126) NimBLE: 0x00
D (36136) event: running post WIFI_EVENT:4 with handler 0x400d8a0c and context 0x3ffdd21c on loop 0x3ffcc91c
0x400d8a0c: wifi_prov_mgr_event_handler_internal at C:/Espressif/v5.1/components/wifi_provisioning/src/manager.c:890

D (36136) NimBLE: 0x00
D (36146) NimBLE: 0x05
D (36146) NimBLE: 0x00
D (36156) NimBLE: 0x01
D (36156) NimBLE: 0x00
D (36156) NimBLE: 0x04
D (36156) NimBLE: 0x00
D (36166) NimBLE: 0x13
D (36166) NimBLE:

D (36236) NimBLE: ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=7 data=
D (36236) NimBLE: 0x03
D (36236) NimBLE: 0x00
D (36236) NimBLE: 0x04
D (36236) NimBLE: 0x00
D (36236) NimBLE: 0x0a
D (36246) NimBLE: 0x0c
D (36246) NimBLE: 0x00
D (36246) NimBLE:

D (36246) protocomm_nimble: Read attempted for characteristic UUID = 1234ff52-90ab-cdef-fedc-ba0987654321, attr_handle = 12
D (36266) NimBLE: host tx hci data; handle=0 length=11

D (36266) NimBLE: ble_hs_hci_acl_tx():
D (36276) NimBLE: 0x00
D (36276) NimBLE: 0x00
D (36276) NimBLE: 0x0b
D (36276) NimBLE: 0x00
D (36286) NimBLE: 0x07
D (36286) NimBLE: 0x00
D (36286) NimBLE: 0x04
D (36286) NimBLE: 0x00
D (36296) NimBLE: 0x0b
D (36296) NimBLE: 0x75
D (36296) NimBLE: 0x43
D (36296) NimBLE: 0x89
D (36306) NimBLE: 0xdd
D (36306) NimBLE: 0x5c
D (36306) NimBLE: 0xd1
D (36306) NimBLE:

D (37166) NimBLE: ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0035 len=5
[... 451 lines cut ...]
D (132436) wifi_prov_mgr: execute_event_cb : 5
D (132436) event: running post WIFI_PROV_EVENT:5 with handler 0x400d7be4 and context 0x3ffcd8f8 on loop 0x3ffcc91c
0x400d7be4: monkey_wifi_event_handler at D:/MonkeySource/4-FW/2-esp32-dev-v5.1/airsmart-fw/common/monkey_wifi/monkey_wifi.c:602

W (132446) monkey_wifi.c: * PROV_EVENT : WIFI_PROV_END
I (132456) monkey_wifi.c: Deinitializing provisioning manager...
D (132466) wifi_prov_mgr: Provisioning not running
D (132466) wifi_prov_mgr: Manager already de-initialized
D (132476) wifi_prov_mgr: execute_event_cb : 6
D (132476) BTDM_INIT: Release DRAM [0x3ffae6e0] - [0x3ffaff10]
D (132486) BTDM_INIT: Release DRAM [0x3ffb0000] - [0x3ffb2730]
D (132486) BTDM_INIT: Release DRAM [0x3ffb8000] - [0x3ffb9a20]
D (132496) BTDM_INIT: Release DRAM [0x3ffbdb28] - [0x3ffbdb5c]
D (132496) BTDM_INIT: Release BTDM BSS [0x3ffc8bcc] - [0x3ffc8e08]
D (132506) BTDM_INIT: Release BTDM Data [0x3ffc1f90] - [0x3ffc1fb4]
D (132516) BTDM_INIT: Release BT BSS [0x3ffc72d0] - [0x3ffc8bcc]
D (132516) BTDM_INIT: Release BT Data [0x3ffc0dac] - [0x3ffc1f90]
I (132526) wifi_prov_scheme_ble: BTDM memory released
D (132536) event: running post WIFI_PROV_EVENT:6 with handler 0x400d7be4 and context 0x3ffcd8f8 on loop 0x3ffcc91c
0x400d7be4: monkey_wifi_event_handler at D:/MonkeySource/4-FW/2-esp32-dev-v5.1/airsmart-fw/common/monkey_wifi/monkey_wifi.c:602

W (132546) monkey_wifi.c: * PROV_EVENT : WIFI_PROV_DEINIT
I (132546) monkey_wifi.c: Provisioning manager deinitialized.

assert failed: vQueueDelete queue.c:2300 (pxQueue)

monkeytronics, 2023-12-20:
Just to rule my app out of the equation, I've repeated the process with the ESP android app. It sticks at exactly the same place. Looks like it's getting the STA_COPNNECTED but never gets the GOT_IP. Both event groups are registered in the event handler, and I can see the GOT_IP during a normal power on after the provisioning is done (or sort of done). 

[image]

idea--list, 2023-12-21:
@monkeytronics 
I also had problems with provisioning on earlier releases. See #11130
In my case one solution was to wait at least 20 minutes before trying to provision the same device again.
Not sure if this will help in your case, but you might just try that.

rahult-github, 2023-12-22:
Hi @monkeytronics ,

Tried the default example for v5.1 with esp32 , and see that no issue is observed .

--- 
 
 I (62695) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (62695) wifi:state: init -> auth (b0)
I (62735) wifi:state: auth -> assoc (0)
I (62785) wifi:state: assoc -> run (10)
I (62785) wifi:connected with AndroidRainmakerOpenNetwork0123, aid = 3, channel 11, BW20, bssid = 40:31:3c:e6:00:7c
I (62785) wifi:security: Open Auth, phy: bgn, rssi: -58
I (62805) wifi:pm start, type: 1

I (62805) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (65155) wifi:<ba-add>idx:0 (ifx:0, 40:31:3c:e6:00:7c), tid:0, ssn:0, winSize:64
I (65815) app: Connected with IP Address:192.168.31.145
I (65815) esp_netif_handlers: sta ip: 192.168.31.145, mask: 255.255.255.0, gw: 192.168.31.1
I (65815) wifi_prov_mgr: STA Got IP
I (65815) app: Provisioning successful
I (65825) app: Hello World!
I (66825) app: Hello World!
I (66845) NimBLE: GAP procedure initiated: stop advertising.

I (66845) NimBLE: GAP procedure initiated: stop advertising.

I (66855) NimBLE: GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19

E (66915) protocomm_nimble: Error setting advertisement data; rc = 30
I (66915) wifi_prov_mgr: Provisioning stopped
I (66915) app: BLE transport: Disconnected!
I (66915) app: BLE transport: Disconnected!
I (66925) wifi_prov_scheme_ble: BTDM memory released
I (67825) app: Hello World!
I (68825) app: Hello World!
I (69825) app: Hello World!
I (70825) app: Hello World!

[image]

--

 So, will need some more information to narrow down the issue. 
 
 1. Can you please confirm if the IDF example too shows same behaviour for you ? 
 2. Can you share your complete application code that can be used by us for this issue ? 
 3. Also please share the sdkconfig file to confirm the features.

rahult-github, 2024-02-06:
Hi @monkeytronics , any further updates ?

rahult-github, 2024-03-26:
Close issue due to no further updates. Please feel free to reopen with details, if needed.

closed 2024-03-26 by rahult-github