espressif/esp-idf #12845: SOC_ADC_DIGI_MIN_BITWIDTH of esp32s2 is incorrect

nopnop2002, 2023-12-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

esp32s2

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s2

### Power Supply used.

USB

### What is the expected behavior?

SOC_ADC_DIGI_MIN_BITWIDTH for esp32 returns 13

### What is the actual behavior?

The ADC resolution of esp32s2 is 13 bits, but SOC_ADC_DIGI_MIN_BITWIDTH returns 12.

### Steps to reproduce.

Build this code.
```
#include <stdio.h>
#include <inttypes.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "soc/soc_caps.h"
#include "soc/adc_channel.h"
#include "esp_adc/adc_oneshot.h"
#include "esp_adc/adc_cali.h"
#include "esp_adc/adc_cali_scheme.h"

void app_main(void)
{
        adc_channel_t adc1_channel = ADC1_GPIO1_CHANNEL;
        int adc1_gpio = 1;

        // Configure ADC
        printf("SOC_ADC_MAX_CHANNEL_NUM=%d\n", SOC_ADC_MAX_CHANNEL_NUM);
        printf("SOC_ADC_DIGI_MIN_BITWIDTH=%d\n", SOC_ADC_DIGI_MIN_BITWIDTH);
        printf("SOC_ADC_DIGI_MAX_BITWIDTH=%d\n", SOC_ADC_DIGI_MAX_BITWIDTH);

    // ADC1 Init
    adc_oneshot_unit_handle_t adc1_handle;
    adc_oneshot_unit_init_cfg_t init_config1 = {
        .unit_id = ADC_UNIT_1,
    };
    ESP_ERROR_CHECK(adc_oneshot_new_unit(&init_config1, &adc1_handle));

    // ADC1 config
    adc_oneshot_chan_cfg_t config = {
        .bitwidth = ADC_BITWIDTH_DEFAULT,
        .atten = ADC_ATTEN_DB_0,
    };
    printf("adc1_gpio=%d adc1_channel=%d\n", adc1_gpio, adc1_channel);
    ESP_ERROR_CHECK(adc_oneshot_config_channel(adc1_handle, adc1_channel, &config));

        for (int i=0;i<10;i++) {
        int adc_raw;
        ESP_ERROR_CHECK(adc_oneshot_read(adc1_handle, adc1_channel, &adc_raw));
                printf("GPIO%02d adc1_channel: %d Raw: %d(0x%04x)\n", adc1_gpio, adc1_channel, adc_raw, adc_raw);
                vTaskDelay(100);
        }
}
```

### Debug Logs.

```plain
SOC_ADC_MAX_CHANNEL_NUM=10
SOC_ADC_DIGI_MIN_BITWIDTH=12
SOC_ADC_DIGI_MAX_BITWIDTH=12
adc1_gpio=1 adc1_channel=0
GPIO01 adc1_channel: 0 Raw: 8191(0x1fff)
GPIO01 adc1_channel: 0 Raw: 8191(0x1fff)
GPIO01 adc1_channel: 0 Raw: 8191(0x1fff)
GPIO01 adc1_channel: 0 Raw: 8191(0x1fff)
GPIO01 adc1_channel: 0 Raw: 8191(0x1fff)
GPIO01 adc1_channel: 0 Raw: 8191(0x1fff)
GPIO01 adc1_channel: 0 Raw: 8191(0x1fff)
GPIO01 adc1_channel: 0 Raw: 8191(0x1fff)
GPIO01 adc1_channel: 0 Raw: 8191(0x1fff)
GPIO01 adc1_channel: 0 Raw: 8191(0x1fff)
```
```

Icarus113, 2023-12-20:
```
SOC_ADC_DIGI_MIN_BITWIDTH=12
SOC_ADC_DIGI_MAX_BITWIDTH=12
```
is for continuous mode

nopnop2002, 2023-12-20:
ESP_ERROR_CHECK(adc_oneshot_config_channel(adc1_handle, adc1_channel, &config));

nopnop2002, 2023-12-20:
@Icarus113 

Where can I find the resolution in oneshot mode?

Icarus113, 2023-12-20:
S2 these two are for oneshot mode
```
#define SOC_ADC_RTC_MIN_BITWIDTH                (13)
#define SOC_ADC_RTC_MAX_BITWIDTH                (13)
```

BTW on S2 the bitwidth is always fixed, I'd suggest you using `ADC_BITWIDTH_DEFAULT`

nopnop2002, 2023-12-20:
>S2 these two are for oneshot mode

Thank you.

>BTW on S2 the bitwidth is always fixed, I'd suggest you using ADC_BITWIDTH_DEFAULT

ADC_BITWIDTH_DEFAULT is always 0.
```
SOC_ADC_DIGI_MIN_BITWIDTH=12
SOC_ADC_DIGI_MAX_BITWIDTH=12
SOC_ADC_RTC_MIN_BITWIDTH=13
SOC_ADC_RTC_MAX_BITWIDTH=13
ADC_BITWIDTH_DEFAULT=0
```

Icarus113, 2023-12-20:
This is difficult for current driver, as we don't think of adding a macro indicating the max bitwidth for ADC (under different modes).

On ESP32 / ESP32S2, `SOC_ADC_RTC_MIN/MAX_BITWIDTH` indicates the oneshot mode max bitwidth. Whereas on later chips `SOC_ADC_DIGI_MIN/MAX_BITWIDTH` is for that.

`ADC_BITWIDTH_DEFAULT` is used for `adc_oneshot_chan_cfg_t:bitwidth`, with this, when you call `adc_oneshot_config_channel` it will automatically select the max bitwidth.

nopnop2002, 2023-12-20:
ESP-IDF V4 had ADC_WIDTH_BIT_DEFAULT, which indicated the default bit number of the ADC, but it disappeared in ESP-IDF V5.

Thank you very much.

closed 2023-12-20 by nopnop2002

Icarus113, 2023-12-20:
v5 is a major version and the ADC new driver was introduced, some of the legacy driver and definitions are deprecated, you can find it in [adc_types_legacy.h](https://github.com/espressif/esp-idf/blob/master/components/driver/deprecated/driver/adc_types_legacy.h#L36)