espressif/esp-idf #12846: unit-test-app build fails on macos with --preview set-target linux

dajtxx, 2023-12-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I was able to build and run the component unit test on an ESP32, but I want to run them on the host.

I was hoping it would build and run as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/host-apps.html#build-and-run).

### What is the actual behavior?

The build fails as shown below.

### Steps to reproduce.

1. cd $IDF_PATH/tools/unit-test-app
2. Modify the unit-test-app CMakefiles.txt file so it can run on the host, as detailed in issue #11966 . Something I could not do was set the `CONFIG_UNITY_ENABLE_IDF_TEST_RUNNER` option because I cannot find the unity component in menuconfig.

```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.16)

list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/components/driver/test_apps/components")

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
set(COMPONENTS main)
set(EXTRA_COMPONENT_DIRS "/x/src/r5t/components")
set(TEST_COMPONENTS "sara_r5" CACHE STRING "Components to test")
project(unit-test-app)
```

3. idf.py fullclean

```
Executing action: fullclean
Executing action: remove_managed_components
Done
```

4. idf.py --preview set-target linux

```
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/x/src/esp-idf-v5.1.2/tools/unit-test-app/build' is empty. Nothing to clean.
Executing action: set-target
Set Target to: linux, new sdkconfig will be created.
Running cmake in directory /x/src/esp-idf-v5.1.2/tools/unit-test-app/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/x/.espressif/python_env/idf5.1_py3.9_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=linux -DCCACHE_ENABLE=0 /x/src/esp-idf-v5.1.2/tools/unit-test-app"...
-- Existing sdkconfig '/x/src/esp-idf-v5.1.2/tools/unit-test-app/sdkconfig' renamed to '/x/src/esp-idf-v5.1.2/tools/unit-test-app/sdkconfig.old'.
[... 60 lines cut ...]
/x/src/esp-idf-v5.1.2/components/freertos/FreeRTOS-Kernel/tasks.c:6160:77: error: result of comparison of constant 18446744073709551615 with expression of type 'volatile uint32_t' (aka 'volatile unsigned int') is always false [-Werror,-Wtautological-constant-out-of-range-compare]
                    configASSERT( pxTCB->ulNotifiedValue[ uxIndexToNotify ] == ~0UL );
                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~
/x/src/esp-idf-v5.1.2/components/freertos/esp_additions/include/freertos/FreeRTOSConfig.h:123:38: note: expanded from macro 'configASSERT'
#define configASSERT( a )    assert( a )
                                     ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk/usr/include/assert.h:99:25: note: expanded from macro 'assert'
    (__builtin_expect(!(e), 0) ? __assert_rtn(__func__, __ASSERT_FILE_NAME, __LINE__, #e) : (void)0)
                        ^
3 errors generated.
[25/42] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.oFAILED: esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.o
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -DESP_PLATFORM -DIDF_VER=\"v5.1.2-dirty\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -DprojCOVERAGE_TEST=0 -I/x/src/esp-idf-v5.1.2/tools/unit-test-app/build/config -I/x/src/esp-idf-v5.1.2/components/freertos/FreeRTOS-Kernel/include -I/x/src/esp-idf-v5.1.2/components/freertos/FreeRTOS-Kernel/portable/linux/include -I/x/src/esp-idf-v5.1.2/components/freertos/esp_additions/include/freertos -I/x/src/esp-idf-v5.1.2/components/freertos/esp_additions/include -I/x/src/esp-idf-v5.1.2/components/freertos/esp_additions/arch/linux/include -I/x/src/esp-idf-v5.1.2/components/esp_hw_support/include -I/x/src/esp-idf-v5.1.2/components/heap/include -I/x/src/esp-idf-v5.1.2/components/log/include -I/x/src/esp-idf-v5.1.2/components/soc/include -I/x/src/esp-idf-v5.1.2/components/soc/linux -I/x/src/esp-idf-v5.1.2/components/soc/linux/include -I/x/src/esp-idf-v5.1.2/components/esp_rom/include -I/x/src/esp-idf-v5.1.2/components/esp_rom/include/linux -I/x/src/esp-idf-v5.1.2/tools/mocks/soc/include -I/x/src/esp-idf-v5.1.2/components/esp_common/include -I/x/src/esp-idf-v5.1.2/components/esp_system/include -I/x/src/esp-idf-v5.1.2/components/linux/include -I/x/src/esp-idf-v5.1.2/components/esp_event/include -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.3.sdk -fcolor-diagnostics -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -Wno-documentation -Wno-typedef-redefinition -Wno-char-subscripts -Wno-format-security -Wno-tautological-overlap-compare -Wno-tautological-pointer-compare -Wno-pointer-bool-conversion -Wno-string-concatenation -Wno-section -Wno-unknown-attributes -Wno-atomic-alignment -Wno-unused-but-set-variable -Wno-unused-function -Wno-gnu-variable-sized-type-not-at-end -Wno-constant-logical-operand -Wno-c2x-extensions -Wno-extern-c-compat -Wwrite-strings -fstack-protector-strong -fmacro-prefix-map=/x/src/esp-idf-v5.1.2/tools/unit-test-app=. -fmacro-prefix-map=/x/src/esp-idf-v5.1.2=/IDF -fno-jump-tables -fno-use-cxa-atexit -std=gnu17 -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.o -MF esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.o.d -o esp-idf/main/CMakeFiles/__idf_main.dir/app_main.c.o -c /x/src/esp-idf-v5.1.2/tools/unit-test-app/main/app_main.c
/x/src/esp-idf-v5.1.2/tools/unit-test-app/main/app_main.c:1:10: fatal error: 'test_utils.h' file not found
#include "test_utils.h"
         ^~~~~~~~~~~~~~
1 error generated.
[32/42] Building C object esp-idf/freertos/CMakeFiles/__idf_freertos.dir/FreeRTOS-Kernel/stream_buffer.c.oninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /x/src/esp-idf-v5.1.2/tools/unit-test-app/build/log/idf_py_stderr_output_91982 and /x/src/esp-idf-v5.1.2/tools/unit-test-app/build/log/idf_py_stdout_output_91982
```

### More Information.

As mentioned above, I could not set the `CONFIG_UNITY_ENABLE_IDF_TEST_RUNNER` option because I cannot find the unity component in menuconfig.

The tests do compile and run on an ESP32.

igrr, 2023-12-20:
Thanks for reporting this @dajtxx. So far we haven't been checking the combination of IDF_TARGET=linux with macOS host, so in `master` this is now pretty much totally broken. We don't have a lot of macOS users in IDF team, so up until recently nobody stumbled upon this problem.

I am now gradually fixing the issues I found with IDF_TARGET=linux + macOS on `master`, and will add a CI job once all the issues are fixed, to prevent further regressions. Once that is all fixed on `master`, we'll take a look at the release branches as well. If you find a fix for this issue in the meantime on release/v5.1, please do post a patch and we'll try to get it merged.

HaydenDekker, 2025-05-09:
Cool, I've been looking for ways I can seperate Linux unity tests from ESP-IDF unity tests but have the same code base. Currently I've allocated a sub folder to host the linux application, and another to host the esp application. The component source and tests are as per the ESP-IDF residing in the /components folder and with include, src and test subfolders. In the test subfolder I am writing all base tests as plain c functions with unity assertions. Then I'm creating two test case files, one for the ESP-IDF's TEST_CASE, and the other for the linux's unity_fixture TEST() macros. These both refer to the test function. It's working well and keeping my concerns in check. I just stumbled on this thread. I'm trying to work out how I may simplifiy the unity test definitions, have a single test definition and running on the host target. I'm using VScode extension. I'm thinking this is the solution, just have to work out how to trigger it. Any advice? Thankyou Muchly.