espressif/esp-idf #12847: Failed to connect to the Amazon Cloud connection using mqtt's tcp routine

LHYYSY, 2023-12-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32 C3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-13

### Power Supply used.

USB

### What is the expected behavior?

Successful connection

### What is the actual behavior?

connection fail

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
I (25) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader
I (26) boot: compile time Dec 12 2023 11:59:34
I (26) boot: chip revision: v0.3
I (29) boot.esp32c3: SPI Speed      : 80MHz
I (34) boot.esp32c3: SPI Mode       : DIO  
I (39) boot.esp32c3: SPI Flash Size : 2MB  
I (43) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=25988h (153992) map
I (119) esp_image: segment 1: paddr=000359b0 vaddr=3fc90600 size=028bch ( 10428) load
I (121) esp_image: segment 2: paddr=00038274 vaddr=40380000 size=07da4h ( 32164) load
I (131) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=a8484h (689284) map
I (243) esp_image: segment 4: paddr=000e84ac vaddr=40387da4 size=08808h ( 34824) load
I (255) boot: Loaded app from partition at offset 0x10000
I (255) boot: Disabling RNG early entropy source...
I (267) cpu_start: Unicore app
I (267) cpu_start: Pro cpu up.
I (276) cpu_start: Pro cpu start user code
I (276) cpu_start: cpu freq: 160000000 Hz
I (276) cpu_start: Application information:
I (279) cpu_start: Project name:     mqtt_tcp
I (284) cpu_start: App version:      1
I (288) cpu_start: Compile time:     Dec 20 2023 15:27:53
I (294) cpu_start: ELF file SHA256:  53e6b0ce35baecc2...
I (300) cpu_start: ESP-IDF:          v5.1.1-dirty
I (306) cpu_start: Min chip rev:     v0.3
I (310) cpu_start: Max chip rev:     v0.99 
I (315) cpu_start: Chip rev:         v0.3
I (320) heap_init: Initializing. RAM available for dynamic allocation:
I (327) heap_init: At 3FC97540 len 000451D0 (276 KiB): DRAM
I (333) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (340) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (347) spi_flash: detected chip: generic
I (351) spi_flash: flash io: dio
W (355) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (368) sleep: Configure to isolate all GPIO pins in sleep state
I (375) sleep: Enable automatic switching of GPIO sleep configuration
I (382) app_start: Starting scheduler on CPU0
I (387) main_task: Started on CPU0
I (387) main_task: Calling app_main()
I (387) MQTT_EXAMPLE: [APP] Startup..
I (397) MQTT_EXAMPLE: [APP] Free memory: 285840 bytes
I (397) MQTT_EXAMPLE: [APP] IDF version: v5.1.1-dirty
I (417) example_connect: Start example_connect.
I (417) pp: pp rom version: 9387209
I (417) net80211: net80211 rom version: 9387209
I (427) wifi:wifi driver task: 3fc9fe4c, prio:23, stack:6656, core=0
I (427) wifi:wifi firmware version: ce9244d
I (427) wifi:wifi certification version: v7.0
I (437) wifi:config NVS flash: enabled
I (437) wifi:config nano formating: disabled
I (447) wifi:Init data frame dynamic rx buffer num: 32
I (447) wifi:Init management frame dynamic rx buffer num: 32
I (457) wifi:Init management short buffer num: 32
I (457) wifi:Init dynamic tx buffer num: 32
I (457) wifi:Init static tx FG buffer num: 2
I (467) wifi:Init static rx buffer size: 1600
I (467) wifi:Init static rx buffer num: 10
I (477) wifi:Init dynamic rx buffer num: 32
I (477) wifi_init: rx ba win: 6
I (477) wifi_init: tcpip mbox: 32
I (487) wifi_init: udp mbox: 6
I (487) wifi_init: tcp mbox: 6
I (497) wifi_init: tcp tx win: 5744
I (497) wifi_init: tcp rx win: 5744
I (497) wifi_init: tcp mss: 1440
I (507) wifi_init: WiFi IRAM OP enabled
I (507) wifi_init: WiFi RX IRAM OP enabled
I (517) phy_init: phy_version 970,1856f88,May 10 2023,17:44:12
I (557) wifi:mode : sta (84:f7:03:53:43:d4)
I (557) wifi:enable tsf
I (567) example_connect: Connecting to HUAWEI-0H14NV...
I (567) example_connect: Waiting for IP(s)
I (2967) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (3647) wifi:state: init -> auth (b0)
I (4647) wifi:state: auth -> init (200)
I (4647) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (4647) example_connect: Wi-Fi disconnected, trying to reconnect...
I (7057) example_connect: Wi-Fi disconnected, trying to reconnect...
I (9467) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (9467) wifi:state: init -> auth (b0)
I (9497) wifi:state: auth -> assoc (0)
I (9507) wifi:state: assoc -> run (10)
I (9537) wifi:connected with HUAWEI-0H14NV, aid = 7, channel 1, 40U, bssid = 30:aa:e4:34:27:80
I (9537) wifi:security: WPA2-PSK, phy: bgn, rssi: -44
I (9537) wifi:pm start, type: 1

I (9537) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (9547) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (10577) wifi:<ba-add>idx:0 (ifx:0, 30:aa:e4:34:27:80), tid:0, ssn:1, winSize:64
I (10577) wifi:<ba-del>idx:0, tid:0
I (10587) wifi:<ba-add>idx:0 (ifx:0, 30:aa:e4:34:27:80), tid:0, ssn:1, winSize:64
I (10587) wifi:<ba-del>idx:0, tid:0
I (10597) wifi:<ba-add>idx:0 (ifx:0, 30:aa:e4:34:27:80), tid:0, ssn:1, winSize:64
I (11047) esp_netif_handlers: example_netif_sta ip: 192.168.3.65, mask: 255.255.255.0, gw: 192.168.3.1
I (11047) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.3.65
I (11407) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:86f7:03ff:fe53:43d4, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (11407) example_common: Connected to example_netif_sta
I (11417) example_common: - IPv4 address: 192.168.3.65,
I (11417) example_common: - IPv6 address: fe80:0000:0000:0000:86f7:03ff:fe53:43d4, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (11427) MQTT_EXAMPLE: Other event id:7
I (11437) main_task: Returned from app_main()
E (21517) esp-tls: [sock=54] select() timeout
E (21517) transport_base: Failed to open a new connection: 32774
E (21517) mqtt_client: Error transport connect
I (21517) MQTT_EXAMPLE: MQTT_EVENT_ERROR
E (21527) MQTT_EXAMPLE: Last error reported from esp-tls: 0x8006
I (21527) MQTT_EXAMPLE: Last errno string (Success)
I (21537) MQTT_EXAMPLE: MQTT_EVENT_DISCONNECTED
I (36537) MQTT_EXAMPLE: Other event id:7
E (46547) esp-tls: [sock=54] select() timeout
E (46547) transport_base: Failed to open a new connection: 32774
E (46547) mqtt_client: Error transport connect
I (46547) MQTT_EXAMPLE: MQTT_EVENT_ERROR
E (46557) MQTT_EXAMPLE: Last error reported from esp-tls: 0x8006
I (46557) MQTT_EXAMPLE: Last errno string (Success)
I (46567) MQTT_EXAMPLE: MQTT_EVENT_DISCONNECTED
I (61567) MQTT_EXAMPLE: Other event id:7
E (71577) esp-tls: [sock=54] select() timeout
E (71577) transport_base: Failed to open a new connection: 32774
E (71577) mqtt_client: Error transport connect
I (71577) MQTT_EXAMPLE: MQTT_EVENT_ERROR
E (71587) MQTT_EXAMPLE: Last error reported from esp-tls: 0x8006
I (71587) MQTT_EXAMPLE: Last errno string (Success)
I (71597) MQTT_EXAMPLE: MQTT_EVENT_DISCONNECTED
I (86597) MQTT_EXAMPLE: Other event id:7
```

LHYYSY, 2023-12-20:
[image]

LHYYSY, 2023-12-20:
[image]

LHYYSY, 2023-12-20:
[image]

ESP-YJM, 2023-12-22:
It seems TCP connection failed due to TCP SYN no response. Could you please help check the domain and port can access with TCP?

LHYYSY, 2023-12-22:
> It seems TCP connection failed due to TCP SYN no response. Could you please help check the domain and port can access with TCP?

LHYYSY, 2023-12-22:
> It seems TCP connection failed due to TCP SYN no response. Could you please help check the domain and port can access with TCP?

Do I want to make sure that tcp routines using mqtt to connect to Amazon Cloud can connect? This process does not use SSL/TSL to encrypt the transmission

ESP-YJM, 2023-12-22:
Yes, you can test whether TCP can access.

LHYYSY, 2023-12-22:
> 是的，您可以测试 TCP 是否可以访问。
I tried to use tcp to connect to other clouds, such as Alibaba Cloud, but connected to Amazon Cloud times wrong, but I can use mqtt tls connection, so I suspect Amazon cloud does not support tcp connection, right?

LHYYSY, 2023-12-22:
> Yes, you can test whether TCP can access.

I tried to use tcp to connect to other clouds, such as Alibaba Cloud, but connected to Amazon Cloud times wrong, but I can use mqtt tls connection, so I suspect Amazon cloud does not support tcp connection, right?

ESP-YJM, 2023-12-22:
Yes, i think the Amazon not open port 1883, because this is for mqtt over TCP. Maybe you should connect it with port 443 and mqtt over TLS.

ESP-YJM, 2023-12-25:
@LHYYSY Any update for this issue?

LHYYSY, 2023-12-25:
> 此问题有任何更新吗？

Thank you very much for your answer. I think I have solved it.

LHYYSY, 2023-12-25:
> @LHYYSY Any update for this issue?

Thank you very much for your answer. I think I have solved it.

closed 2023-12-25 by LHYYSY

vectorcrumb, 2024-04-27:
@SeanYang-7 what was the solution?