espressif/esp-idf #12850: Cannot build bootloader on IDFv5.1 for ESP32S3 + 32MB OSPI flash due to bootloader API missing

huming2207, 2023-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2-472-gb38ee33b6d

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build successfully

### What is the actual behavior?

Got build error like this, as long as the Kconfig's `CONFIG_ESPTOOLPY_OCT_FLASH=y` and `CONFIG_SPI_FLASH_32BIT_ADDRESS=y` (i.e. build any firmware for ESP32-S3-WROOM-2-N32R8V modules with octal SPI enabled):

```
/Users/hu/esp/esp-idf/components/bootloader_support/bootloader_flash/src/bootloader_flash_config_esp32s3.c: In function 'bootloader_init_spi_flash':
/Users/hu/esp/esp-idf/components/bootloader_support/bootloader_flash/src/bootloader_flash_config_esp32s3.c:276:48: error: implicit declaration of function 'bootloader_flash_get_spi_mode'; did you mean 'bootloader_flash_set_dummy_out'? [-Werror=implicit-function-declaration]
  276 |     bootloader_flash_32bits_address_map_enable(bootloader_flash_get_spi_mode());
      |                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                bootloader_flash_set_dummy_out
cc1: some warnings being treated as errors
```

### Steps to reproduce.

1. Clone the latest ESP-IDF v5.1 by `git clone https://github.com/espressif/esp-idf --recursive -b release/v5.1`
2. Run `install.sh`
3. Run `export.sh`
4. Grab a `hello_world` project from the `examples`
5. Run `idf.py set-target esp32s3`
6. Run `idf.py menuconfig`, enable `CONFIG_ESPTOOLPY_OCT_FLASH=y` and `CONFIG_SPI_FLASH_32BIT_ADDRESS=y` by going to "Serial flasher config" and tick "Enable Octal Flash", and select "Flash size" to 32MB:

[image]

7. Then compile by `idf.py build`, you will see the error.

### Build or installation Logs.

```plain
hu@hu-mac  ~/Projects/hello_world  idf.py build
Executing action: all (aliases: build)
Running ninja in directory /Users/hu/Projects/hello_world/build
Executing "ninja all"...
[0/1] Re-running CMake...-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file /Users/hu/Projects/hello_world/sdkconfig
Compiler supported targets: xtensa-esp32s3-elf

-- App "hello_world" version: 1
-- Adding linker script /Users/hu/Projects/hello_world/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/hu/esp/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script /Users/hu/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /Users/hu/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
[... 8 lines cut ...]
-- Build files have been written to: /Users/hu/Projects/hello_world/build
[9/945] Generating ../../partition_table/partition-table.binPartition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[149/945] Building C object esp-idf/bootloader_support/CMakeFil....dir/bootloader_flash/src/bootloader_flash_config_esp32s3.c.objFAILED: esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/bootloader_flash/src/bootloader_flash_config_esp32s3.c.obj
/Users/hu/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.1.2-472-gb38ee33b6d\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/hu/Projects/hello_world/build/config -I/Users/hu/esp/esp-idf/components/bootloader_support/include -I/Users/hu/esp/esp-idf/components/bootloader_support/bootloader_flash/include -I/Users/hu/esp/esp-idf/components/bootloader_support/private_include -I/Users/hu/esp/esp-idf/components/newlib/platform_include -I/Users/hu/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/hu/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/hu/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/Users/hu/esp/esp-idf/components/freertos/esp_additions/include -I/Users/hu/esp/esp-idf/components/freertos/esp_additions/arch/xtensa/include -I/Users/hu/esp/esp-idf/components/esp_hw_support/include -I/Users/hu/esp/esp-idf/components/esp_hw_support/include/soc -I/Users/hu/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/Users/hu/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -I/Users/hu/esp/esp-idf/components/esp_hw_support/port/esp32s3/private_include -I/Users/hu/esp/esp-idf/components/heap/include -I/Users/hu/esp/esp-idf/components/log/include -I/Users/hu/esp/esp-idf/components/soc/include -I/Users/hu/esp/esp-idf/components/soc/esp32s3 -I/Users/hu/esp/esp-idf/components/soc/esp32s3/include -I/Users/hu/esp/esp-idf/components/hal/esp32s3/include -I/Users/hu/esp/esp-idf/components/hal/include -I/Users/hu/esp/esp-idf/components/hal/platform_port/include -I/Users/hu/esp/esp-idf/components/esp_rom/include -I/Users/hu/esp/esp-idf/components/esp_rom/include/esp32s3 -I/Users/hu/esp/esp-idf/components/esp_rom/esp32s3 -I/Users/hu/esp/esp-idf/components/esp_common/include -I/Users/hu/esp/esp-idf/components/esp_system/include -I/Users/hu/esp/esp-idf/components/esp_system/port/soc -I/Users/hu/esp/esp-idf/components/esp_system/port/include/private -I/Users/hu/esp/esp-idf/components/xtensa/include -I/Users/hu/esp/esp-idf/components/xtensa/esp32s3/include -I/Users/hu/esp/esp-idf/components/lwip/include -I/Users/hu/esp/esp-idf/components/lwip/include/apps -I/Users/hu/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/hu/esp/esp-idf/components/lwip/lwip/src/include -I/Users/hu/esp/esp-idf/components/lwip/port/include -I/Users/hu/esp/esp-idf/components/lwip/port/freertos/include -I/Users/hu/esp/esp-idf/components/lwip/port/esp32xx/include -I/Users/hu/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/Users/hu/esp/esp-idf/components/spi_flash/include -I/Users/hu/esp/esp-idf/components/mbedtls/port/include -I/Users/hu/esp/esp-idf/components/mbedtls/mbedtls/include -I/Users/hu/esp/esp-idf/components/mbedtls/mbedtls/library -I/Users/hu/esp/esp-idf/components/mbedtls/esp_crt_bundle/include -I/Users/hu/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -I/Users/hu/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -I/Users/hu/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -I/Users/hu/esp/esp-idf/components/efuse/include -I/Users/hu/esp/esp-idf/components/efuse/esp32s3/include -I/Users/hu/esp/esp-idf/components/esp_app_format/include -mlongcalls  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=/Users/hu/Projects/hello_world=. -fmacro-prefix-map=/Users/hu/esp/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/bootloader_flash/src/bootloader_flash_config_esp32s3.c.obj -MF esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/bootloader_flash/src/bootloader_flash_config_esp32s3.c.obj.d -o esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/bootloader_flash/src/bootloader_flash_config_esp32s3.c.obj -c /Users/hu/esp/esp-idf/components/bootloader_support/bootloader_flash/src/bootloader_flash_config_esp32s3.c
/Users/hu/esp/esp-idf/components/bootloader_support/bootloader_flash/src/bootloader_flash_config_esp32s3.c: In function 'bootloader_init_spi_flash':
[... cut ...]
[158/945] Building C object esp-idf/app_update/CMakeFiles/__idf_app_update.dir/esp_ota_ops.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/hu/Projects/hello_world/build/log/idf_py_stderr_output_17111 and /Users/hu/Projects/hello_world/build/log/idf_py_stdout_output_17111
```
```

### More Information.

This was working fine in previous IDF 5.1 codebase (at least definitely working fine in less than 20 days ago). Somehow it's broken due to some latest changes on ESP-IDF 5.1 branch.

huming2207, 2023-12-21:
I can confirm this issue doesn't exist on IDF 5.1.2 release, i.e. https://github.com/espressif/esp-idf/commit/482a8fb2d78e3b58eb21b26da8a5bedf90623213

huming2207, 2023-12-21:
I had a quick look and I think this issue is a regression from this commit: https://github.com/espressif/esp-idf/commit/bb7544e65aba750aad7dfd641a2b166b9e0a2ffe

cc @igrr @mythbuster5 @ginkgm

mythbuster5, 2023-12-22:
Thanks for the reporting, fixing..

Alvin1Zhang, 2024-01-12:
Thanks for reporting and sharing the updates, the fix is available, feel free to reopen.

closed 2024-01-12 by Alvin1Zhang