espressif/esp-idf #12851: Imposible to Revoke Secure Boot public key

MigliusBudriunas, 2023-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.2

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

In esp_ota_ops.c file in esp_ota_revoke_secure_boot_public_key finction in line 943 "all_zeroes = all_zeroes && (*(uint8_t *)(trusted_keys.key_digests[i] + j) == 0);" trusted_keys should return SECURE_BOOT_DIGEST1/2/3. Read access is not locked.

### What is the actual behavior?

In esp_ota_ops.c file in esp_ota_revoke_secure_boot_public_key finction in line 943 "all_zeroes = all_zeroes && (*(uint8_t *)(trusted_keys.key_digests[i] + j) == 0);" trusted_keys is always returned as 0.
It looks like you can't access it as 8bit registers. It returns real keys if you access it as 32bit registers.

### Steps to reproduce.

1. Flash secure boot keys 1/2/3
2. Add following code in esp_ota_revoke_secure_boot_public_key function after 952 line. This code prints Secure Boot Digests just accesses the efuse registers as 8bit and 32bit. 32bit digest is printed in different MSB format than espesfuse.py summary.

    for (unsigned j = 0; j < 3; j++) {
		printf("\n");
		printf("Boot digest %d: ", j);
		printf("\n  8bit access: ");
		for (int i = 0; i < ESP_SECURE_BOOT_DIGEST_LEN; i++)
			printf("%02X", *((uint8_t *)trusted_keys.key_digests[j] + i));
		printf("\n 32bit access: ");
		for (int i = 0; i < ESP_SECURE_BOOT_DIGEST_LEN/4; i++)
			printf("%08X", *((unsigned int *)trusted_keys.key_digests[j] + i));
	}
3. Step call esp_ota_revoke_secure_boot_public_key function. This function always fails to revoke keys.
...

### Debug Logs.

```plain
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:2
Valid secure boot key blocks: 0 1 2       
[... 276 lines cut ...]
USB_EXCHG_PINS (BLOCK0)                            Set this bit to exchange USB D+ and D- pins        = False R/W (0b0)
USB_EXT_PHY_ENABLE (BLOCK0)                        Set this bit to enable external PHY                = False R/W (0b0)
DIS_USB_JTAG (BLOCK0)                              Set this bit to disable function of usb switch to  = False R/W (0b0)
                                                   jtag in module of usb device
DIS_USB_SERIAL_JTAG (BLOCK0)                       Set this bit to disable usb device                 = False R/W (0b0)
USB_PHY_SEL (BLOCK0)                               This bit is used to switch internal PHY and extern
   = internal PHY is assigned to USB Device while external PHY is assigned to USB OTG R/W (0b0)
                                                   al PHY for USB OTG and USB Device
DIS_USB_SERIAL_JTAG_ROM_PRINT (BLOCK0)             USB printing                                       = Enable R/W (0b0)
DIS_USB_SERIAL_JTAG_DOWNLOAD_MODE (BLOCK0)         Set this bit to disable UART download mode through = False R/W (0b0)
                                                    USB
DIS_USB_OTG_DOWNLOAD_MODE (BLOCK0)                 Set this bit to disable download through USB-OTG   = False R/W (0b0)

Vdd fuses:
VDD_SPI_XPD (BLOCK0)                               SPI regulator power up signal                      = True R/W (0b1)
VDD_SPI_TIEH (BLOCK0)                              If VDD_SPI_FORCE is 1; determines VDD_SPI voltage
   = VDD_SPI connects to 1.8 V LDO R/W (0b0)
VDD_SPI_FORCE (BLOCK0)                             Set this bit and force to use the configuration of = True R/W (0b1)
                                                    eFuse to configure VDD_SPI

Wdt fuses:
WDT_DELAY_SEL (BLOCK0)                             RTC watchdog timeout threshold; in unit of slow cl = 40000 R/W (0b00)
                                                   ock cycle

Flash voltage (VDD_SPI) set to 1.8V by efuse.

hmalpani, 2023-12-26:
Hello @MigliusBudriunas 
I think you are unable to revoke the key because as the logs suggest, your application might not be signed with another key. Can you please try signing the application with two different keys and then try to revoke a key. Let me know if this helps.

Thanks!

MigliusBudriunas, 2023-12-28:
That is not the problem. Every trusted key in trusted_keys struct is being marked as "Empty trusted key block..." in line 948. Keys are not being copied to trusted_digests struct, because line 946 is not being executed.

hmalpani, 2024-01-03:
Hello @MigliusBudriunas 
Yes you are right. It seems to be a bug. Reading from the peripheral should be 32 bit aligned. I will create a fix for this issue. Until then can you help me with sharing your sdkconfig file. Have you enabled any optimization in the menuconfig?

MigliusBudriunas, 2024-01-03:
CONFIG_COMPILER_OPTIMIZATION_DEFAULT=y
I don't think that default optimizations was changed. I attach sdkconfig:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/13816463/sdkconfig.txt)

hmalpani, 2024-01-09:
Hello @MigliusBudriunas 
Recently we updated the key revocation which is more secure as it verifies the bootloader as well before revoking the key in efuse. Can you try cherry-picking the commit and please let me know if that helps to solve the issue.

`git cherry-pick 1df186d4e1a7797d7fbd8a2edf5504c7b8a4f59f --strategy-option theirs`

If this works, we can backport the commit to previous releases.

MigliusBudriunas, 2024-01-12:
Everything works now, thanks. Great work 👍