espressif/esp-idf #12853: Failed spratically to connect with MQTT and end in a lockup state, with watchdog timer being triggered

al-micro, 2023-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-rc2

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

Sample data with ULP, after 2 minutes wakeup and upload data through mqtt with a pem certificate.

### What is the actual behavior?

Sporatically ends in a lockup state tying to connect with MQTT.

### Steps to reproduce.

The current FW is tested on several boards, one work cycle for sampling data and uploading with MQTT is ~2.5min. The issue have orccured after different intervals, for the same board this could happen within half an hour and next time 48 hours runtime.

### Debug Logs.

```plain
idf.py monitor --timestamps

2023-12-21 09:09:37
2023-12-21 09:09:41 E (27468952) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
2023-12-21 09:09:41 E (27468952) task_wdt:  - IDLE (CPU 1)
2023-12-21 09:09:41 E (27468952) task_wdt: Tasks currently running:
2023-12-21 09:09:41 E (27468952) task_wdt: CPU 0: IDLE
2023-12-21 09:09:41 E (27468952) task_wdt: CPU 1: mqtt_task
2023-12-21 09:09:41 E (27468952) task_wdt: Print CPU 1 backtrace
2023-12-21 09:09:41
2023-12-21 09:09:41
2023-12-21 09:09:41 Backtrace: 0x40378F5F:0x3FC9C370 0x403773A1:0x3FC9C390 0x42093807:0x3FCC59F0 0x4206ED7B:0x3FCC5A10 0x4206BA64:0x3FCC5A50 0x4206A13E:0x3FCC5A80 0x42066961:0x3FCC5B80 0x42066A1D:0x3FCC5BF0 0x42066A51:0x3FCC5CA0 0x42066B81:0x3FCC5D10 0x420636A9:0x3FCC5D30 0x420637AD:0x3FCC5ED0 0x4205853E:0x3FCC5F10 0x420579A0:0x3FCC5F50 0x42057BD3:0x3FCC5F90 0x42058A8E:0x3FCC5FD0 0x420C30AB:0x3FCC6000 0x420168D7:0x3FCC6020
2023-12-21 09:09:41 0x40378f5f: esp_crosscore_isr at /home/mt/esp/esp-idf-v5.1/components/esp_system/crosscore_int.c:96

2023-12-21 09:09:41 0x403773a1: _xt_lowint1 at /home/mt/esp/esp-idf-v5.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1226
2023-12-21 09:09:41 0x42093807: aes_ll_dma_enable at /home/mt/esp/esp-idf-v5.1/components/hal/esp32s3/include/hal/aes_ll.h:200
2023-12-21 09:09:41  (inlined by) aes_hal_transform_dma_start at /home/mt/esp/esp-idf-v5.1/components/hal/aes_hal.c:55
2023-12-21 09:09:41 0x4206ed7b: mbedtls_base64_decode at /home/mt/esp/esp-idf-v5.1/components/mbedtls/mbedtls/library/base64.c:176
2023-12-21 09:09:41 0x4206ba64: gcm_mult at /home/mt/esp/esp-idf-v5.1/components/mbedtls/port/aes/esp_aes_gcm.c:230 (discriminator 2)
2023-12-21 09:09:41 0x4206a13e: esp_aes_crypt_xts at /home/mt/esp/esp-idf-v5.1/components/mbedtls/port/aes/esp_aes_xts.c:284
2023-12-21 09:09:41 0x42066961: mbedtls_entropy_free at /home/mt/esp/esp-idf-v5.1/components/mbedtls/mbedtls/library/entropy.c:101
2023-12-21 09:09:41  (inlined by) mbedtls_entropy_free at /home/mt/esp/esp-idf-v5.1/components/mbedtls/mbedtls/library/entropy.c:81
2023-12-21 09:09:41 0x42066a1d: mbedtls_md_free at /home/mt/esp/esp-idf-v5.1/components/mbedtls/mbedtls/library/md.c:236
2023-12-21 09:09:41 0x42066a51: mbedtls_md_free at /home/mt/esp/esp-idf-v5.1/components/mbedtls/mbedtls/library/md.c:294
2023-12-21 09:09:41 0x42066b81: mbedtls_md_setup at /home/mt/esp/esp-idf-v5.1/components/mbedtls/mbedtls/library/md.c:415 (discriminator 2)
2023-12-21 09:09:41 0x420636a9: mbedtls_ctr_drbg_random_with_add at /home/mt/esp/esp-idf-v5.1/components/mbedtls/mbedtls/library/ctr_drbg.c:556
2023-12-21 09:09:41 0x420637ad: ecdh_free_internal at /home/mt/esp/esp-idf-v5.1/components/mbedtls/mbedtls/library/ecdh.c:223
2023-12-21 09:09:41  (inlined by) mbedtls_ecdh_free at /home/mt/esp/esp-idf-v5.1/components/mbedtls/mbedtls/library/ecdh.c:264
2023-12-21 09:09:41 0x4205853e: esp_transport_list_clean at /home/mt/esp/esp-idf-v5.1/components/tcp_transport/transport.c:83
2023-12-21 09:09:41 0x420579a0: esp_tls_plain_tcp_connect at /home/mt/esp/esp-idf-v5.1/components/esp-tls/esp_tls.c:509
2023-12-21 09:09:42 0x42057bd3: esp_mbedtls_init_pk_ctx_for_ds at /home/mt/esp/esp-idf-v5.1/components/esp-tls/esp_tls_mbedtls.c:992
2023-12-21 09:09:42  (inlined by) set_pki_context at /home/mt/esp/esp-idf-v5.1/components/esp-tls/esp_tls_mbedtls.c:382
2023-12-21 09:09:42 0x42058a8e: tcp_read at /home/mt/esp/esp-idf-v5.1/components/tcp_transport/transport_ssl.c:292 (discriminator 3)
2023-12-21 09:09:42 0x420c30ab: esp_transport_ssl_set_cert_data_der at ??:?
2023-12-21 09:09:42 0x420168d7: esp_mqtt_set_ssl_transport_properties at /home/mt/esp/esp-idf-v5.1/components/mqtt/esp-mqtt/mqtt_client.c:151
2023-12-21 09:09:42  (inlined by) esp_mqtt_task at /home/mt/esp/esp-idf-v5.1/components/mqtt/esp-mqtt/mqtt_client.c:1556
```
```

amrsoll, 2023-12-21:
I want to add some clarification to the bug report.

To boil it down a bit more, this trace happens every 10 000 -ish usage of the MQTT client. The watchdog resets the task itself, but it goes right back to printing that trace, in a loop.

In the meantime, we can live with this error if we can have some guidance for how we can ask the watchdog to soft-reset instead of just resetting the MQTT task, that would be very helpful 😄 

Is there a known issue with AES hardware acceleration?

amrsoll, 2023-12-23:
this also seems to be hardware dependent. we only experience this with 30% of our boards

AxelLin, 2023-12-23:
> this also seems to be hardware dependent. we only experience this with 30% of our boards

Do you mean it only happens on certain boards?  Some boards never have such issue?

amrsoll, 2023-12-23:
Exactly, we have only been able to experience it on certain boards, although it is a significant number of them.

amrsoll, 2024-01-02:
We can also help with recreating the issue @euripedesrocha 

Do you have a quick fix for triggering a soft system reset rather than a task reset from the watchdog?

euripedesrocha, 2024-01-03:
@amrsoll There is no known issue with hardware accelerator, you could try without it to isolate. 

I would appreciate it if you can provide a reproducer, so I can try to identify the possible source of the issue.

euripedesrocha, 2024-04-02:
I'm closing this since I couldn't reproduce the issue. 
Feel free to reopen, adding more information.

closed 2024-04-02 by euripedesrocha

maenkai, 2024-04-11:
E (176656) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (176656) task_wdt:  - IDLE (CPU 1)
E (176656) task_wdt: Tasks currently running:
E (176656) task_wdt: CPU 0: IDLE
E (176656) task_wdt: CPU 1: mqtt_task
E (176656) task_wdt: Aborting.

abort() was called at PC 0x40193ff6 on core 0

Backtrace:0x400910e6:0x3ffbf310 0x40091a09:0x3ffbf330 0x40098bba:0x3ffbf350 0x40193ff6:0x3ffbf3c0 0x40083109:0x3ffbf3e0 0x401f4cf7:0x3ffbd860 0x400d99b9:0x3ffbd880 0x40092f67:0x3ffbd8a0

ELF file SHA256: f94520234e1b4e89

Rebooting...
ets Jun  8 2016 00:22:57

I also have this problem.Please help me to slove it.
Make an appearance
```shell
0x400910e6: panic_abort at /home/zy/esp/esp-idf/components/esp_system/panic.c:379
0x40091a09: esp_system_abort at /home/zy/esp/esp-idf/components/esp_system/system_api.c:112
0x40098bba: abort at /home/zy/esp/esp-idf/components/newlib/abort.c:46
0x40193ff6: task_wdt_isr at /home/zy/esp/esp-idf/components/esp_common/src/task_wdt.c:182 (discriminator 1)
0x40083109: _xt_lowint1 at /home/zy/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105
0x401f4cf7: cpu_ll_waiti at /home/zy/esp/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at /home/zy/esp/esp-idf/components/esp_pm/pm_impl.c:827
0x400d99b9: esp_vApplicationIdleHook at /home/zy/esp/esp-idf/components/esp_common/src/freertos_hooks.c:63
0x40092f67: prvIdleTask at /home/zy/esp/esp-idf/components/freertos/tasks.c:3839

maenkai, 2024-04-11:
> E （176656） task_wdt：任务看门狗被触发。以下任务未及时重置看门狗： E （176656） task_wdt： - 空闲 （CPU 1） E （176656） task_wdt： 当前正在运行的任务： E （176656） task_wdt： CPU 0： 空闲 E （176656） task_wdt： CPU 1： mqtt_task E （176656） task_wdt： 中止。
> 
> abort（） 在 PC 0x40193ff6 上调用了核心 0
> 
> 回溯：0x400910e6：0x3ffbf310 0x40091a09：0x3ffbf330 0x40098bba：0x3ffbf350 0x40193ff6：0x3ffbf3c0 0x40083109：0x3ffbf3e0 0x401f4cf7：0x3ffbd860 0x400d99b9：0x3ffbd880 0x40092f67：0x3ffbd8a0
> 
> ELF 文件 SHA256： f94520234e1b4e89
> 
> 重 启。。。 ets Jun 8 2016 00：22：57
> 
> 我也有这个问题。请帮我把它弄好。出场
> 
> ```shell
> 0x400910e6: panic_abort at /home/zy/esp/esp-idf/components/esp_system/panic.c:379
> 0x40091a09: esp_system_abort at /home/zy/esp/esp-idf/components/esp_system/system_api.c:112
> 0x40098bba: abort at /home/zy/esp/esp-idf/components/newlib/abort.c:46
> 0x40193ff6: task_wdt_isr at /home/zy/esp/esp-idf/components/esp_common/src/task_wdt.c:182 (discriminator 1)
> 0x40083109: _xt_lowint1 at /home/zy/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105
> 0x401f4cf7: cpu_ll_waiti at /home/zy/esp/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
>  (inlined by) esp_pm_impl_waiti at /home/zy/esp/esp-idf/components/esp_pm/pm_impl.c:827
> 0x400d99b9: esp_vApplicationIdleHook at /home/zy/esp/esp-idf/components/esp_common/src/freertos_hooks.c:63
> 0x40092f67: prvIdleTask at /home/zy/esp/esp-idf/components/freertos/tasks.c:3839
> ```

My problem only appears on the hardware, not on the development board, maybe the problem depends on the hardware?

euripedesrocha, 2024-04-11:
@maenkai the watchdog could be triggered in many situations, could you please open a new issue describing your problem and giving enough information for us to reproduce the issue?

maenkai, 2024-04-12:
> ```shell
> 0x400910e6: panic_abort at /home/zy/esp/esp-idf/components/esp_system/panic.c:379
> 0x40091a09: esp_system_abort at /home/zy/esp/esp-idf/components/esp_system/system_api.c:112
> 0x40098bba: [...]

OK , this is my issue link https://github.com/espressif/esp-idf/issues/13596