espressif/esp-idf #12855: Random time 5min-hours crash in timer ISR on invoke spi transaction

mmar22, 2023-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

framework-arduinoespressif32 @ 3.20009.0 (2.0.9)

### Espressif SoC revision.

ESP32S3 v0.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom

### Power Supply used.

External 3.3V

### What is the expected behavior?

In timer interrupt 👍 
`void IRAM_ATTR irqCallbackStopFunc()
{
//on moment low pwm  
  gpio_set_level(DSP_EN,1);

  esp_err_t ret;
//start spi and complete set EN low
	myled.nextline+=1;
	if(myled.nextline>7) myled.nextline=0;
    transline.tx_buffer=LEDBufferlines[myled.nextline];        //finally send the line data
    transline.length=7*8;          //Data length, in bits
		
	  ret=spi_device_queue_trans(spiDISP, &transline, portMAX_DELAY);
}`
this works ok  but randomly on last line crash

### What is the actual behavior?

random crash

### Steps to reproduce.

1. used timer to generate pwm ()
2. in timer end cycle callback call code
3. sometimes crash after 5min sometimes after hours
...
cant def reproduce only Dependency Graph
|-- lvgl @ 8.3.7
|-- GFX Library for Arduino @ 1.3.4
|-- **ESP32_PWM @ 1.3.3**
|-- IRremoteESP8266 @ 2.8.4
|-- ESP32-audioI2S @ 2.1.0
|-- Wire @ 2.0.0
|-- ft3267
|-- Preferences @ 2.0.0
|-- WiFi @ 2.0.0
|-- spdif

### Debug Logs.

```plain
assert failed: xQueueGenericSend queue.c:825 (!( ( xTaskGetSchedulerState() == ( ( BaseType_t ) 0 ) ) 
&& ( xTicksToWait != 0 ) ))

Backtrace: 0x40377d92:0x3fc98290 0x4037d955:0x3fc982b0 0x40384b25:0x3fc982d0 0x4037e40b:0x3fc98400 0x4205de95:0x3fc98440 0x403754ed:0x3fc98480 0x40375576:0x3fc984a0 0x40375624:0x3fc984d0 0x40375e40:0x3fc984f0 0x40379295:0x3fc98510 0x400559dd:0x3fcf1560 |<-CORRUPTED

  #0  0x40377d92:0x3fc98290 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4037d955:0x3fc982b0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40384b25:0x3fc982d0 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x4037e40b:0x3fc98400 in xQueueGenericSend at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c:821 (discriminator 2)
  #4  0x4205de95:0x3fc98440 in spi_device_queue_trans at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/spi_master.c:810
      (inlined by) spi_device_queue_trans at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/spi_master.c:786
  #5  0x403754ed:0x3fc98480 in irqCallbackStopFunc() at src/matrix_led.cpp:68 (discriminator 1)       
  #6  0x40375576:0x3fc984a0 in ESP32_PWM::run() at .pio/libdeps/esp32s3mykorvo/ESP32_PWM/src/ESP32_PWM_ISR_Impl.h:126
  #7  0x40375624:0x3fc984d0 in TimerHandler(void*) at src/matrix_led.cpp:45
  #8  0x40375e40:0x3fc984f0 in timer_isr_default at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/timer.c:208
      (inlined by) timer_isr_default at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/timer.c:186
  #9  0x40379295:0x3fc98510 in _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/xtensa_vectors.S:1114
  #10 0x400559dd:0x3fcf1560 in ?? ??:0
```

### More Information.

i comment out this one last line and all works anytime ok.

`ret=spi_device_queue_trans(spiDISP, &transline, portMAX_DELAY);`

My project is big and i dont test update PIO framework to latest. I mean is irelevant.

igrr, 2023-12-21:
Please see https://github.com/espressif/esp-idf/issues/12260#issuecomment-1723068405.

mmar22, 2023-12-21:
Ofcourse i read this before write my ticket. "I think that function is not allowed to run in an ISR context." I think isnt real result. And my code dont use SPI in other place as this line then what semaphore? Call this func return OK allways and when crash then return nothing because crash inside spidriver.

mmar22, 2023-12-23:
One new info i upgrade to 2.0.11 (pio 6.4.0) and now crash is immediately and always. Need help with spi dma in timer isr

mmar22, 2023-12-28:
Nobody? I reqiure DMA offload perfectly timed SPI transfers. Seems on 1.x works, 2.0.9 work some minutes and on 2.0.11 crash .
I know its arduino espidf builds, but problem is in rtos core and non existence SPI func for ISR. I require help with this.

Now only working solution is change to 
`ret=spi_device_polling_transmit(spiDISP, &transline);`
but this is no DMA and no offload MCU

pnt325, 2024-01-31:
hi @mmar22 

Because on `spi_device_queue_trans` call to `xQueueSend` instead of `xQueueSendFromISR`. If you're call `spi_device_queue_trans` from peripheral interrupt it's always crash.

You must update `spi_master.c` to make it's work.

Replace
```c
BaseType_t r = xQueueSend(handle->trans_queue, (void *)&trans_buf, ticks_to_wait);
```
By
```c
BaseType_t task_woken = pdFALSE;
BaseType_t r = xQueueSendFromISR(handle->trans_queue, (void *)&trans_buf, &task_woken);
portYIELD_FROM_ISR(task_woken);
```

mmar22, 2024-01-31:
I know this, but i cant uderstand why in 21 century somebody writes library not prepared for ISR use? Yes your edit maybe helps , but 100% not if i require use other SPI outside ISR.
I accept this for new lib , but IDF is how 5 year ???