espressif/esp-idf #12856: WIFI MESH false cyclic found

mpierasco, 2023-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am developing a project without a router, self organization is off and all devices use fixed root configuration.
My issue is the following: I start the device A, it become the root, then power device B which connect to A, the I power off the device A, the device B get the PARENT_DISCONNECTED event and later it become the root. Once I power on again the device A it is not able to connect to device B because it see a cyclic connection and deauth the node A.
Looks like the node B keep thinking that node A is still his parent also if it is the root, it is not connected to it.
I tried calling esp_mesh_disconnect and also esp_wifi_disconnect but none worked.
Is there any API to clean the parent status of the root?
 
On the device B, once it become root and node A power on again i see this:

<pre>I (31004) mesh: [IO]disable self-organizing&lt;reconnect&gt;
V (31004) esp_adapter: thread sem get: sem=0x3ffec4dc
V (31004) esp_adapter: thread sem get: sem=0x3ffec4dc
V (31014) esp_adapter: thread sem get: sem=0x3ffec4dc
I (31014) mesh: [MANUAL]designated as root and router is not set
<font color="#4E9A06">I (31014) iotty mesh - : No other nodes</font>
<font color="#4E9A06">I (31014) iotty mesh - : -----------------  I am ROOT   ------------------</font>
<font color="#CC0000">E (31014) iotty mesh - : Mesh Type: 1, Devices: 1</font>

V (35914) esp_adapter: thread sem get: sem=0x3ffec4dc
V (65914) esp_adapter: thread sem get: sem=0x3ffec4dc
V (95914) esp_adapter: thread sem get: sem=0x3ffec4dc
V (125914) esp_adapter: thread sem get: sem=0x3ffec4dc
D (130434) wifi:ap recv deauth/disassoc from sta ff:ff:ff:ff:ff:ff, reason:8
D (134164) wifi:ap recv deauth/disassoc from sta ff:ff:ff:ff:ff:ff, reason:3
D (134174) wifi:ap recv assoc/reassoc request
I (134174) wifi:new:&lt;11,2&gt;, old:&lt;11,0&gt;, ap:&lt;11,2&gt;, sta:&lt;11,2&gt;, prof:11
V (134174) wifi:ht40d freq=2452, chan=11
D (134174) wifi:bss=0x3ffbc470, resp=16 join
I (134174) wifi:station: 24:4c:ab:32:b9:b4 join, AID=1, bgn, 40D
D (134174) wifi:join success, ap send assoc response
D (134174) event: running post WIFI_MESH_EVENT:37 with handler 0x40177388 and context 0x3ffdf84c on loop 0x3ffb7154
<font color="#C4A000">0x40177388: mesh_wifi_event_cb at ??:?</font>

I (134174) mesh: [CYCLIC]cyclic found, deauth aid:1, 24:4c:ab:32:b9:b4
V (134174) esp_adapter: thread sem get: sem=0x3ffec4dc
I (134174) wifi:station: 24:4c:ab:32:b9:b4 leave, AID = 1, bss_flags is 658530, bss:0x3ffbc470
I (134184) wifi:new:&lt;11,0&gt;, old:&lt;11,2&gt;, ap:&lt;11,2&gt;, sta:&lt;11,2&gt;, prof:11
V (134184) wifi:ht20 freq=2462, chan=11
V (134184) esp_adapter: thread sem get: sem=0x3ffec4dc
V (134184) esp_adapter: thread sem get: sem=0x3ffec4dc
D (134184) wifi:no bcast bss, send deauth: reason not authed(6)
D (134184) event: running post WIFI_EVENT:15 with handler 0x400e1274 and context 0x3ffddbf8 on loop 0x3ffb7154
<font color="#C4A000">0x400e1274: ip_event_mesh_handler at /home/enventive/esp/dimmer_mesh/main/iotty_mesh.c:934</font>

<font color="#4E9A06">I (134184) iotty mesh - : WIFI Event id:15</font>
D (134184) event: running post WIFI_EVENT:15 with handler 0x40177388 and context 0x3ffd61dc on loop 0x3ffb7154
<font color="#C4A000">0x40177388: mesh_wifi_event_cb at ??:?</font>

D (134184) event: running post MESH_EVENT:25 with handler 0x400e2280 and context 0x3ffdec5c on loop 0x3ffb7154
<font color="#C4A000">0x400e2280: mesh_event_handler at /home/enventive/esp/dimmer_mesh/main/iotty_mesh.c:603</font>

V (134184) esp_adapter: thread sem get: sem=0x3ffec4dc
<font color="#4E9A06">I (134184) iotty mesh - : &lt;MESH_EVENT_PS_CHILD_DUTY&gt;cidx:0, 24:4c:ab:32:b9:b4, duty:0</font>
</pre>

swaroopsuthar, 2024-01-09:
Try this

```
    if ( is_root_device ) {
        printf("SET ROOT NODE\n");
        ESP_ERROR_CHECK(esp_mesh_set_type(MESH_ROOT));
        esp_mesh_set_self_organized(false, false);
    } else {
        printf("SET CHILD NODE\n");
        ESP_ERROR_CHECK(esp_mesh_set_type(MESH_IDLE));
        esp_mesh_fix_root(true);
        esp_mesh_set_self_organized(true, true);
    }
  ```

mpierasco, 2024-01-09:
> Try this
>
> ```
>     if ( is_root_device ) {
>         printf("SET ROOT NODE\n");
>         ESP_ERROR_CHECK(esp_mesh_set_type(MESH_ROOT));
>         esp_mesh_set_self_organized(false, false);
>     } else [...]

Hi @swaroopsuthar, thank you for the reply, I do not get your point, in which situation should I try that code? 
When I power off the root on the child side the MESH_EVENT_PARENT_DISCONNECTED event is triggered, then the child starts a scan, if it does not find any devices and it become root and I call esp_mesh_set_type(MESH_ROOT). 
esp_mesh_set_self_organized(false, false) is called at parent disconnect event, and esp_mesh_set_self_organized(true, false) is called when a node connect to a parent. In my application I do not have a router so last parameter is always false for me, I call esp_mesh_fix_root(true) at the beginning and I am not sure of the outcome of calling esp_mesh_set_self_organized(true, true) once the device found a new parent candidate.

andy-danieal, 2024-01-09:
```
  printf("SET ROOT NODE\n");
  ESP_ERROR_CHECK(esp_mesh_set_type(MESH_ROOT));
  esp_mesh_set_self_organized(false, false);
```

Add this code on *Root device* before `esp_mesh_start();`. That means we have fixed root and stop auto scanning of parent availablility.

IF you want without ROUTER then comment that code only not require any other changes.
```
     /* router */
    cfg.channel = CONFIG_MESH_CHANNEL;
    // cfg.router.ssid_len = strlen(CONFIG_MESH_ROUTER_SSID);
    // memcpy((uint8_t *) &cfg.router.ssid, CONFIG_MESH_ROUTER_SSID, cfg.router.ssid_len);
    // memcpy((uint8_t *) &cfg.router.password, CONFIG_MESH_ROUTER_PASSWORD,
    //        strlen(CONFIG_MESH_ROUTER_PASSWORD));
    /* mesh softAP */
    ESP_ERROR_CHECK(esp_mesh_set_ap_authmode(CONFIG_MESH_AP_MODE));
    cfg.mesh_ap.max_connection         = CONFIG_MESH_AP_DEVICES;
    cfg.mesh_ap.nonmesh_max_connection = CONFIG_MESH_NON_AP_DEVICES;
    memcpy((uint8_t *)&cfg.mesh_ap.password, CONFIG_MESH_NON_AP_PASSWORD, strlen(CONFIG_MESH_NON_AP_PASSWORD));
    ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));
```

```
  printf("SET CHILD NODE\n");
  ESP_ERROR_CHECK(esp_mesh_set_type(MESH_IDLE));
  esp_mesh_fix_root(true);
  esp_mesh_set_self_organized(true, true);
```

Add this code on *Child device* before `esp_mesh_start();`. That means we have fixed root and start auto scanning of parent availablility. and without need manual scanning.

mpierasco, 2024-01-09:
Hi @andy-danieal, my issue came from root node power off and its child becoming the root, so I cannot call those functions before esp_mesh_start(), I have three steps, an init function where I call initialization functions like

esp_mesh_set_topology(MESH_TOPO_TREE)
esp_mesh_fix_root(true)
esp_mesh_set_config(&cfg)
esp_mesh_start()

then at the MESH_EVENT_STARTED I start the scanning for already active nodes, so I have 
```
	case MESH_EVENT_STARTED: 
		esp_mesh_get_id(&id);
		ESP_LOGI(TAG, "<MESH_EVENT_MESH_STARTED>ID:"MACSTR"", MAC2STR(id.addr));
		mesh.layer = esp_mesh_get_layer();
		ESP_ERROR_CHECK(esp_mesh_set_self_organized(0, 0));
		esp_wifi_scan_stop();
		scan_config.show_hidden = 1;
		scan_config.scan_type = WIFI_SCAN_TYPE_PASSIVE;
		scan_config.channel = DEFAULT_AP_CHANNEL;
		esp_wifi_scan_start(&scan_config, 0);
		mesh.connected = false;
```
After the scan phase I call the scan_done_handler where I decide if the device must be root or should be a child and connect to someone.
This is usually working, if I power off, let say, the first two layer then the devices on third layer elect a new root and rebuild the network, then if I power on again the devices which were in the first two layer they will connect to the network without issue.
My issue happen when a device become root and it remembers, for some reason, which device was its parent and if this device tries to become the child of the new root a cyclic will be detected. If the old root power on and connect to another device I do not get the cyclic issue.

mpierasco, 2024-01-23:
> @mpierasco can you provide a demo for me to test?

Hi @zhangyanjiaoesp, yes I can provide it, I just need some time to strip off all the non relevant code and the hw related code so it will run on every ESP32 module

zhangyanjiaoesp, 2024-01-26:
@mpierasco Do you use ESP32? Which IDF commit do you use?

mpierasco, 2024-01-26:
Hi @zhanzhaocheng I use ESP32, I am currently using ESP-IDF v4.4,I tried this firmware also with ESP-IDF v5.0 but nothing changed

zhangyanjiaoesp, 2024-01-26:
@mpierasco please tell me the commit id info, like this bb8dd9d35b4961a8a7181cd856a41e79e436081e

mpierasco, 2024-01-26:
This is what I get from git: commit 8153bfe4125e6a608abccf1561fd10285016c90a (HEAD, tag: v4.4)

mahboobkarimian, 2024-02-02:
The same problem just posted here: https://esp32.com/viewtopic.php?f=13&t=38133
My problem is that after the root node reboots, its child node does not get PARENT_DISCONECTED event. Then it never tries to reconnect. Part of this issue is that there is no authentication event for STA which can handle very fast AP reboots. I'm using commit `e49823f10c6b6b8dadbad376002ea2bc531b7b0f / release/v5.2`

mahboobkarimian, 2024-02-02:
Here is more detail:
```
D (456238) wifi:bssid equal: ss_state=0x4
[0;32mI (457058) mesh_main: Received Routing table [0] 40:4c:ca:41:6d:dc[0m
[0;32mI (457058) mesh_main: Received Routing table [1] 40:4c:ca:41:b7:90[0m
V (457138) transport_base: poll_read: select - Timeout before any socket was ready![0m
V (457138) transport_base: poll_read: select - Timeout before any socket was ready![0m
D (457138) mqtt_client: esp_mqtt_handle_transport_read_error: transport_read(): call timed out before data was ready![0m
[0;33mW (457148) user_tasks: Sensing with DS18B20 failed[0m
[0;32mI (457158) user_tasks: Temperature: -999.0C[0m
D (457258) wifi:mms: 0->0
D (457258) wifi:bssid equal: ss_state=0x4
[0;32mI (458008) mesh_main: Tried to publish layer:2 IP:10.0.0.2[0m
D (458008) mqtt_client: mqtt_enqueue id: 46576, type=3 successful[0m
D (458008) outbox: ENQUEUE msgid=46576, msg_type=3, len=39, size=39[0m
D (458008) mesh_netif: Sending to root, dest addr: 40:4c:ca:41:6d:dd, size: 122[0m
[0;32mI (458128) user_tasks: systemtime is 458148106[0m
V (458158) transport_base: poll_read: select - Timeout before any socket was ready![0m
[0;33mW (458158) user_tasks: Sensing with DS18B20 failed[0m
[0;32mI (458158) user_tasks: Temperature: -999.0C[0m
[0;33mW (459158) user_tasks: Sensing with DS18B20 failed[0m
[0;32mI (459158) user_tasks: Temperature: -999.0C[0m
W (459218) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:485, no_wnd_count:0, timeout_count:0
V (459258) esp_adapter: thread sem get: sem=0x40830f34[0m
[0;32mI (460128) user_tasks: systemtime is 460148100[0m
[0;33mW (460158) user_tasks: Sensing with DS18B20 failed[0m
[0;32mI (460158) user_tasks: Temperature: -999.0C[0m
W (460418) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:485, no_wnd_count:0, timeout_count:1
[0;33mW (461158) user_tasks: Sensing with DS18B20 failed[0m
[0;32mI (461158) user_tasks: Temperature: -999.0C[0m
W (461618) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:485, no_wnd_count:0, timeout_count:2
[0;32mI (462128) user_tasks: systemtime is 462148100[0m
[0;33mW (462158) user_tasks: Sensing with DS18B20 failed[0m
[0;32mI (462158) user_tasks: Temperature: -999.0C[0m
W (462818) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:485, no_wnd_count:0, timeout_count:3
[0;33mW (463158) user_tasks: Sensing with DS18B20 failed[0m
[0;32mI (463158) user_tasks: Temperature: -999.0C[0m
W (464018) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:485, no_wnd_count:0, timeout_count:4
[0;31mE (464018) mesh_netif: Send with err code 16394 ESP_ERR_MESH_TIMEOUT[0m
[0;32mI (464018) mesh_mqtt: sent publish returned msg_id=46576[0m
V (464028) transport_base: poll_read: select - Timeout before any socket was ready![0m
D (464038) mqtt_client: esp_mqtt_handle_transport_read_error: transport_read(): call timed out before data was ready![0m
[0;32mI (464128) user_tasks: systemtime is 464148105[0m
[0;33mW (464158) user_tasks: Sensing with DS18B20 failed[0m
[0;32mI (464158) user_tasks: Temperature: -999.0C[0m
D (464768) mesh_netif: Sending to root, dest addr: 40:4c:ca:41:6d:dd, size: 122[0m
V (465058) transport_base: poll_read: select - Timeout before any socket was ready![0m
V (465058) transport_base: poll_read: select - Timeout before any socket was ready![0m
D (465058) mqtt_client: esp_mqtt_handle_transport_read_error: transport_read(): call timed out before data was ready![0m
D (465068) mqtt_client: Sending Duplicated QoS1 message with id=46576[0m
[0;33mW (465158) user_tasks: Sensing with DS18B20 failed[0m
[0;32mI (465158) user_tasks: Temperature: -999.0C[0m
I (465928) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
D (465928) wifi:mms: 0->0
D (465928) wifi:bssid equal: ss_state=0x4
D (465928) event: running post WIFI_MESH_EVENT:54 with handler 0x4207ab2a and context 0x40830b78 on loop 0x40825550[0m
I (465938) mesh: <MESH_NWK_ROUTER>new router bssid:00:00:00:00:00:00
V (465948) esp_adapter: thread sem get: sem=0x40830f34[0m
W (465968) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:2, 1200 ms timeout, seqno:0, xseqno:486, no_wnd_count:0, timeout_count:5
D (466018) wifi:mms: 0->0
D (466018) wifi:bssid equal: ss_state=0x4
[0;32mI (466028) mesh_main: Tried to publish layer:2 IP:10.0.0.2[0m
[0;32mI (466128) user_tasks: systemtime is 466148106[0m
```

Note that I rebooted the root at time t=457285.
Look at this line a little after the root reboot at t=465928
```
D (465928) event: running post WIFI_MESH_EVENT:54 with handler 0x4207ab2a and context 0x40830b78 on loop 0x40825550[0m
I (465938) mesh: <MESH_NWK_ROUTER>new router bssid:00:00:00:00:00:00
[... 40 lines cut ...]
D (682388) wifi:stop CSA timer
I (682388) wifi:<ba-del>idx:0, tid:5
I (682398) wifi:new:<6,0>, old:<6,1>, ap:<6,1>, sta:<6,0>, prof:6
V (682398) wifi:ht20 freq=2437, chan=6
D (682408) wifi:filter: set rx policy=8
D (682408) wifi:Send disconnect event, reason=209, AP number=1
D (682418) event: running post WIFI_EVENT:5 with handler 0x4207ab2a and context 0x40830fd8 on loop 0x40825550[0m
V (682428) esp_adapter: thread sem get: sem=0x40830f34[0m
V (682428) esp_adapter: thread sem get: sem=0x40830f34[0m
I (682438) mesh: 5961<scan>parent layer:1, rssi:0, assoc:0(cnx rssi threshold:-120)
[... cut ...]
I (682488) mesh: [wifi]disconnected reason:209(), continuous:1/max:12, non-root, vote(,stopped)<><>
I (682498) mesh: [scan]new scanning time:600ms, beacon interval:300ms
V (682508) esp_adapter: thread sem get: sem=0x40830f34[0m
D (682508) nvs: nvs_set bcn.interval 2 300[0m
V (682528) esp_adapter: thread sem get: sem=0x40830f34[0m
V (682528) esp_adapter: thread sem get: sem=0x40830f34[0m
```
What is the solution for this? This is the very basic scenario in every mesh network, it must be handled in someway.
Please note that in some instances, even after  hours my node can not recover!

zhangyanjiaoesp, 2024-02-06:
@mahboobkarimian Has reply you on the forum https://esp32.com/viewtopic.php?f=13&t=38133

yanmartins, 2024-03-08:
I was using esp-idf 5.0.4, after update to 5.2.1 the nodes start to show this message too:
`Send with err code 16394 ESP_ERR_MESH_TIMEOUT`

zhangyanjiaoesp, 2024-05-10:
@mpierasco do you still have this issue?

mpierasco, 2024-05-13:
> @mpierasco do you still have this issue?

Yes, the last replies refer to a scenario with a router so they do not apply to my issue, did you try the demo I sent? Was you able to reproduce my issue?

zhangyanjiaoesp, 2024-05-13:
@mpierasco I have tried with your demo on IDF v4.4, but the device can't run correctly, it always show the following log.
[image]

mpierasco, 2024-05-13:
[demo_mesh_test.zip](https://github.com/espressif/esp-idf/files/15291112/demo_mesh_test.zip)
 I have just redone the project starting from the Hello World sample project, I did not touch the config file, I also removed the certificate and I tried on an standard evaluation board programming it through USB and looking at the monitor through the same USB.
I used ESP-IDF v4.4 to build it.
This is what I got from monitor

> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0030,len:6612
> load:0x40078000,len:14780
> load:0x40080400,len:3792
> 0x40080400: _init at ??:?
> 
> entry 0x40080694
> I (27) boot: ESP-IDF v4.4-dirty 2nd stage bootloader
> I (27) boot: compile time 09:44:57
> I (27) boot: chip revision: 1
> I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
> I (37) boot.esp32: SPI Speed      : 40MHz
> I (42) boot.esp32: SPI Mode       : DIO
> I (46) boot.esp32: SPI Flash Size : 2MB
> I (51) boot: Enabling RNG early entropy source...
> I (56) boot: Partition Table:
> I (60) boot: ## Label            Usage          Type ST Offset   Length
> I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (82) boot:  2 factory          factory app      00 00 00010000 00100000
> I (89) boot: End of partition table
> I (94) boot_comm: chip revision: 1, min. application chip revision: 0
> I (101) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1dd3ch (122172) map
> I (154) esp_image: segment 1: paddr=0002dd64 vaddr=3ffb0000 size=022b4h (  8884) load
> I (157) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=9e0d0h (647376) map
> I (394) esp_image: segment 3: paddr=000ce0f8 vaddr=3ffb22b4 size=01640h (  5696) load
> I (396) esp_image: segment 4: paddr=000cf740 vaddr=40080000 size=142f4h ( 82676) load
> I (433) esp_image: segment 5: paddr=000e3a3c vaddr=50000000 size=00010h (    16) load
> I (443) boot: Loaded app from partition at offset 0x10000
> I (443) boot: Disabling RNG early entropy source...
> I (456) cpu_start: Pro cpu up.
> I (456) cpu_start: Starting app cpu, entry point is 0x40081174
> 0x40081174: call_start_cpu1 at /home/enventive/esp/esp-idf/components/esp_system/port/cpu_start.c:156
> 
> I (0) cpu_start: App cpu up.
> I (472) cpu_start: Pro cpu start user code
> I (472) cpu_start: cpu freq: 160000000
> I (472) cpu_start: Application information:
> I (477) cpu_start: Project name:     hello_world
> I (482) cpu_start: App version:      1
> I (486) cpu_start: Compile time:     May 13 2024 09:44:52
> I (492) cpu_start: ELF file SHA256:  45c3a9ebd017a3e2...
> I (498) cpu_start: ESP-IDF:          v4.4-dirty
> I (504) heap_init: Initializing. RAM available for dynamic allocation:
> I (511) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
> I (517) heap_init: At 3FFB8808 len 000277F8 (157 KiB): DRAM
> I (523) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
> I (530) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
> I (536) heap_init: At 400942F4 len 0000BD0C (47 KiB): IRAM
> I (543) spi_flash: detected chip: gd
> I (546) spi_flash: flash io: dio
> W (550) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
> I (565) cpu_start: Starting scheduler on PRO CPU.
> I (0) cpu_start: Starting scheduler on APP CPU.
> I (645) wifi:wifi driver task: 3ffc19a4, prio:23, stack:6656, core=0
> I (645) system_api: Base MAC address is not set
> I (645) system_api: read default base MAC address from EFUSE
> I (655) wifi:wifi firmware version: 7679c42
> I (655) wifi:wifi certification version: v7.0
> I (655) wifi:config NVS flash: enabled
> I (655) wifi:config nano formating: disabled
> I (665) wifi:Init data frame dynamic rx buffer num: 32
[... 102 lines cut ...]
> I (4115) iotty mesh - : <MESH_EVENT_SCAN_DONE>number:4
> W (4115) iotty mesh - : My type: 0
> I (4125) iotty mesh - : [0]IOTTY_176709547626952, a0:b7:65:d0:91:c9, channel:11, rssi:-75
> I (4135) iotty mesh - : [1]IOTTY_211423786030988, c0:49:ef:a9:d7:8d, channel:11, rssi:-72
> I (4145) iotty mesh - : [2], 32:de:4b:e2:66:e5, channel:11, rssi:-83
> I (4145) iotty mesh - : [3]Iotty_WiFi, d8:0d:17:ff:5f:56, channel:11, rssi:-77
> I (4515) iotty mesh - : WIFI Event id:1
> E (4515) iotty mesh - : Mesh Type: 0, Devices: 1
> 
> 
> I (4515) mesh: 4016<scan>self-organized is disabled, users shall read out scan results by themselves.
> I (4525) iotty mesh - : <MESH_EVENT_SCAN_DONE>number:4
> W (4535) iotty mesh - : My type: 0
> I (4535) iotty mesh - : [0]IOTTY_176709547626952, a0:b7:65:d0:91:c9, channel:11, rssi:-74
> I (4545) iotty mesh - : [1], 32:de:4b:e2:66:e5, channel:11, rssi:-85
> I (4555) iotty mesh - : [2]IOTTY_211423786030988, c0:49:ef:a9:d7:8d, channel:11, rssi:-72
> I (4565) iotty mesh - : [3]Iotty_WiFi, d8:0d:17:ff:5f:56, channel:11, rssi:-77
> W (4565) iotty mesh - : try_count > 5
> W (4575) iotty mesh - : scan count: 0, root candidate: 1
> I (4575) mesh: [CONFIG]set parent->sta.ssid to be a valid value
> I (4585) mesh: [IO]disable self-organizing<reconnect>
> I (4585) mesh: [MANUAL]designated as root and router is not set
> I (4595) iotty mesh - : No other nodes
> I (4595) iotty mesh - : -----------------  I am ROOT   ------------------
> E (4605) iotty mesh - : Mesh Type: 1, Devices: 1

zhangyanjiaoesp, 2024-05-13:
I will test the new demo, thanks.

zhangyanjiaoesp, 2024-05-13:
@mpierasco Now I can reproduce the problem using the new demo, will debug it and find the root cause ASAP.

zhangyanjiaoesp, 2024-05-15:
@mpierasco 
I have changed your test code, and now it seems to work well. The changes are the following:
(the left is the original code, the right is the modified code)
[image]
[image]
[image]

 In the original code, `esp_mesh_set_parent()` returns error due to the parent is zero, you should set a valid value for the parent, then node's parent will be updated. At the same time, `esp_mesh_set_self_organized(0,0)` to stop the root reconnect.

By the way, I think the function calls for `esp_mesh_set_self_organized(1,0)` in these two places are redundant.
[image]
[image]

zhangyanjiaoesp, 2024-05-20:
> Did the demo worked for you after your changes?
yes, it works for me.

zhangyanjiaoesp, 2024-05-20:
> @zhangyanjiaoesp By the way is there any way to clear parent info set by esp_mesh_set_parent() ? Is it possible to set to no parent as at beginning? I think this could solve my issue Thank you

In our design, when the mesh network is set to root-fixed, then the layer2 node won't change it's parent, so if the root node leaves, the layer2 node will scan or reconnect to the root node until it returns. 
The `esp_mesh_set_parent()` API can't set to NULL to clear the parent information, so I set a valid router value for it when the node change to ROOT node. This will clear the parent information which is set when the node is layer2 node. And the `esp_mesh_set_self_organized(0,0)` will stop the root reconnect.

zhangyanjiaoesp, 2024-05-20:
@mpierasco We will discuss internally if can set `parent` to NULL for the `esp_mesh_set_parent()` API.

mpierasco, 2024-05-20:
@zhangyanjiaoesp, Thank you for your help.

I made other tests, I assume that you say that the demo works because once the old root starts again there is a scan session then it become root again and the other node connect to it, but this is not a properly working scenario in my opinion.
Once the original root disappear on the other note I can see a scan session then this:

<pre><font color="#C4A000">W (931981) iotty mesh - : try_count &gt; 5</font>
<font color="#C4A000">W (931981) iotty mesh - : scan count: 0, root candidate: 1</font>
I (931991) mesh: [IO]disable self-organizing&lt;reconnect&gt;
I (932001) mesh: [MANUAL]connect to router:iotty_mesh, 00:00:00:00:00:00[layer:0], ID:64:65:6d:6f:5f:6d&lt;connect&gt;
<font color="#4E9A06">I (932011) iotty mesh - : esp_mesh_set_parent return value: 0</font>
I (932011) mesh: [IO]disable self-organizing&lt;reconnect&gt;
I (932051) mesh: [CONFIG]connect to router:iotty_mesh, 00:00:00:00:00:00
I (932081) mesh: &lt;mesh_connect_to_router,540&gt;parent is set&lt;stop reconnect&gt;g_is_wifi_connecting:1, g_is_wifi_disconnecting:0, g_mesh_stop_reconnection:0
<font color="#4E9A06">I (932081) iotty mesh - : No other nodes</font>
<font color="#4E9A06">I (932081) iotty mesh - : -----------------  I am ROOT   ------------------</font>
I (932091) mesh: [IO]disable self-organizing&lt;stop reconnect&gt;
<font color="#CC0000">E (932101) iotty mesh - : Mesh Type: 0, Devices: 1</font>
</pre>

What is not right, in my opinion, is that the node take the decision to become the root, it claims this role for itself, but the mesh type return to 0, this means that other node connected to it would disconnect because it says it is outside the mesh and no one would connect to him as root because it says it is not a root.
This is the code:

```
		MON_LOGW(TAG, "try_count > 5");
		MON_LOGW(TAG, "scan count: %d, root candidate: %d", scan_next_entry, root_candidate);			
		if( /*(scan_next_entry == 0) ||*/ (root_candidate)) {
		        wifi_config_t parent = {
			        .sta = {
				        .ssid = DEFAULT_SSID,
				        .password = DEFAULT_PASSWORD,
				        .channel = DEFAULT_AP_CHANNEL
			        }
		        };
		        esp_err_t err = esp_mesh_set_parent(&parent, NULL, MESH_ROOT, MESH_ROOT_LAYER);
		        MON_LOGI(TAG, "esp_mesh_set_parent return value: %d", err);
			if(err_check(TAG, esp_mesh_set_type(MESH_ROOT), 101)) {
				MON_LOGI(TAG, "No other nodes");
				MON_LOGI(TAG, "-----------------  I am ROOT   ------------------");
				esp_mesh_set_self_organized(0, 0);
				esp_mesh_comm_start();
				mesh_scan_entries_clean();
				MON_LOGE(TAG, "Mesh Type: %d, Devices: %d\n\n", esp_mesh_get_type(), esp_mesh_get_routing_table_size());
			}
		}

```
Is the reason that after calling **esp_mesh_set_self_organized(0, 0);** the mesh type is reset to default state? Should I call it before setting the type to root?
Sorry for all these questions but I do not understand which is the expected flow in the library.

zhangyanjiaoesp, 2024-05-21:
@mpierasco I will check the flow again.

zhangyanjiaoesp, 2024-05-21:
@mpierasco Please add `esp_mesh_disconnect()` after the `esp_mesh_set_parent()` API, and have a try.
This is the code:
[image]

When the root node call `esp_mesh_set_parent()` API, it will enable reconnect and call `esp_wifi_connect()` to connect to the router, but your case don't need connect to a router, so we call `esp_mesh_disconnect()` to disconnect. 
The `esp_mesh_set_type()` API will also enable reconnect when type is root, so we call `esp_mesh_set_self_organized(0,0)` to stop reconnect.
The mesh type will change when the node is connecting, is not caused by `esp_mesh_set_self_organized(0,0)`  .

mpierasco, 2024-05-21:
@zhangyanjiaoesp 
It still does not work, if I set the parent the device try to connect, I call esp_mesh_disconnect, then esp_mesh_set_type(MESH_ROOT) but the type does not change to 1, if I call esp_mesh_is_root() it returns false.
If I avoid to set the parent and call the disconnect function then the type become 1 after esp_mesh_set_type(MESH_ROOT).
How am I supposed to elect a root node if there is no router? It looks like the node can become a root only if it has no parent or if it can connect to it, or was able to connect to it in the past, as was my initial issue.

zhangyanjiaoesp, 2024-05-21:
> In our design, when the mesh network is set to root-fixed, then the layer2 node won't change it's parent, so if the root node leaves, the layer2 node will scan or reconnect to the root node until it returns.

@mpierasco This is the root cause, you want to change the layer2 node in the fix-root network to make it root, which is against our design.  Why you want to change fixed root between two device? How about use self-organized network?

mpierasco, 2024-05-21:
@zhangyanjiaoesp The point is that in the real environment there will be 30 or more devices, there is no guarantee that the root node will always be available and if not the mesh will be split in the sub mesh isolated, so the nodes in one sub mesh will not be able to send message to the nodes in the other sub mesh.
I would use self-organized network but it works only with a router because the metric used to vote nodes to become root is based on the router signal seen by each node, without a router it doesn't work, or this is what I understand.

Sherry616, 2024-07-17:
Hi @mpierasco, could you please share your latest updates on this issue? Thanks.

Sherry616, 2024-07-31:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. 
Thanks for using our Espressif product!

closed 2024-07-31 by Sherry616

avehbiaslanturk, 2024-10-11:
Hi mates,

I have been struggling with the same problem for a fixed root application. As you said, when I reset the root, the leaves can't detect the parent disconnected event and go into a silly loop. I tried adding a 10-second delay at the beginning of the root's code so that the clients can detect the parent disconnected event, and it works! :)

[image]

[image]

Ahmet Vehbi,

zhangyanjiaoesp, 2024-10-12:
@avehbiaslanturk this issue has been solved in [85c4c96e](https://github.com/espressif/esp-idf/commit/85c4c96e72a53a9c0fb1a25f36caa9f990897a2f), and the solution has been backported to v5.3/5.2/5.1/5.0, please update your IDF version, thanks.