espressif/esp-idf #12857: Wifi to Serial Tunnel Example

seankovacs, 2023-12-21:
### Is your feature request related to a problem?

No

### Describe the solution you'd like.

It would be nice to see an example of a native ESP-IDF Wifi to Serial tunnel. This is a very common use-case handled by uBlox and Lantronix chips out the box. 

### Describe alternatives you've considered.

Many Arduino based wifi to serial bridges, but nothing I've seen using just ESP-IDF.

MaxwellAlan, 2023-12-22:
Hi @seankovacs 

Could you provide some more details about this feature request? Such as application scenario or some other reference examples?
PS: we also have [esp-hosted solution](https://github.com/espressif/esp-hosted), does it meet your request? 

Thanks.

seankovacs, 2023-12-22:
The esp-hosted solutions are a step in the right direction, but seem like overkill. The esp-hosted solutions don't support Wifi to Uart either. The idea is to simply connect to an STM32 over Wifi (STA or AP) via UDP or TCP, with the ESP32 interfacing to the STM32 using a UART. In particular, I'm passing [mavlink ](https://mavlink.io/en/) packets between a host computer and an STM32 MCU.

MaxwellAlan, 2023-12-22:
@seankovacs how about ESP-AT(https://github.com/espressif/esp-at) solution?

seankovacs, 2023-12-22:
This seems to do what I am asking, though is it open source? I  need to do man-in-the-middle packet inspection on the ESP.

MaxwellAlan, 2023-12-25:
@seankovacs

Yeah, most of the AT solution code is open. Maybe you can submit your detail request in the [AT issue](https://github.com/espressif/esp-at/issues)

MaxwellAlan, 2025-03-06:
This feat can be done in APP layer, will close this feat request.
Thanks.

closed 2025-03-06 by MaxwellAlan