espressif/esp-idf #12862: Sharing Digital Signature Peripheral Across Threads

xieliwei, 2023-12-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, I'm working on a firmware that requires TLS device authentication with the secure key+cert stored in the DS of an ESP32-S3. Thus we have ESP-TLS being using by CoreMQTT for sensor reports, and ESP-TLS being used by esp_websocket_client for a separate configuration and command channel. Both the MQTT and websocket are authenticated and secure channel formed using the certificate stored on the DS.

While experiencing network issues that require re-connections on MQTT and websocket simultaneously, we regularly hit the following assert failure (this also sometimes happen on boot, probably due to varying server and network delays, despite us trying to stagger the initialisation of MQTT and websocket tasks):

`assert failed: xQueueGenericSend IDF\components\freertos\queue.c:832 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTas`

And the backtrace always goes like this (I'm using PlatformIO, with IDF 4.4.6):

`components/esp_system/panic.c:408
components/esp_system/esp_system.c:137
components\newlib/assert.c:85
components\freertos/queue.c:830 (discriminator 8)
components\mbedtls\port\esp_ds/esp_rsa_sign_alt.c:108
components\esp-tls/esp_tls_mbedtls.c:317
components\esp-tls/esp_tls_mbedtls.c:255
components\esp-tls/esp_tls.c:107
..........`

With the origin of the backtrace either from esp_websocket_client or coreMQTT. The backtrace differs depending on what is being done, but they always reach the esp_rsa_sign_alt.c:108 point.

From what I understand, this is because both coreMQTT and esp_websocket_client are sharing the DS peripheral without proper coordination across different threads/tasks. While it should be easily solved using a mutex for operations involving the DS peripheral, it is not easy to do so as re-connection is handled by the coreMQTT and esp_websocket_client components.

I think I'm doing this wrongly, since the assert is coming from code that is precisely trying to gain (or release) exclusive access to the DS peripheral, so shouldn't it already be safe to try to access it from different threads? Can somebody advise me?

xieliwei, 2023-12-23:
I think I found the issue. It seems like this is occurring due to a `esp_ds_release_ds_lock()` (and hence `xSemaphoreGive()`) that is always called in `esp_mbedtls_cleanup()`, regardless of whether a lock was previously acquired. 

`esp_mbedtls_cleanup()` is called whenever a TLS connection attempt fails. Thus if the TLS connection fails before `esp_ds_init_data_ctx()` (which contains the `xSemaphoreTake()`) is called, we will be calling `xSemaphoreGive()` without prior locking.

This occurs for example when there is any issue with the certificate provided when building the TLS context, or there is a problem establishing the underlying TCP connection. There are more cases when this occurs for sure, as there are many points where the TLS connection can fail. This also applies when we are setting up a TLS server connection as well.

This is already a bad thing, but it seems like doing so does not cause a crash as `xSemaphoreGive()` will return a failure from the underlying queue.

The assert fails when we do successfully use the DS peripheral via ESP-TLS from another task prior to this and has yet to return the lock. Since the IDF defines `configCHECK_MUTEX_GIVEN_BY_OWNER`, we will see the assert mentioned in the issue fail as the failed TLS connection tries to call `xSemaphoreGive()` on a mutex that was previously successfully acquired (and not yet given) from another thread.

The following code snippet provides a minimal proof of this issue. It should demonstrate the issue even with an un-configured DS peripheral, as we just fake the required structures.

`CONFIG_ESP_TLS_USE_DS_PERIPHERAL=y` must be in sdkconfig
```
#include <esp_log.h>
#include <esp_wifi.h>
#include <esp_tls.h>
#include <esp_http_client.h>

void do_ds_task(void *p) {
    // We don't have to actually configure the DS peripheral in this POC
    // Just fake the structs so that esp_ds_init_data_ctx() will be happy
    esp_ds_data_t esp_ds_data = {
        .rsa_length = ESP_DS_RSA_2048,
    };
    esp_ds_data_ctx_t ds_data = {
        .esp_ds_data = &esp_ds_data,
        .rsa_length_bits = 2048,
    };
    esp_err_t esp_ret = esp_ds_init_data_ctx(&ds_data);
    if (esp_ret == ESP_OK) {
        ESP_LOGI( "DS_TASK", "DS Data Context Initialised" );
    }
    // Let's extend the bug's window by simulating a DS operation that
    // takes very long to complete
    vTaskDelay(pdMS_TO_TICKS(20000));
    // We will never reach here
    esp_ds_release_ds_lock();
    vTaskDelete(NULL);
}

void do_tls_fail(void *p) {
    vTaskDelay(pdMS_TO_TICKS(1000));
    ESP_LOGI( "TLS_FAIL", "This task will cause an assertion failure in 10 seconds" );
    vTaskDelay(pdMS_TO_TICKS(10000));
    esp_http_client_config_t config = {
        // We use an invalid domain to cause the TLS connection to fail
        .host = "fakedomain.faketld",
        .port = 12345,
        .path = "/",
        .transport_type = HTTP_TRANSPORT_OVER_SSL,
    };
    esp_http_client_handle_t client = esp_http_client_init(&config);
    // This will cause the assertion fail
    esp_http_client_perform(client);
    esp_http_client_cleanup(client);
    vTaskDelete(NULL);
}

void app_main() {
    esp_netif_init();
    esp_wifi_set_mode( WIFI_MODE_AP );
    xTaskCreate(do_ds_task, "T1", 4096, NULL, 1, NULL);
    xTaskCreate(do_tls_fail, "T2", 4096, NULL, 1, NULL);
    while (1) {
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}
```

Resolving this issue might be a bit tricky as we have to persist some information across different tasks. This likely means we have to modify the `esp_tls_t` handle to contain at least a boolean on whether a lock was previously acquired. Modifying `esp_tls_t` however could break compatibility.

Alternatively ESP-TLS can track the task handle that last acquired the lock and only calls `esp_ds_release_ds_lock()` if the currently running task matches it.

Ideally, the solution should tend towards using `esp_ds_release_ds_lock()` with more finesse in `esp_mbedtls_cleanup()`. Meanwhile, a simple solution is to patch `esp_ds_release_ds_lock()` to determine if the lock is held by the current task (using `xSemaphoreGetMutexHolder()`) before calling `xSemaphoreGive()`. The performance impact should be minimal.

mahavirj, 2024-01-04:
@xieliwei This issue was fixed through https://github.com/espressif/esp-idf/pull/11402. May I know which IDF release you are using?

xieliwei, 2024-01-04:
After checking again, it seems the project was using 4.4.5. Odd, this is a new PIO project that's created after 4.4.6 came out. I assumed it was already on the latest IDF supported by the Arduino framework. Since this was already resolved in the latest version, I'll close this issue. Thanks for looking into it!

closed 2024-01-04 by xieliwei