espressif/esp-idf #12864: ESP32C6 incorrect temp_sensor reading after disable and re-enable

AxelLin, 2023-12-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2-496-g3629f070b6

### Espressif SoC revision.

ESP32C6

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

temperature_sensor_get_celsius() should get correct temperature reading.

### What is the actual behavior?

If calling temperature_sensor_disable() and temperature_sensor_enable() again, the reading becomes wrong.

### Steps to reproduce.

A modified code of examples/peripherals/temperature_sensor as below shows the symptom.
Only the first reading shows correct tsens_value.
If calling temperature_sensor_disable() and temperature_sensor_enable() again, the reading becomes wrong.

void app_main(void)
{
    ESP_LOGI(TAG, "Install temperature sensor, expected temp ranger range: 10~50 ℃");
    temperature_sensor_handle_t temp_sensor = NULL;
    temperature_sensor_config_t temp_sensor_config = TEMPERATURE_SENSOR_CONFIG_DEFAULT(10, 50);
    ESP_ERROR_CHECK(temperature_sensor_install(&temp_sensor_config, &temp_sensor));

    //ESP_LOGI(TAG, "Enable temperature sensor");
    //ESP_ERROR_CHECK(temperature_sensor_enable(temp_sensor));

    ESP_LOGI(TAG, "Read temperature");
    int cnt = 20;
    float tsens_value;
    while (cnt--) {
        temperature_sensor_enable(temp_sensor);

        // The first reading after disable/enable is incorrect
        ESP_ERROR_CHECK(temperature_sensor_get_celsius(temp_sensor, &tsens_value));
        ESP_LOGI(TAG, "Temperature value %.02f ℃", tsens_value);
        vTaskDelay(pdMS_TO_TICKS(1000));

#if 0   // Note, the second reading is correct.
        ESP_ERROR_CHECK(temperature_sensor_get_celsius(temp_sensor, &tsens_value));
        ESP_LOGI(TAG, "Temperature value %.02f ℃", tsens_value);
        vTaskDelay(pdMS_TO_TICKS(1000));
#endif
        temperature_sensor_disable(temp_sensor);

    }
}

Seems only the first reading after disable/enable is incorrect, the second reading looks correct.
Change #if 0 to 1 in above code will show that the second reading is correct.

suda-morris, 2023-12-25:
@AxelLin Thanks for reporting. It looks like we need to add some delay after `temperature_sensor_enable`. We will look into this issue ASAP.

AxelLin, 2023-12-25:
I add the temperature sensor reading code to examples/bluetooth/nimble/bleprph.
It shows after disable-reenable the reading is wrong.
But **after bt init**, the reading becomes correct.

Below is the log, I also attached the modified code:

```
I (334) cpu_start: Min chip rev:     v0.0
I (339) cpu_start: Max chip rev:     v0.99
I (344) cpu_start: Chip rev:         v0.0
I (349) heap_init: Initializing. RAM available for dynamic allocation:
I (356) heap_init: At 408158D0 len 00066D40 (411 KiB): D/IRAM
I (362) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (369) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (376) spi_flash: detected chip: generic
I (380) spi_flash: flash io: dio
I (384) sleep: Configure to isolate all GPIO pins in sleep state
I (391) sleep: Enable automatic switching of GPIO sleep configuration
I (398) coexist: coex firmware version: e41c5cb
I (403) coexist: coexist rom version 5b8dcfa
I (408) app_start: Starting scheduler on CPU0
I (413) main_task: Started on CPU0
I (413) main_task: Calling app_main()
I (453) example: Install temperature sensor, expected temp ranger range: 10~50 ℃
I (453) temperature_sensor: Range [-10°C ~ 80°C], error < 1°C
I (463) example: Enable temperature sensor
I (463) example: Read temperature
I (463) example: Temperature value 12.00 ℃
I (1473) example: Temperature value 35.00 ℃
I (2473) example: Temperature value 35.00 ℃
I (3473) example: Temperature value 35.00 ℃
I (4473) example: Temperature value 35.00 ℃
I (5473) example: Temperature value 35.00 ℃
I (6473) example: Temperature value 35.00 ℃
I (7473) example: Temperature value 35.00 ℃
I (8473) example: Temperature value 35.00 ℃
I (9473) example: Temperature value 35.00 ℃
I (10473) BLE_INIT: Using main XTAL as clock source
controller lib commit: [0ecb36f]
I (10473) phy_init: phy_version 230,c773401,Oct 30 2023,15:07:16
I (10513) phy: libbtbb version: 7243671, Oct 30 2023, 15:07:30
I (10513) example: Temperature value 14.00 ℃
I (11513) example: Temperature value 14.00 ℃
I (12513) example: Temperature value 14.00 ℃
I (13513) example: Temperature value 14.00 ℃
I (14513) example: Temperature value 15.00 ℃
I (15513) example: Temperature value 15.00 ℃
I (16513) example: Temperature value 15.00 ℃
I (17513) example: Temperature value 15.00 ℃
I (18513) example: Temperature value 15.00 ℃
I (19513) example: Temperature value 15.00 ℃
I (20513) example: Temperature value 15.00 ℃
I (21513) example: Temperature value 15.00 ℃
I (22513) example: Temperature value 15.00 ℃
I (23513) example: Temperature value 15.00 ℃
I (24513) example: Temperature value 15.00 ℃
I (25513) example: Temperature value 15.00 ℃
I (26513) example: Temperature value 15.00 ℃
I (27513) example: Temperature value 15.00 ℃
I (28513) example: Temperature value 15.00 ℃
I (29513) example: Temperature value 15.00 ℃
I (30513) NimBLE_BLE_PRPH: BLE Host Task Started
I (30513) NimBLE: Device Address:
I (30513) NimBLE: 40:4c:ca:41:15:ce
I (30513) NimBLE:

I (30513) NimBLE: GAP procedure initiated: extended advertise; instance=0

I (30523) uart: queue free spaces: 8
I (30523) example: Temperature value 16.00 ℃
I (31533) example: Temperature value 15.00 ℃
I (32533) example: Temperature value 15.00 ℃
I (33533) example: Temperature value 15.00 ℃
I (34533) example: Temperature value 15.00 ℃
I (35533) example: Temperature value 15.00 ℃
I (36533) example: Temperature value 15.00 ℃
I (37533) example: Temperature value 15.00 ℃
I (38533) example: Temperature value 15.00 ℃
I (39533) example: Temperature value 15.00 ℃
I (40533) main_task: Returned from app_main()

```
[nimble_bleprph_read_temp.diff.txt](https://github.com/espressif/esp-idf/files/13764688/nimble_bleprph_read_temp.diff.txt)

closed 2023-12-29 by espressif-bot

AxelLin, 2024-01-27:
Just remind v5.1 and v5.2 need this fix.

AxelLin, 2024-01-31:
A follow up fix: bc1478f0b03dab7afc742aed2ccf605880290125

AxelLin, 2024-02-27:
@mythbuster5 v5.1.3 does not include this fix.