espressif/esp-idf #12865: Unknown Platform Linux-armv6l (Raspberry pi zero)

ZXTube, 2023-12-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-1157-g341a8f2d65

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Running `./install.sh esp32c3` should install all of the requirements and set up everything needed

### What is the actual behavior?

But instead I get this error:

    Traceback (most recent call last):
      File "/home/ziyad/esp/esp-idf/tools/idf_tools.py", line 207, in <module>
        CURRENT_PLATFORM = parse_platform_arg(PYTHON_PLATFORM)
      File "/home/ziyad/esp/esp-idf/tools/idf_tools.py", line 202, in parse_platform_arg
        fatal(f'unknown platform: {platform}')
    NameError: name 'fatal' is not defined

After research and reading what is in this pull request [`Use Linux i386/AMD64 binaries on FreeBSD`](https://github.com/espressif/esp-idf/pull/7226) I think that this error is happening because the raspberry pi zero isn't supported.

Also I'm not sure why `name 'fatal' is not defined` that could be another bug.

### Steps to reproduce.

1. Open a raspberry pi zero
2. Make sure it's platform and cpu is Linux-armv6l
3. Git Clone the esp-idf repository
4. cd into the esp-idf folder
5. run ./install.sh esp32-c3 (I think writing all instead of esp32-c3 will also give the same error)

### Build or installation Logs.

```plain
$ ./install.sh esp32c3
    Detecting the Python interpreter
    Checking "python3" ...
    Python 3.9.2
    "python3" has been detected
    Checking Python compatibility
    Installing ESP-IDF tools
    Traceback (most recent call last):
      File "/home/ziyad/esp/esp-idf/tools/idf_tools.py", line 207, in <module>
        CURRENT_PLATFORM = parse_platform_arg(PYTHON_PLATFORM)
      File "/home/ziyad/esp/esp-idf/tools/idf_tools.py", line 202, in parse_platform_arg
        fatal(f'unknown platform: {platform}')
    NameError: name 'fatal' is not defined
```

### More Information.

This is what I ran to know the system and machine:

    $ python
    Python 3.9.2 (default, Mar 12 2021, 04:06:34) 
    [GCC 10.2.1 20210110] on linux
    Type "help", "copyright", "credits" or "license" for more information.
    >>> import platform
    >>> print(platform.system() + '-' + platform.machine())
    Linux-armv6l

dobairoland, 2023-12-27:
Hi @ZXTube.

Yes, you are right about `fatal()` not working. We will fix it.

As regarding the `Linux-armv6l` support, could you please try to add the following line: 

```
 'Linux-armv6l': PLATFORM_LINUX_ARM32,
```

at

https://github.com/espressif/esp-idf/blob/master/tools/idf_tools.py#L161C14-L161C14.

Please report back to us if the given hack is sufficient and everything works as should be after installing the tools.

sidreco214, 2024-01-12:
I met same problem. I try adding line as @dobairoland mentioned, but the same error arise.

## esp-idf/tools/idf_tools.py
```console
PLATFORM_LINUX_ARM32: PLATFORM_LINUX_ARM32,
'arm-linux-gnueabi': PLATFORM_LINUX_ARM32,
'Linux-armv7l': PLATFORM_LINUX_ARM32,
'Linux-arm': PLATFORM_LINUX_ARM32,
'Linux-armv6l': PLATFORM_LINUX_ARM32,
```

## Error
```console
./install.sh
Detecting the Python interpreter
Checking "python3" ...
Python 3.11.2
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Traceback (most recent call last):
  File "/home/rasp-find/esp/esp-idf/tools/idf_tools.py", line 208, in <module>
    CURRENT_PLATFORM = parse_platform_arg(PYTHON_PLATFORM)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/rasp-find/esp/esp-idf/tools/idf_tools.py", line 203, in parse_platform_arg
    fatal(f'unknown platform: {platform}')
    ^^^^^
NameError: name 'fatal' is not defined
```

## Additional information
Linux raspberrypi 6.1.0-rpi7-rpi-v6 #1 Raspbian 1:6.1.63-1+rpt1 (2023-11-24) armv6l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

closed 2024-02-28 by espressif-bot