espressif/esp-idf #12868: [Feature] Flashing an ESP32-s3 with WebUSB - is it possible?

chipweinberger, 2023-12-24:
Has anyone tried this?

Any resources for implementing this?

Where should I look into esptool.py for reference?

chipweinberger, 2023-12-24:
@dobairoland

igrr, 2023-12-24:
WebSerial: https://github.com/espressif/esptool-js

WebUSB: https://github.com/espressif/esptool-js/issues/109#issue-1844510397

chipweinberger, 2023-12-24:
That's awesome! I assume WebSerial uses USB under the hood?

Look at us working on Christmas Eve 😅

chipweinberger, 2023-12-25:
> The Web Serial API provides a way for websites to read from and write to a serial device with JavaScript. Serial devices are connected either through a serial port on the user's system or through removable USB and Bluetooth devices that emulate a serial port

Seems like yes, it will work over USB Serial Jtag

closed 2023-12-25 by chipweinberger