espressif/esp-idf #12869: ota时出现了这个报错：“transport_base: esp_tls_conn_read error, errno=No more processes”

caidamao, 2023-12-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2-dirty

### Espressif SoC revision.

ESP32 32D

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

null

### Power Supply used.

USB

### What is the expected behavior?

正常OTA

### What is the actual behavior?

ota出现这个报错，然后可能就会一直卡住，直到软件超时导致复位。
这个问题是概率性出现，不是每次都会这样。

### Steps to reproduce.

1. 链接WiFi
2. 观察log 的进度 percent = x%
3. 概率性卡住

### Debug Logs.

```plain
I (3437) esp_netif_handlers: sta ip: 192.168.31.152, mask: 255.255.255.0, gw: 192.168.31.1
I (3437) wifi station: got ip:192.168.31.152
I (3437) wifi station: connected to ap SSID:??? password:==
I (3447) main_task: Returned from app_main()
W (8447) esp_https_ota: Continuing with insecure option because CONFIG_ESP_HTTPS_OTA_ALLOW_HTTP is set.
I (9447) CPU_INFO: Minimum free heap size: 92556 bytes

I (9447) wifi station: ota timer = 1
I (9667) esp_https_ota: Starting OTA...
I (9667) esp_https_ota: Writing to partition subtype 17 at offset 0x200000
I (10447) wifi station: image_size = 1514256ota_read_image_size = 13312 percent = 0
I (10447) CPU_INFO: Minimum free heap size: 87780 bytes

I (10447) wifi station: ota timer = 2
I (11457) wifi station: image_size = 1514256ota_read_image_size = 49152 percent = 3
I (11467) CPU_INFO: Minimum free heap size: 84488 bytes

I (11467) wifi station: ota timer = 3
[... 313 lines cut ...]
I (88447) wifi station: image_size = 1514256ota_read_image_size = 1495545 percent = 98
I (88447) CPU_INFO: Minimum free heap size: 79972 bytes

I (88447) wifi station: ota timer = 80
I (89447) wifi station: image_size = 1514256ota_read_image_size = 1495545 percent = 98
I (89447) CPU_INFO: Minimum free heap size: 79972 bytes

I (89447) wifi station: ota timer = 81
I (90447) wifi station: image_size = 1514256ota_read_image_size = 1495545 percent = 98
I (90447) CPU_INFO: Minimum free heap size: 79972 bytes

I (90447) wifi station: ota timer = 82
I (91447) wifi station: image_size = 1514256ota_read_image_size = 1495545 percent = 98
I (91447) CPU_INFO: Minimum free heap size: 79972 bytes

I (91447) wifi station: ota timer = 83
I (92447) wifi station: image_size = 1514256ota_read_image_size = 1495545 percent = 98
I (92447) CPU_INFO: Minimum free heap size: 79972 bytes

I (92447) wifi station: ota timer = 84
I (93447) wifi station: image_size = 1514256ota_read_image_size = 1495545 percent = 98
I (93447) CPU_INFO: Minimum free heap size: 79972 bytes

I (93447) wifi station: ota timer = 85
```

caidamao, 2023-12-25:
特别的操作，我开启了两个特殊的选项
[image]
[image]

caidamao, 2023-12-25:
我测试这个问题的时候，发现使用小米AX6000相对容易复现。我用另一把手机测试时，好像目前没有遇到此问题。

caidamao, 2023-12-25:
我准备了复现用的代码，请配置ssid跟WiFi密码。跟一个ESP32可以使用的OSS服务器上的固件，这个固件请在分区表允许的情况下尽量大，较大的固件会更容易复现。并且不清楚是不是错觉，蓝牙连接中的情况下好像更容易出现问题。
同时，存在一个特殊的情况：我更新了这个问题的修复在idf内https://github.com/espressif/esp-idf/issues/12718
但问题在5.1.1内可复现，应该影响不大
[gatt_server_service_table.zip](https://github.com/espressif/esp-idf/files/13764971/gatt_server_service_table.zip)

ESP-YJM, 2023-12-25:
从日志看，是设备与服务器之间没有数据交互导致 OTA 流程卡死了。你试一下修改 `esp_http_client_config_t config`, 添加 keep_alive_enable 来检测此时服务器与设备之间的 TCP 是否异常。另外，如果你可以抓包的话，也可以抓下包看下。

caidamao, 2023-12-25:
> 从日志看，是设备与服务器之间没有数据交互导致 OTA 流程卡死了。你试一下修改 `esp_http_client_config_t config`, 添加 keep_alive_enable 来检测此时服务器与设备之间的 TCP 是否异常。另外，如果你可以抓包的话，也可以抓下包看下。

这个？
[image]

ESP-YJM, 2023-12-25:
对

caidamao, 2023-12-25:
> 对
[image]

我试了一下，卡住的问题似乎有改善，但是还是很高频率的出现：
transport_base: esp_tls_conn_read error, errno=No more processes
我上面的工程已经是删减掉很多东西的代码了，但是也会出现这个问题。
@ESP-YJM

ESP-YJM, 2023-12-25:
`transport_base: esp_tls_conn_read error, errno=No more processes` 这个错其实可以忽略，这个日志的意思是读数据没有读到，但没有读到并不是错误。主要看 OTA 后续还能不能继续下去。如果不能，一直卡着不动，这个时候 `keep-alive` 功能使能后就会检测 TCP 链路，如果 keep-alive 检测到链路断了，设备就会自行断开连接，OTA 就以失败告终；如果检测不到链路断了，那这个时候就需要抓包看下，看看是服务器的原因还是设备的原因。

ESP-YJM, 2024-03-06:
@caidamao Any update for this issue? I am closing this issue. If you have, feel free to open again. Thanks.

closed 2024-03-06 by ESP-YJM