espressif/esp-idf #12870: esp_http_client_open() should return specific error code when certificate is invalid

JB-DX, 2023-12-25:
### Is your feature request related to a problem?

esp_http_client_open() returns just ESP_ERR_HTTP_CONNECT even when the certificate has expired or is invalid. There's no specific error code reflecting this particular reason.
I see the reason nicely in the debug-log, but is there a way for the caller to know that reason in order to take further action?
If not, could an event or similar be added to report that reason?

hmalpani, 2023-12-26:
Hello @JB-DX 
You can get the error code using the event_handler set in the esp_http_client_config_t. On `HTTP_EVENT_ERROR`, esp_tls_error_handle_t is sent in the handler argument. You can use that to get the error codes.

```
esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
    static char *output_buffer;  // Buffer to store response of http request from event handler
    static int output_len;       // Stores number of bytes read
    switch(evt->event_id) {
        case HTTP_EVENT_ERROR:
            esp_tls_error_handle_t err_handle = (esp_tls_error_handle_t)evt->data;
            int esp_tls_code = 0, esp_tls_flags = 0;
            esp_err_t ret = esp_tls_get_and_clear_last_error(err_handle, &esp_tls_code, &esp_tls_flags);
            ESP_LOGI(TAG, "HTTP_EVENT_ERROR: %s: esp_tls_code: 0x%04x, esp_tls_flags: 0x%04x", esp_err_to_name(ret), esp_tls_code, esp_tls_flags);
            break;
.
.

```

JB-DX, 2023-12-27:
Thank you @hmalpani,
I tried it but don't get a HTTP_EVENT_ERROR 
I still use IDF v4.4-dev-479
Was  it only implemented in newer versions?

hmalpani, 2023-12-28:
Hello @JB-DX 
Can you try it on release/v4.4 branch. It was not implemented on v4.4-dev. I have verified it on release/v4.4 branch. I think cherry-picking 9812879 can also help. Hope it helps.

Thanks!

hmalpani, 2024-01-12:
Hello @JB-DX 
Using the latest 4.4 release should solve this issue.
Closing this issue. Please feel free to reopen if you still face the issue

closed 2024-01-12 by hmalpani

JB-DX, 2024-01-23:
I migrated to v5.1 so it took a bit longer to finish.
It now works as you described.
Thanks you !