espressif/esp-idf #12871: Not able to provision the ESP - BLE-MESH Example.

Bosemani, 2023-12-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hai,
  My sdk version is ESP-IDF v5.3-dev-1157 & Esp32-c3 development board.
I tried onoff_client, onoff_server, sensor ble mesh examples under this location `esp-idf\examples\bluetooth\esp_ble_mesh`
I haven't modified anything on code. I just changed BLE mesh log level.
I'm using NRF BLE MESH Android app for provisioning but I'm not able to complete the provisioning process. It always fails.

Here is the on-off server example on esp32-c3 log:
```c
I (30) boot: ESP-IDF v5.3-dev-1157-g341a8f2d65 2nd stage bootloader
I (30) boot: compile time Dec 26 2023 10:33:55
I (31) boot: chip revision: v0.3
I (35) boot.esp32c3: SPI Speed      : 80MHz
I (40) boot.esp32c3: SPI Mode       : DIO
I (44) boot.esp32c3: SPI Flash Size : 2MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00177000
I (88) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=3c0e0020 size=3b4c8h (242888) map
I (140) esp_image: segment 1: paddr=0004b4f0 vaddr=3fc90a00 size=02f28h ( 12072) load
I (142) esp_image: segment 2: paddr=0004e420 vaddr=40380000 size=01bf8h (  7160) load
I (147) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=dc5fch (902652) map
I (301) esp_image: segment 4: paddr=0012c624 vaddr=40381bf8 size=0ec34h ( 60468) load
I (317) boot: Loaded app from partition at offset 0x10000
I (317) boot: Disabling RNG early entropy source...
I (329) cpu_start: Unicore app
I (338) cpu_start: Pro cpu start user code
I (338) cpu_start: cpu freq: 160000000 Hz
I (338) cpu_start: Application information:
I (341) cpu_start: Project name:     onoff_server
I (347) cpu_start: App version:      1
I (351) cpu_start: Compile time:     Dec 26 2023 11:09:40
I (357) cpu_start: ELF file SHA256:  562da1cc4...
I (363) cpu_start: ESP-IDF:          v5.3-dev-1157-g341a8f2d65
I (369) cpu_start: Min chip rev:     v0.3
I (374) cpu_start: Max chip rev:     v1.99
I (379) cpu_start: Chip rev:         v0.3
I (383) heap_init: Initializing. RAM available for dynamic allocation:
I (391) heap_init: At 3FC9B410 len 00024BF0 (146 KiB): RAM
I (397) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (404) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (411) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (418) spi_flash: detected chip: generic
I (422) spi_flash: flash io: dio
W (426) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (439) sleep: Configure to isolate all GPIO pins in sleep state
I (446) sleep: Enable automatic switching of GPIO sleep configuration
I (453) coexist: coex firmware version: c02915e
I (467) coexist: coexist rom version 9387209
I (468) main_task: Started on CPU0
I (468) main_task: Calling app_main()
I (468) EXAMPLE: Initializing...
I (468) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (478) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (488) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (508) BLE_INIT: BT controller compile version [b877d66]
I (508) BLE_INIT: Bluetooth MAC: 7c:df:a1:c2:0e:82
I (508) phy_init: phy_version 1130,b4e4b80,Sep  5 2023,11:09:30
Bluetooth Mesh v1.1 commit: [154b4fcc98]
I (568) BLE_MESH: Friend not supported
I (628) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (628) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (628) EXAMPLE: BLE Mesh Node initialized
I (638) main_task: Returned from app_main()
I (6818) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
W (8468) BLE_MESH: No Health Server context provided
I (9368) BLE_MESH: Algorithm:   0x00
I (9368) BLE_MESH: Public Key:  0x00
I (9368) BLE_MESH: Auth Method: 0x00
I (9368) BLE_MESH: Auth Action: 0x00
I (9378) BLE_MESH: Auth Size:   0x00
I (9638) BLE_MESH: Remote Random: 977f71785a107b753fa2ab3ddfec264f
I (9738) BLE_MESH: Primary Element: 0x001c
I (9738) BLE_MESH: net_idx 0x0000 flags 0x00 iv_index 0x0000
I (9738) BLE_MESH: DevKey 0600416b9d5b3221629100404c71b29f
I (9738) BLE_MESH: NetKey ae415409468ef870f8559d1279eaaa97
W (9748) BLE_MESH: No Health Server context provided
I (9758) BLE_MESH: Primary address 0x001c, element count 3
I (9758) BLE_MESH: Scan is already started
I (9768) EXAMPLE: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-GATT
I (9768) EXAMPLE: ESP_BLE_MESH_NODE_PROV_COMPLETE_EVT
I (9778) EXAMPLE: net_idx: 0x0000, addr: 0x001c
I (9778) EXAMPLE: flags: 0x00, iv_index: 0x00000000
W (10808) BT_HCI: hcif disc complete: hdl 0x2, rsn 0x13
I (17158) BLE_MESH: recv, app_idx 0xfffe src 0x0001 dst 0x001c
I (17158) BLE_MESH: recv, len 3: 800800
I (17158) BLE_MESH: send, app_idx 0xfffe src 0x001c dst 0x0001
I (17168) BLE_MESH: send, len 32: 0200e502000000000a0003000000020000000010000001000010000001000010
I (17178) BLE_MESH: Seg, send_tag 0x03, send_szmic 0, aszmic 0
I (17178) BLE_MESH: Use security credentials 0x00, proxy 0
I (17188) BLE_MESH: Network PDU, count 2, interval 20
I (17188) BLE_MESH: Use security credentials 0x00, proxy 0
I (17198) BLE_MESH: Network PDU, count 2, interval 20
I (17208) BLE_MESH: Use security credentials 0x00, proxy 0
I (17208) BLE_MESH: Network PDU, count 2, interval 20
I (18228) BLE_MESH: Attempts: 3
I (18228) BLE_MESH: Resending 0/2, cred 0x00
I (18228) BLE_MESH: Resending 1/2, cred 0x00
I (18228) BLE_MESH: Resending 2/2, cred 0x00
I (19248) BLE_MESH: Attempts: 2
I (19248) BLE_MESH: Resending 0/2, cred 0x00
I (19248) BLE_MESH: Resending 1/2, cred 0x00
I (19248) BLE_MESH: Resending 2/2, cred 0x00
I (20268) BLE_MESH: Attempts: 1
I (20268) BLE_MESH: Resending 0/2, cred 0x00
I (20268) BLE_MESH: Resending 1/2, cred 0x00
I (20268) BLE_MESH: Resending 2/2, cred 0x00
I (21288) BLE_MESH: Attempts: 0
I (21288) BLE_MESH: Resending 0/2, cred 0x00
I (21288) BLE_MESH: Resending 1/2, cred 0x00
I (21288) BLE_MESH: Resending 2/2, cred 0x00
W (22308) BLE_MESH: Ran out of retransmit attempts
W (358928) BT_HCI: hcif disc complete: hdl 0x2, rsn 0x13

``` 

In Nrf mesh app it's getting stop at "Sending composition data get...."
[image]

Why is this happening? How to resolve this.

Best Regards
_Bose_

Bosemani, 2023-12-28:
update
I tried in `ESP-IDF v5.1.1-439-gcb174b0fe1-dirty` version Its working.
But `ESP-IDF v5.3-dev-1157-g341a8f2d65` not able to provision.
Why it's not working in master branch?

Bosemani, 2024-02-13:
hai @forx157 
Any update?

Herbrant, 2024-02-20:
Hi @Bosemani! I have the same problem using the nrf mesh application for Android. I tried doing the same with the Apple appand it worked without any problem.
If you take a look to the source code of the nrf mesh application it seems that the Android app supports BLE Mesh 1.0.1 while the Apple app supports BLE Mesh 1.1. Maybe the problem is related to the version of the supported protocol by the application.

If you don't have an apple product, you can try to do the same using an esp32 as provisioner.

Bosemani, 2024-02-20:
Hai @Herbrant 
Thank you for your response.
I tested by using another esp32 as provisioner. Its working well.
In NRF mesh android app, we need to wait for their update.

Herbrant, 2024-02-20:
Right now I have a problem with the provisioner and the onoff_client of the examples directory.
It seems that the provisioner has some problem in the appkey configuration phase:
```
...
W (58547) BLE_MESH: Receive generic status message timeout
I (58547) EXAMPLE: example_ble_mesh_generic_client_cb, error_code = 0x00, event = 0x03, addr: 0x0008, opcode: 0x8201
I (58557) BLE_MESH: Client message 0x00008201 with timeout 4000ms
I (58567) BLE_MESH: send, app_idx 0x0000 src 0x0001 dst 0x0008
I (58567) BLE_MESH: send, len 2: 8201
I (58577) BLE_MESH: Unseg, send_tag 0x00, send_szmic 0, aszmic 0
I (58587) BLE_MESH: Use security credentials 0x00, proxy 0
I (58587) BLE_MESH: Network PDU, count 2, interval 20
W (62597) BLE_MESH: Receive generic status message timeout
I (62597) EXAMPLE: example_ble_mesh_generic_client_cb, error_code = 0x00, event = 0x03, addr: 0x0008, opcode: 0x8201
I (62607) BLE_MESH: Client message 0x00008201 with timeout 4000ms
I (62607) BLE_MESH: send, app_idx 0x0000 src 0x0001 dst 0x0008
I (62617) BLE_MESH: send, len 2: 8201
I (62617) BLE_MESH: Unseg, send_tag 0x00, send_szmic 0, aszmic 0
I (62627) BLE_MESH: Use security credentials 0x00, proxy 0
I (62637) BLE_MESH: Network PDU, count 2, interval 20
W (66647) BLE_MESH: Receive generic status message timeout
I (66647) EXAMPLE: example_ble_mesh_generic_client_cb, error_code = 0x00, event = 0x03, addr: 0x0008, opcode: 0x8201
I (66647) BLE_MESH: Client message 0x00008201 with timeout 4000ms
I (66657) BLE_MESH: send, app_idx 0x0000 src 0x0001 dst 0x0008
I (66657) BLE_MESH: send, len 2: 8201
I (66667) BLE_MESH: Unseg, send_tag 0x00, send_szmic 0, aszmic 0
I (66677) BLE_MESH: Use security credentials 0x00, proxy 0
I (66677) BLE_MESH: Network PDU, count 2, interval 20
```

onoff_client:
```
...
I (3530) EXAMPLE: net_idx 0x0000, app_idx 0x0000
I (3540) AppKey: 12 12 12 12 12 12 12 12 12 12 12 12 12 12 12 12 
I (3680) BLE_MESH: recv, app_idx 0xfffe src 0x0001 dst 0x0008
I (3680) BLE_MESH: recv, len 8: 803d080000000010
I (3680) BLE_MESH: send, app_idx 0xfffe src 0x0008 dst 0x0001
I (3690) BLE_MESH: send, len 9: 803e02080000000010
I (3700) BLE_MESH: Unseg, send_tag 0x02, send_szmic 0, aszmic 0
I (3700) BLE_MESH: Use security credentials 0x00, proxy 0
I (3710) BLE_MESH: Network PDU, count 2, interval 20
I (3780) BLE_MESH: recv, app_idx 0x0000 src 0x0001 dst 0x0008
I (3780) BLE_MESH: recv, len 2: 8201
I (7830) BLE_MESH: recv, app_idx 0x0000 src 0x0001 dst 0x0008
I (7830) BLE_MESH: recv, len 2: 8201
I (11880) BLE_MESH: recv, app_idx 0x0000 src 0x0001 dst 0x0008
I (11880) BLE_MESH: recv, len 2: 8201
I (15920) BLE_MESH: recv, app_idx 0x0000 src 0x0001 dst 0x0008
I (15920) BLE_MESH: recv, len 2: 8201
I (19960) BLE_MESH: recv, app_idx 0x0000 src 0x0001 dst 0x0008
I (19960) BLE_MESH: recv, len 2: 8201
I (24020) BLE_MESH: recv, app_idx 0x0000 src 0x0001 dst 0x0008
I (24020) BLE_MESH: recv, len 2: 8201
I (28060) BLE_MESH: recv, app_idx 0x0000 src 0x0001 dst 0x0008
I (28060) BLE_MESH: recv, len 2: 8201
I (32100) BLE_MESH: recv, app_idx 0x0000 src 0x0001 dst 0x0008
I (32100) BLE_MESH: recv, len 2: 8201
I (36150) BLE_MESH: recv, app_idx 0x0000 src 0x0001 dst 0x0008
I (36150) BLE_MESH: recv, len 2: 8201
I (40190) BLE_MESH: recv, app_idx 0x0000 src 0x0001 dst 0x0008
I (40190) BLE_MESH: recv, len 2: 8201
I (44240) BLE_MESH: recv, app_idx 0x0000 src 0x0001 dst 0x0008
I (44240) BLE_MESH: recv, len 2: 8201
```

You are able to send messages from the client to the server? @Bosemani

ankit-thealchemist, 2024-02-22:
I have done provisioning on the tag v5.1.2 on esp32 c3. Its working fine for me

Bosemani, 2024-02-22:
> Right now I have a problem with the provisioner and the onoff_client of the examples directory.
> It seems that the provisioner has some problem in the appkey configuration phase:
> ```
> ...
> W (58547) [...]

Hai @Herbrant 
I tried the same example, but I'm not facing any issues. But I haven't modified any code. Just loaded in two boards and checked the logs. It's working. I'm using the master branch.

Bosemani, 2024-02-22:
> I have done provisioning on the tag v5.1.2 on esp32 c3. Its working fine for me

Hai @ankit-thealchemist 
Did you tried the master branch? In the 5.1 version it's working.but recent versions it's not working using the nrf app.

ankit-thealchemist, 2024-02-22:
@Bosemani No, I never work on the master branch, its in the dev phase and buggy. I always use the tags, they are stable. 

Check this, it happened it before also 
https://github.com/espressif/esp-idf/issues/6997

chegewara, 2024-02-23:
https://github.com/espressif/esp-idf/issues/12254#issuecomment-1960775828

Fighter19, 2024-02-24:
> I have done provisioning on the tag v5.1.2 on esp32 c3. Its working fine for me

I can confirm, had problem with the master, I built today. No problem with the nRF App an v5.1.2.
Thanks for the hint.

Bosemani, 2024-02-26:
> > I have done provisioning on the tag v5.1.2 on esp32 c3. Its working fine for me
> 
> I can confirm, had problem with the master, I built today. No problem with the nRF App an v5.1.2. Thanks for the hint.

Hai @Fighter19 
I'm using v5.3-dev-2032-g4d90eedb6e.
What is your version? 
also are u using android nrf mesh app?

Bosemani, 2024-02-26:
> [#12254 (comment)](https://github.com/espressif/esp-idf/issues/12254#issuecomment-1960775828)

Hai @chegewara 
I don't have any provisioned devices to try this. Also, We are planning MESH V1.1 devices only in a network.
Do you know any other Android app, that will support BLE Mesh V1.1 devices?

chegewara, 2024-02-26:
Hi,
I am 90% sure its issue with espressif mesh code, in particular with proxy out characteristic in proxy server.
I tested with nrf mesh, sti mesh and silab mesh app, also with espressif espmesh app. None can provision mesh node from master.

If you dont have any provisioned esp32 devices, then its easy solution. Grab esp32 devkit and flash any mesh example built with idf 5.1. You have tool to work with mesh 5.1, because you can continue provisioning and play with mesh v1.1.

Specs says that mesh 1.1 shall be backward compatible, so any provisioner which works with mesh 1.0 should works with v1.1, just wont handle new features.

Bosemani, 2024-02-26:
> Hi,
> I am 90% sure its issue with espressif mesh code, in particular with proxy out characteristic in proxy server.
> I tested with nrf mesh, sti mesh and silab mesh app, also with espressif espmesh [...]

Hai @chegewara 
Thank you so much for your reply.
I will try to use idf version 5.1 and master brach for testing.

chegewara, 2024-02-27:
There is one more thing i just learned.
esp32 with ble mesh v1.1 works with silab mesh app, but this time is the issue (small) with the app.
- after initial provisioning app stuck
- when you restart app you can work with esp32 node, which is provisioned
- the way this app is handling models kinda sucks, at least comparing to nRF mesh, but you dont need this extra esp32 node i mentioned earlier to work

chegewara, 2024-02-29:
Looks like i dont know what i am saying about, maybe except how to workaround provisioning issues with nrf mesh.

Here is what i found:
- even esp32 ble mesh provisioner v1.0 cant provision device with mesh v1.1 
- composition data in v1.1 looks different (additional 2 bytes value in front), according to logs
-- composition data from v1.0 `e502 0000 0000 0a00 0300` and v1.1 `0200 e502 0000 0000 0a00 0300`
- ble mesh v1.1 specs does not mention about this additional value - table 4.2 -  https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=554899

Bosemani, 2024-02-29:
> Looks like i dont know what i am saying about, maybe except how to workaround provisioning issues with nrf mesh.
>
> Here is what i found:
> - even esp32 ble mesh provisioner v1.0 cant provision device [...]

Hai @chegewara 
 I haven't looked into it.
It's Great. This means, we need a V1.1 provisioner to provision the v1.1 device.
Nrf iOS app will support v1.1, that's the reason we can provision using iOS.
But Android nrf doesn't support v1.1.

chegewara, 2024-02-29:
I believe this is relevant info

mesh v1.0
[image]

mesh v1.1
[image]

chegewara, 2024-03-02:
Ok, here is finally the proof why mesh v1.1 node cant be provisioned (what i suspected earlier).
Here is test case:
- provisioner - esp32s3 with custom provisioner firmware - master branch
- node - esp32 with custom node firmware - master branch

Logs from node (relevant only):
```
I (42151) mesh-cb: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
I (42301) BLE_MESH: Algorithm:   0x00
I (42302) BLE_MESH: Public Key:  0x00
I (42302) BLE_MESH: Auth Method: 0x00
I (42304) BLE_MESH: Auth Action: 0x00
I (42308) BLE_MESH: Auth Size:   0x00
W (42413) BT_GATT: attribute value too long, to be truncated to 20
Mismatch proxy client type 0x00
Mismatch proxy client type 0x00
Mismatch proxy client type 0x00
W (42481) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x13
W (42486) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13
I (42497) mesh-cb: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-GATT
E (42661) BT_APPL: bta_gattc_process_indicate indication/notif for unknown device, ignore
Mismatch proxy client type 0x00
Mismatch proxy client type 0x00
Mismatch proxy client type 0x00
W (43776) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x8
W (43780) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x8
```
and this is most important line:
```
E (42661) BT_APPL: bta_gattc_process_indicate indication/notif for unknown device, ignore
```
From bluetooth specs
```
7.1.3.2.1 Characteristic behavior
The Mesh Provisioning Data Out characteristic is used to Proxy PDU message containing Provisioning
PDU from a Provisioning Server to a Provisioning Client.
A Provisioning Server is not required to support bonding. As a bonding relationship is not established
between the Provisioning Client and the Provisioning Server, the Provisioning Client **shall enable**
notifications (**write value 0x0001**) to the Mesh Provisioning Data Out Client Characteristic Configuration
Descriptor after a connection is established and before sending the first message of the Proxy PDU.
```

Thanks

chegewara, 2024-03-02:
Ok, here is finally the proof why mesh v1.1 node cant be provisioned (what i suspected earlier).
Here is test case:
- provisioner - esp32s3 with custom provisioner firmware - master branch
- node - esp32 with custom node firmware - master branch

Logs from node (relevant only):
```
I (42151) mesh-cb: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
I (42301) BLE_MESH: Algorithm:   0x00
I (42302) BLE_MESH: Public Key:  0x00
I (42302) BLE_MESH: Auth Method: 0x00
I (42304) BLE_MESH: Auth Action: 0x00
I (42308) BLE_MESH: Auth Size:   0x00
W (42413) BT_GATT: attribute value too long, to be truncated to 20
Mismatch proxy client type 0x00
Mismatch proxy client type 0x00
Mismatch proxy client type 0x00
W (42481) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x13
W (42486) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13
I (42497) mesh-cb: ESP_BLE_MESH_NODE_PROV_LINK_CLOSE_EVT, bearer PB-GATT
E (42661) BT_APPL: bta_gattc_process_indicate indication/notif for unknown device, ignore
Mismatch proxy client type 0x00
Mismatch proxy client type 0x00
Mismatch proxy client type 0x00
W (43776) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x8
W (43780) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x8
```
From what i understand, due to `proxy mismatch` node is disconnecting from provisioner.
There is no problem with provisioning over PB-ADV tho.

Thanks

Bosemani, 2024-03-05:
> CONFIG_BLE_MESH_PROV_EPA

Hai @chegewara 
Thank you for your great effort.
do you have any workaround for this?

chegewara, 2024-03-09:
Please ignore my previous posts, it was bunch of random issues not related and most likely due to some misconfig in my mesh nodes
I even got at some point issue like this, but i cant reproduce it anymore
`BLE_MESH: Invalid length 18 for type 0x03`

Concluding, my first thought about nrf mesh not being able to provisioning esp32 was most likely correct 
@Bosemani can you grab logs from esp32 node when provisioning with iOS?
nrf mesh provisioning logs
```
W (35463) BLE_MESH: PDU check: length 1 for type 0x00
W (46683) BLE_MESH: PDU check: length 5 for type 0x02
I (46684) BLE_MESH: Algorithm:   0x00
I (46684) BLE_MESH: Public Key:  0x00
I (46687) BLE_MESH: Auth Method: 0x00
I (46691) BLE_MESH: Auth Action: 0x00
I (46696) BLE_MESH: Auth Size:   0x00
W (46745) BLE_MESH: PDU check: length 64 for type 0x03
W (46863) BLE_MESH: PDU check: length 16 for type 0x05
W (46923) BLE_MESH: PDU check: length 16 for type 0x06
I (46924) BLE_MESH: Remote Random: 7cd0b4b28c6633eda17b8e4941f3e895
W (46984) BLE_MESH: PDU check: length 33 for type 0x07
I (46987) BLE_MESH: Primary Element: 0x0078
I (46987) BLE_MESH: net_idx 0x0000 flags 0x00 iv_index 0x0000
I (46990) BLE_MESH: DevKey ec629c1da74fd3a64618c31618d1f153
I (46997) BLE_MESH: NetKey 0b82d9555a000a44e15d1f71acac3656
I (47008) BLE_MESH: Primary address 0x0078, element count 1
Mismatch proxy client type 0x00
Mismatch proxy client type 0x00
Mismatch proxy client type 0x00
W (48064) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x13
W (48068) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13
I (51011) BLE_MESH: Use security credentials 0x00, proxy 1
I (51027) BLE_MESH: Use security credentials 0x00, proxy 1
I (53105) BLE_MESH: recv, app_idx 0xfffe src 0x1990 dst 0x0078
I (53105) BLE_MESH: recv, len 3: 800800
I (53106) BLE_MESH: send, app_idx 0xfffe src 0x0078 dst 0x1990
I (53112) BLE_MESH: send, len 20: 0200e502000000000a0003000000020000000010
I (53119) BLE_MESH: Seg, send_tag 0x03, send_szmic 0, aszmic 0
I (53127) BLE_MESH: Use security credentials 0x00, proxy 0
I (53133) BLE_MESH: Network PDU, count 2, interval 20
```

vs esp32 provisioning over GAT  logs
```
E (1122) BT_APPL: bta_gattc_process_indicate indication/notif for unknown device, ignore
W (2351) BLE_MESH: PDU check: length 1 for type 0x00
W (2442) BLE_MESH: PDU check: length 5 for type 0x02
I (2443) BLE_MESH: Algorithm:   0x00
I (2444) BLE_MESH: Public Key:  0x00
I (2446) BLE_MESH: Auth Method: 0x00
I (2450) BLE_MESH: Auth Action: 0x00
I (2454) BLE_MESH: Auth Size:   0x00
W (2471) BLE_MESH: PDU check: length 64 for type 0x03
W (2652) BLE_MESH: PDU check: length 16 for type 0x05
W (2711) BLE_MESH: PDU check: length 16 for type 0x06
I (2712) BLE_MESH: Remote Random: 541284747a8a001f986843e2f61ef7d9
W (2772) BLE_MESH: PDU check: length 33 for type 0x07
I (2775) BLE_MESH: Primary Element: 0x0005
I (2775) BLE_MESH: net_idx 0x0000 flags 0x00 iv_index 0x0000
I (2778) BLE_MESH: DevKey 1cc7f604a80b1e6d2ceda3c85bbb4d96
I (2784) BLE_MESH: NetKey 5463144a3c4af3c698ac3d5d07797527
I (2796) BLE_MESH: Primary address 0x0005, element count 1
Mismatch proxy client type 0x00
Mismatch proxy client type 0x00
Mismatch proxy client type 0x00
W (2866) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x13
W (2868) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13
I (2943) BLE_MESH: recv, app_idx 0xfffe src 0x0001 dst 0x0005
I (2943) BLE_MESH: recv, len 3: 800800
I (2944) BLE_MESH: send, app_idx 0xfffe src 0x0005 dst 0x0001
I (2949) BLE_MESH: send, len 20: 0200e502000000000a0003000000020000000010
I (2957) BLE_MESH: Seg, send_tag 0x03, send_szmic 0, aszmic 0
I (2964) BLE_MESH: Use security credentials 0x00, proxy 0
I (2970) BLE_MESH: Network PDU, count 2, interval 20
```

nRF does not try to write to 0x2902 in proxy characteristic, which is not required according to bluetooth mesh specs. 
`E (1122) BT_APPL: bta_gattc_process_indicate indication/notif for unknown device, ignore`
mesh node suppose to do that on its own during connection.
Since its not enabled, proxy out characteristic cant send notifications

chegewara, 2024-03-12:
Ive found it, finally.
It has nothing to do with anything i said earlier, and for sure with ble mesh v1.1.
I dont know why, probably a bug, but provisioning cant pass this check
https://github.com/espressif/esp-idf/blob/master/components/bt/esp_ble_mesh/core/net.c#L1813
when i remove this exact check, then i can provisioning with nrf connect, like usual
```
bt_mesh_private_gatt_proxy_state_get() == BLE_MESH_PRIVATE_GATT_PROXY_DISABLED &&
```

I hope espressif can clarify why is this happen.

Thanks

ruslanslobozhenuk, 2024-03-13:
> Наконец-то я нашел его. Это не имеет ничего общего ни с чем, о чем я говорил ранее, и уж точно с ble mesh v1.1. подготовка не может пройти эту проверку [https://github.com/espressif/esp-idf/blob/master/comComponents/bt/esp_ble_mesh/core/net.c#L1813](https://github.com/espressif/esp-idf/blob/master/components/bt/esp_ble_mesh/core/net.c#L1813) Это исправление
> 
> ```
>     if (IS_ENABLED(CONFIG_BLE_MESH_GATT_PROXY_SERVER) &&
>        bt_mesh_private_gatt_proxy_state_get() != BLE_MESH_PRIVATE_GATT_PROXY_ENABLED &&
>        net_if == BLE_MESH_NET_IF_PROXY) {
> ```
> 
> Спасибо
> 
> PS теперь он работает и с nRF Connect, и с приложением Silab.

Very good work!!! Thank YOU!!!

forx157, 2024-03-13:
@chegewara 
Thank you very much for your excellent work and apologies for the oversight in our code, we will be fixing this as soon as possible, thanks again for your efforts.

chegewara, 2024-03-13:
No problem, im glad it is that easy fix
I honestly believe this ble mesh component is very good job of you guys. 
If lame like me can easy build any mesh device, node models or provisioner, without issue and get clear info when something is wrong (like missing setup server or or light models) then its very good.

I can only complain that mesh in esp-idf 5.1 is not C++ friendly (macros wont let use it in C++), but its fixed in 5.3.

Thanks

Bosemani, 2024-03-13:
> No problem, im glad it is that easy fix
> I honestly believe this ble mesh component is very good job of you guys.
> If lame like me can easy build any mesh device, node models or provisioner, without [...]

@chegewara Very good work! Thank you.

forx157, 2024-03-14:
@chegewara 
Could you show me the code for this operation?

chegewara, 2024-03-14:
@forx157 
Sorry, i deleted my previous post because i want to perform more tests to make sure i am not doing anything stupid and later to confirm its related to vendor or all models.

I am trying to figure out whats wrong.
All i know right now, that `esp_ble_mesh_model_publish` is somehow changing data in `esp_ble_mesh_model_pub_t` net buffer. (when i pass data straight to publish function, then all works fine)
I have this code
```
        ESP_LOG_BUFFER_HEX("TAG update 3", pub.msg->data, pub.msg->len);
        esp_ble_mesh_model_publish(_model, set_opcode, pub.msg->len, pub.msg->data, ROLE_NODE);
        ESP_LOG_BUFFER_HEX("TAG update 4", pub.msg->data, pub.msg->len);
```
and this output:
```
I (47917) TAG update 3: 12 34 <---- before call to publish
I (47917) TAG update 4: c4 34 12 c4 34  <----- after call to publish
```

I have more or less complicated C++ library, which should not be the cause of the problem, but who know.

Thanks for attention

chegewara, 2024-03-14:
Im not sure if my deduction is correct, but i found something.
To perform this test i am doing some "stupid" operations, but results are interesting.
- i have this variable as a class member (its correctly initialized) `esp_ble_mesh_model_pub_t pub = {};`
- i update value in net buffer with
```
        memcpy(pub.msg->data, data, len);
        pub.msg->len = len;
        ESP_LOG_BUFFER_HEX("TAG update 2", pub.msg->data, pub.msg->len);
```
and get correct result
```
TAG update 2: 44 44 44 44 <--- data sent to vendor model from nrf mesh
```
now the strange part only for this test purpose
```
    void publishState(size_t len, uint8_t* data)
    {
        esp_ble_mesh_model_publish(_model, set_opcode, len, data, ROLE_NODE); <--- here i pass data
        ESP_LOG_BUFFER_HEX("TAG update 3", pub.msg->data, pub.msg->len);
        esp_ble_mesh_model_publish(_model, set_opcode, pub.msg->len, pub.msg->data, ROLE_NODE); <---- here i pass `pub` with previously updated data
        ESP_LOG_BUFFER_HEX("TAG update 4", pub.msg->data, pub.msg->len);
    }
```
and here is log
```
I (56527) TAG update 3: c4 34 12 44 44 44 44 <--- correctly updated pub.msg->data with opcode
I (56537) TAG update 4: c4 34 12 c4 34 12 c4 44 12 c4 <--- data corrupted
```

we can observe 2 things:
- on first call to publish i pass data straight to the function, even tho `pub` data is updated with opcode (which is probably expected in normal useage)
- on second call to publish `pub` is updated again with opcode, but the pointer is not reset to begin of data buffer, so data are corrupted

Bear in mind that in normal use case i call only one time publish function with data passed from `pub`

I hope i described it that it can be understand, even if its a bit messy

Thanks

forx157, 2024-03-14:
@chegewara 
Is the `pub msg` you are using being used to initialise the publish model?

chegewara, 2024-03-14:
@forx157 
yes, it is (i believe)
```
    BLE_MESH_MODEL_PUB_DEFINE(&pub, 3 + 20, ROLE_NODE, __COUNTER__);
    _model = new esp_ble_mesh_model_t(ESP_BLE_MESH_VENDOR_MODEL(cid, vid, vnd_op, &pub, NULL));
```
here is my updated macro, so i could use the same `pub` name in many class objects
```
#define BLE_MESH_MODEL_PUB_DEFINE(_name, _msg_len, _role, __n__)     \
    NET_BUF_SIMPLE_DEFINE_STATIC(bt_mesh_pub_msg_##__n__, _msg_len); \
    esp_ble_mesh_model_pub_t *_tmp = _name;                          \
    _tmp->msg = &bt_mesh_pub_msg_##__n__;                            \
    _tmp->update = (uint32_t)NULL;                                   \
    _tmp->dev_role = _role;
```

It is working with on-off server w/o issue

forx157, 2024-03-14:
@chegewara 
This is the problem, when you use this `pub` to initialise the publish model you should not pass this memory in `esp_ble_mesh_model_publish`. `pub.msg` this memory is used on the stack to hold the information passed in by the user, and this is caused when you pass `pub.msg` into this function ` memcpy(pub.msg.data, pub.msg.data, pub.msg.data_len)` .

For details you can refer to the implementation in the function `esp_ble_mesh_model_publish`:
```c
    if (act == BTC_BLE_MESH_ACT_MODEL_PUBLISH) {
        bt_mesh_model_msg_init(model->pub->msg, opcode);
         // if you passed pub.msg.data in,The essence of this line of code becomes
        // net_buf_simple_add_mem(model->pub->msg, model->pub->msg.data, model->pub->msg.data_len);
        net_buf_simple_add_mem(model->pub->msg, data, length);
    } else {
        msg_data = (uint8_t *)bt_mesh_malloc(op_len + length);
        if (msg_data == NULL) {
            return ESP_ERR_NO_MEM;
        }
        esp_ble_mesh_model_msg_opcode_init(msg_data, opcode);
        memcpy(msg_data + op_len, data, length);
    }
```

chegewara, 2024-03-14:
Ok, so i dont have to update data in `pub`, just pass data into publsih function and low level stack will update `pub` for me?

closed 2024-04-04 by espressif-bot