espressif/esp-idf #12872: 我采用esp32s3fh4r2作为控制器时，烧录官方蓝牙例程可以使用，但wifi不行，天线方案采用的合宙esp32s3的方案

HAN-anpang, 2023-12-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

采用esp32s3fh4r2作为控制器时，烧录官方蓝牙例程可以使用，但wifi不行，天线方案采用的合宙esp32s3的方案。主要问题是没有信号，无法连接热点也无法发出热点信号

Xiehanxin, 2023-12-26:
1.你这边的问题很可能是天线那部分的频偏没有调好。合宙应该是自己设计的模组，射频电路这一块不是官方的设计，建议问下合宙那边的支持人员
2.可以看下是只有一块出现这个问题还是好几块都是这样的问题
3.没有信号的情况下运行的 LOG 是否正常？如果 LOG 正常就是软件正常运行了，但是硬件端无反应
4.如果你是自己画PCB 可以参考官方的硬件设计指南

HAN-anpang, 2023-12-27:
> 1.你这边的问题很可能是天线那部分的频偏没有调好。合宙应该是自己设计的模组，射频电路这一块不是官方的设计，建议问下合宙那边的支持人员 2.可以看下是只有一块出现这个问题还是好几块都是这样的问题 3.没有信号的情况下运行的 LOG 是否正常？如果 LOG 正常就是软件正常运行了，但是硬件端无反应 4.如果你是自己画PCB 可以参考官方的硬件设计指南

很感谢您的建议，我的pcb是自己画的，天线做了50欧姆的阻抗，LOG正常，硬件端做测试时，我的每一块板子都不行

Sherry616, 2024-03-04:
@HAN-anpang 您好👋，请分享您关于该问题的最新进展，问题是否仍然存在？谢谢。

closed 2024-03-22 by Alvin1Zhang