espressif/esp-idf #12873: Wi-Fi does not working in 5.1.2 ESP-IDF

dm128, 2023-12-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I were checked both examples protocol/esp_http_client and wifi/fast_scan on boards S3 and C3. Example projects did not connect to the home wifi router with wpa wpa2-psk setting. Same projects works with esp-idf 5.1. Changing negotiating inn the project to wpa or wpa2-psk only has no effect. However example connected to the shared wifi on smartphone with wpa2-psk option

AxelLin, 2023-12-26:
Could you enable CONFIG_ESP_WIFI_DEBUG_PRINT=y and share the failure log?

shreyassheth1, 2023-12-27:
Hey @dm128 
We tried to reproduce the same scenario in our test setup and for us, it works fine. wifi/fast_scan on s3 can connect to AP having security wpa-wpa2 on v5.1.2. However, if you are still facing the issue can you please help us by providing logs to debug the issue?
In wifi/fast_scan directory:
- $ idf.py menuconfig # Change as below
- [image]
- [image]

dm128, 2023-12-28:
Found mistake on AP side. The ssid name was contained whitespace on the end. Phones used string obtained from the air and connecting to the same string.
Thanks a lot, closing issue

closed 2023-12-28 by dm128