espressif/esp-idf #12875: gatt_security_client can't connect mouse

jjj19950710, 2023-12-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When the gatt_security_client connects to a Bluetooth mouse, it will not trigger the secure pairing process, and it will automatically disconnect after about 110 seconds of connection.
the configure:
```
#define GATTC_TAG             "RAPOO 5.0MS"
#define REMOTE_SERVICE_UUID   ESP_GATT_UUID_HID_SVC
#define REMOTE_NOTIFY_UUID    ESP_GATT_UUID_HID_REPORT
```
the client log with esp32s3：
```
I (59427) RAPOO 5.0MS: EVT 46, gattc if 3
I (59497) RAPOO 5.0MS: EVT 18, gattc if 3
I (59497) RAPOO 5.0MS: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 23, conn_id 0
I (59497) RAPOO 5.0MS: EVT 7, gattc if 3
I (59497) RAPOO 5.0MS: SEARCH RES: conn_id = 0 is primary service 1
I (59507) RAPOO 5.0MS: start handle 25 end handle 44 current handle value 25
I (59517) RAPOO 5.0MS: UUID16: 1812
I (59517) RAPOO 5.0MS: EVT 6, gattc if 3
I (59517) RAPOO 5.0MS: Get service information from remote device
I (59527) RAPOO 5.0MS: EVT 38, gattc if 3
I (59567) RAPOO 5.0MS: EVT 9, gattc if 3
I (59567) RAPOO 5.0MS: write descr success
I (59597) RAPOO 5.0MS: EVT 4, gattc if 3
I (59597) RAPOO 5.0MS: Write char success
W (174227) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x8
W (174227) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x8
I (174227) RAPOO 5.0MS: EVT 41, gattc if 3
I (174227) RAPOO 5.0MS: ESP_GATTC_DISCONNECT_EVT, reason = 0x8
I (174237) RAPOO 5.0MS: EVT 5, gattc if 3
```
the client log with esp32:
```
I (18425) RAPOO 5.0MS: EVT 46, gattc if 3
I (18425) RAPOO 5.0MS: EVT 18, gattc if 3
I (18425) RAPOO 5.0MS: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 23, conn_id 0
I (18435) RAPOO 5.0MS: EVT 7, gattc if 3
I (18435) RAPOO 5.0MS: SEARCH RES: conn_id = 0 is primary service 1
I (18445) RAPOO 5.0MS: start handle 21 end handle 24 current handle value 21
I (18455) RAPOO 5.0MS: UUID16: 180f
I (18455) RAPOO 5.0MS: EVT 6, gattc if 3
I (18465) RAPOO 5.0MS: Get service information from remote device
I (18475) RAPOO 5.0MS: EVT 38, gattc if 3
I (18745) RAPOO 5.0MS: EVT 9, gattc if 3
I (18745) RAPOO 5.0MS: write descr success
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
I (168524) RAPOO 5.0MS: EVT 41, gattc if 3
I (168524) RAPOO 5.0MS: ESP_GATTC_DISCONNECT_EVT, reason = 0x8
I (168524) RAPOO 5.0MS: EVT 5, gattc if 3
```

jjj19950710, 2023-12-27:
[Uploading esp32-mouse.zip…]()
This is the Bluetooth packet I caught.I hope it will be helpful to you.

zhp0406, 2024-01-02:
@jjj19950710 
thanks for report the issu,I will check.

zhp0406, 2024-01-05:
The API for initiating pairing is 'esp_ble_set_encryption,' which is not actively called in the 'gatt_security_client' example. You can modify the example to initiate pairing when ESP_GATTC_CONNECT_EVT occurs.

jjj19950710, 2024-01-09:
Now the client can initiate pairing requests proactively.Thanks very much！

closed 2024-01-09 by jjj19950710