espressif/esp-idf #12876: RISC V ULP does not run in parallel w/ main CPUs

mickeyl, 2023-12-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-beta1-263-ge49823f10c

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

LilyGo T-Display S3

### Power Supply used.

USB

### What is the expected behavior?

Since the RISCV ULP is marketed as a co-processor, I would have expected it to be able to run _in parallel_ with the main CPU(s).

### What is the actual behavior?

The RISCV ULP starts to run, but halts immediately after a few cycles.

### Steps to reproduce.

I have based on the `uart_print` example, which I modified like that:

```c
void app_main(void)
{
    esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();
    /* not a wakeup from ULP, load the firmware */
    if (cause != ESP_SLEEP_WAKEUP_ULP) {
        printf("Not a ULP-RISC-V wakeup, initializing it! \n");
        init_ulp_program();
    }

    /* ULP Risc-V read and detected a change in GPIO_0, prints */
    if (cause == ESP_SLEEP_WAKEUP_ULP) {
        printf("ULP-RISC-V woke up the main CPU! \n");

    }

    while(true) {
        printf("CPU still running...\n");
        vTaskDelay(pdMS_TO_TICKS(1000));
    }

    /* Go back to sleep, only the ULP Risc-V will run */
    printf("Entering in deep sleep\n\n");

    /* Small delay to ensure the messages are printed */
    vTaskDelay(100);

    ESP_ERROR_CHECK( esp_sleep_enable_ulp_wakeup());
    esp_deep_sleep_start();
}
```

### Debug Logs.

```plain
Hello World from ULP-RISCV!
                           Cnt: 0x00000000
```

### More Information.

If I change the example to move into deep sleep, the RISCV-ULP runs forever.

ESP-Marius, 2023-12-28:
Hi,

The ULP can indeed run in parallel, it doesn't require the main processor to be sleeping. 

I've ran the example with your changes and I can still see the continuous UART output from the ULP. Maybe there is an issue on the receiving end? Are you able to connect a logic analyzer and see the output?

You could also change the `cnt` variable in the ULP to be global, and then print this one from the main cpu as well (just to check that the `while(1)` in the ULP is running)

mickeyl, 2024-01-02:
I have stripped down my example to a very basic one that only toggles a GPIO (connected to a LED). I guess I have been mistaken by problem that a reset indicated by the `idf.py` flash tool does not reset the ULP core if it is already running.

Now why the UART sometimes suddenly stops receiving data I don't know, but I'll keep poking and will open another bug, if I'm sure there is a problem.

Thanks and sorry for the noise.

closed 2024-01-02 by mickeyl