espressif/esp-idf #12877: In CMakelist.txt, idf_component_register(SRCS (C_file)) doesn't work

ElinhoHD, 2023-12-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-D0WD (revision v1.0)

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32 DevKit V1

### Power Supply used.

USB

### What is the expected behavior?

I expected my project be compiled.

### What is the actual behavior?

It shows me this 
```cmd
> idf.py build flash monitor
Executing action: all (aliases: build)
Running ninja in directory H:\git\ESP32\Wipy\build
Executing "ninja all"...
[1/1] cmd.exe /C "cd /D H:\git\ESP32\Wipy\build\bootloader\esp-idf\esptool_py && C:...-offset 0x8000 bootloader 0x1000 H:/git/ESP32/Wipy/build/bootloader/bootloader.bin"Bootloader binary size 0x6830 bytes. 0x7d0 bytes (7%) free.
[2/4] Linking CXX executable Wipy.elfFAILED: Wipy.elf
cmd.exe /C "cd . && C:\Users\mrntu\.espressif\dist\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  -g -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=H:/git/ESP32/Wipy/build/Wipy.map -Wl,--no-warn-rwx-segments -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld @CMakeFiles\Wipy.elf.rsp -o Wipy.elf  && cd ."
c:/users/mrntu/.espressif/dist/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(Wipy.cpp.obj):(.literal.app_main+0x0): undefined reference to `Arret_Urgence::run()'
c:/users/mrntu/.espressif/dist/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(Wipy.cpp.obj): in function `app_main':
H:/git/ESP32/Wipy/arret_urgence/include/arret_urgence.h:77: undefined reference to `Arret_Urgence::run()'
```

### Steps to reproduce.

The method run is declared in arret_urgence.cpp like this :
```c++
#include "arret_urgence.h"

void Arret_Urgence::run()
    {
        cout << "[" << ARRET_URGENCE_OBJET << "] Lancement du thread ARRET_URGENCE" << endl;
        if (this->st_au_attributs->fonctionnement != true)
        {
            cout << "[" << ARRET_URGENCE_OBJET << "] On est dans la boucle" << endl;
            vTaskDelay(1000 / portTICK_PERIOD_MS);
            this->st_au_attributs->fonctionnement = true;

            /* Create a thread on core 0 that spawns another thread, they will both have the same name etc */
            cout << "[" << ARRET_URGENCE_OBJET << "] Thread Config" << endl;
            this->st_au_attributs->cfg = esp_pthread_get_default_config();
            this->st_au_attributs->cfg.thread_name = "Thread 1";
            this->st_au_attributs->cfg.pin_to_core = (int)0;
            this->st_au_attributs->cfg.stack_size = 3 * 1024;
            this->st_au_attributs->cfg.prio = 5;

            /* Setting Button */
            cout << "[" << ARRET_URGENCE_OBJET << "] Setting buttons" << endl;
            esp_rom_gpio_pad_select_gpio(AU_BUTTON);
            this->st_au_attributs->au_button = {
                .pin_bit_mask = BIT64(AU_BUTTON),
                .mode = gpio_mode_t::GPIO_MODE_INPUT,
                .pull_up_en = gpio_pullup_t::GPIO_PULLUP_DISABLE,
                .pull_down_en = gpio_pulldown_t::GPIO_PULLDOWN_ENABLE,
                .intr_type = gpio_int_type_t::GPIO_INTR_DISABLE,
            };
            // gpio_intr_enable(AU_BUTTON);

            vTaskDelay(1000 / portTICK_PERIOD_MS);
            this->st_au_attributs->cfg.inherit_cfg = true;
            cout << "[" << ARRET_URGENCE_OBJET << "] Lancement Thread" << endl;
            esp_pthread_set_cfg(&this->st_au_attributs->cfg);
            vTaskDelay(1000 / portTICK_PERIOD_MS);

            BaseType_t res = xTaskCreatePinnedToCore(
                reinterpret_cast<TaskFunction_t>(arretUrgenceTask), // Task function
                "Thread 1",                                         // Task name
                3 * 1024,                                           // Stack size (in words)
                &this->st_au_attributs,                             // Task parameters
                5,                                                  // Priority
                &this->arretUrgenceHandle,                          // Task handle
                0                                                   // Core number (0 for core 0)
            );
            if (res != pdPASS)
            {
                cout << "Failed to create the thread" << endl;
            }
            cout << "Thread lancé et on passe à la suite\n";
        }
    }
```
This signature is present in arret_urgence.h : 
```c++
class Arret_Urgence
    {
        ST_ARRET_URGENCE_VARS* st_au_attributs = getptStruct_ArretUrgence();
        Arret_Urgence() = default;
        static void arretUrgenceTask(ST_ARRET_URGENCE_VARS* st_au_attributs);

    public:
        int data = 0;
        TaskHandle_t arretUrgenceHandle = nullptr;
        // On interdit la copie d'instance
        Arret_Urgence(const Arret_Urgence &) = delete;
        Arret_Urgence &operator=(const Arret_Urgence &) = delete;
        
        // Recuperation de l'instantiation
        static Arret_Urgence &getInstance()
        {
            cout << "Recuperation de l'instance" << endl;
            // Instantiation unique
            static Arret_Urgence arret_urgence;
            return arret_urgence;
        }
        ~Arret_Urgence() {}
        void run(void);
    };
```
And finaly my Cmakelist.txt is this : 
```Cmake
idf_component_register(SRCS "arret_urgence.cpp" 
INCLUDE_DIRS "include" "../common/include")
target_compile_options(${COMPONENT_LIB} PUBLIC -std=gnu++23)
```

### Debug Logs.

```plain
[1/6] Performing build step for 'bootloader'

[1/1] cmd.exe /C "cd /D H:\git\ESP32\Wipy\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v5.1.2/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 H:/git/ESP32/Wipy/build/bootloader/bootloader.bin"

Bootloader binary size 0x6830 bytes. 0x7d0 bytes (7%) free.

[2/4] Linking CXX executable Wipy.elf

FAILED: Wipy.elf 

cmd.exe /C "cd . && C:\Users\mrntu\.espressif\dist\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  -g -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=H:/git/ESP32/Wipy/build/Wipy.map -Wl,--no-warn-rwx-segments -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld @CMakeFiles\Wipy.elf.rsp -o Wipy.elf  && cd ."

c:/users/mrntu/.espressif/dist/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(Wipy.cpp.obj):(.literal.app_main+0x0): undefined reference to `Arret_Urgence::run()'

c:/users/mrntu/.espressif/dist/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(Wipy.cpp.obj): in function `app_main':

H:/git/ESP32/Wipy/arret_urgence/include/arret_urgence.h:77: undefined reference to `Arret_Urgence::run()'

collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

ESP-Marius, 2023-12-28:
Are you able to push a minimal project to github that can reproduce this error? 

I don't think it is a problem with `SRCS`, but there might be something else in your project layout that causes it.

ElinhoHD, 2023-12-29:
Yeah im sorry I had also to modify CMakelist.txt  in the main component

ElinhoHD, 2023-12-30:
resolved

closed 2023-12-30 by ElinhoHD