espressif/esp-idf #12879: Stack overflow for esp_hid_device

mitchellcairns, 2023-12-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The issue was incorrectly closed:
https://github.com/espressif/esp-idf/issues/9720

Here is the official code :

> esp-idf/components/esp_hid/src/ble_hidd.c line 974

```
esp_event_loop_args_t event_task_args = {
        .queue_size = 5,
        .task_name = "ble_hidd_events",
        .task_priority = uxTaskPriorityGet(NULL),
        .task_stack_size = 2048,
        .task_core_id = tskNO_AFFINITY
    };
    ret = esp_event_loop_create(&event_task_args, &s_dev->event_loop_handle);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "HID device event loop could not be created");
        ble_hidd_dev_free();
        return ret;
    }
```

The task_stack_size is static as 2048 bytes, and it should be dynamic. Either tie this to an SDK config parameter, or have it as a member of esp_hid_device_config_t as a suggestion. This needs to be accessible to change the task size otherwise it's unusable in its default state.

zhp0406, 2024-01-02:
Which version are you using?

zhp0406, 2024-01-05:
60354c3

Alvin1Zhang, 2024-01-10:
@mitchellcairns Thanks for reporting, would you please help share if any updates for the issue? Thanks.

mitchellcairns, 2024-01-13:
> @mitchellcairns This is a known issue. If you are using the ESP32S3 chip, please set the task_stack_size to 4096. For ESP32C3, setting it to 2048 is fine. I have not tested it with other chips yet. Fixed in： [60354c3](https://github.com/espressif/esp-idf/commit/60354c39a9338961adce8e2fb536f83571ebd82c)

I don't think it's a good long-term fix especially since the callback allocated by the task stack size is user-space... It needs to be an SDK config parameter in my opinion.

zhp0406, 2024-01-15:
@mitchellcairns 
Thank you for your suggestion,
we will take it into consideration.

zhp0406, 2024-01-20:
@mitchellcairns 
I believe there is no need to change the stack size of this task, as the stack size consumed by the task is fixed.
What do you think?

Alvin1Zhang, 2024-02-20:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates, thanks.

closed 2024-02-20 by Alvin1Zhang

mitchellcairns, 2024-02-21:
> @mitchellcairns I believe there is no need to change the stack size of this task, as the stack size consumed by the task is fixed. What do you think?

I disagree-- the user defines the callback function, which uses the task stack size. If the callback that is meant to be defined is only meant to hand it off to another task, the API should reflect this or offer an option to adjust the task size to meet the needs of the developer.