espressif/esp-idf #12880: ESP HID Vendor ID and Product ID do not apply in BTC mode

mitchellcairns, 2023-12-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM

### Power Supply used.

USB

### What is the expected behavior?

When setting the vendor_id and product_id members of esp_hid_device_config_t, then calling esp_hidd_dev_init, the appropriate Vendor ID and Product ID are reflected on the host device upon successful connection.

### What is the actual behavior?

The Vendor ID is always shown as 0xFFFF and the Product ID is always shown as 0x0000

### Steps to reproduce.

1. Create an example HID project (The Mouse is fine)
2. Set the SDK settings to only use BT Classic HID
3. Adjust the Vendor/Product IDs 
4. Compile and run, then pair to the device
5. Using a software such as HID Viewer on Windows, observe the reported Product ID and Vendor ID

mitchellcairns, 2023-12-27:
[image]

See the HID example code and how it does not send the appropriate VID or PID.

alex-freeaim, 2024-02-09:
I'm also getting the same problem.

IDF version: 5.1
Espressif SoC revision: ESP32
Operating System used: Windows
How did you build your project?: VS Code IDE (with PlatformIO)
If you are using Windows, please specify command line type: None
Development Kit: Adafruit Feather ESP32 V2
Power Supply used: USB

The example is here:
https://github.com/espressif/esp-idf/blob/release/v5.1/examples/bluetooth/esp_hid_device/main/esp_hid_device_main.c
(Set HID_DEV_MODE to HIDD_BT_MODE)

If there are any updates on the progress of the fix, please let us know. Thanks!

boblane1, 2024-02-18:
Hi @alex-freeaim，

@mitchellcairns has created a pull request(https://github.com/espressif/esp-idf/pull/12889) to solve this issue, but some changes still need to be made. You can refer to it for now.

closed 2024-09-14 by espressif-bot