espressif/esp-idf #12884: tx_io_num error

Takashi426, 2023-12-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF 5.1

### Espressif SoC revision.

ESP32 -WROOM-32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 -WROOM-32

### Power Supply used.

External 5V

### What is the expected behavior?

it want to print dust sensor value which is connect to GPIO 34 as RX and GPIO 35 as TX 

### What is the actual behavior?

E (335) uart: uart_set_pin(645): tx_io_num error

### Steps to reproduce.

1. Step
2. Step
3. Step
...
#include "driver/uart.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include <stdio.h>

#define FW_VERSION "AAS_LDS_UNO_v1.3"
#define SAMPLE_RATE 2000
#define SENSOR_BAUD 9600
#define uart UART_NUM_1
#define TXD_PIN 35 // Define your ESP32 TX pin
#define RXD_PIN 34 // Define your ESP32 RX pin

int PM1value;
int PM25value;
int PM10value;

static const int RX_BUF_SIZE = 1024;

int checkDustSensor();
void getDustValue();
void dust_sensor_task(void *pvParameters);

void app_main() {
  uart_config_t uart_config = {.baud_rate = 115200,
                               .data_bits = UART_DATA_8_BITS,
                               .parity = UART_PARITY_DISABLE,
                               .stop_bits = UART_STOP_BITS_1,
                               .flow_ctrl = UART_HW_FLOWCTRL_DISABLE};

  uart_param_config(uart, &uart_config);
  uart_set_pin(uart, TXD_PIN, RXD_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
  uart_driver_install(uart, RX_BUF_SIZE, 0, 0, NULL, 0);

  printf("Amphenol Advanced Sensors\n");
  printf("%s\n", FW_VERSION);

  if (checkDustSensor()) {
    printf("Dust Sensor Detected\n");
  } else {
    printf("Dust Sensor not detected. Please check connections.\n");
  }

  xTaskCreate(dust_sensor_task, "dust_sensor_task", 4096, NULL, 5, NULL);
}

int checkDustSensor() {
  printf("check dust sensor enterf");
  const char checkCmd[] =
      "BM\xe1\x00\x01\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
      "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00";
  uart_write_bytes(uart, checkCmd, sizeof(checkCmd));
  vTaskDelay(10 / portTICK_PERIOD_MS);

  uint8_t data;
  uart_get_buffered_data_len(uart, &data);
  return data > 0;
}

void getDustValue() {
  printf("get dust sensor entered ");
  uint8_t data[32];
  uint8_t message[64];
  int CRC = 0;

  uart_read_bytes(uart, message, 64, portMAX_DELAY);

  for (uint8_t i = 0; i < 32; i++) {
    if (message[i] == 0x42 && message[i + 1] == 0x4D) {
      for (uint8_t j = 0; j < 32; j++) {
        data[j] = message[i];
        i++;
      }
      break;
    }
  }

  for (int i = 0; i < 30; i++) {
    CRC += data[i];
  }

  if ((data[30] * 256 + data[31]) == CRC) {
    PM1value = (data[4] & 0x3F) << 8 | data[5];
    PM25value = (data[6] & 0x3F) << 8 | data[7];
    PM10value = (data[8] & 0x3F) << 8 | data[9];
  } else {
    PM25value = 999;
  }
}

void dust_sensor_task(void *pvParameters) {
  while (1) {
    printf("while loop entered");
    // if (millis() >= SAMPLE_RATE) {
    getDustValue();
    printf(",PMA1: %d, PMA25: %d, PMA10: %d\n", PM1value, PM25value, PM10value);

    // Reset the timer
    // SAMPLE_RATE = millis() + 2000;
    // }
    // vTaskDelay(  100 / portTICK_PERIOD_MS); // Adjust delay according to your
    // requirement
  }
}

### Debug Logs.

```plain
ELF file SHA256: 59b7e52ed82d4e09

E (388) esp_apptrace: Trace destination 1 not supported!
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7084
ho 0 tail 12 room 4
load:0x40078000,len:15584
load:0x40080400,len:4
.0x40080400: _init at ??:?

load:0x40080404,len:3876
entry 0x4008064c
I (31) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader
I (31) boot: compile time Dec 28 2023 09:38:19
I (31) boot: Multicore bootloader
I (36) boot: chip revision: v3.0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0a858h ( 43096) map
I (120) esp_image: segment 1: paddr=0001a880 vaddr=3ffb0000 size=02664h (  9828) load
I (124) esp_image: segment 2: paddr=0001ceec vaddr=40080000 size=0312ch ( 12588) load
I (131) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=1694ch ( 92492) map
I (168) esp_image: segment 4: paddr=00036974 vaddr=4008312c size=0a42ch ( 42028) load
I (192) boot: Loaded app from partition at offset 0x10000
I (193) boot: Disabling RNG early entropy source...
I (204) cpu_start: Multicore app
I (204) cpu_start: Pro cpu up.
I (204) cpu_start: Starting app cpu, entry point is 0x40081260
0x40081260: call_start_cpu1 at C:/Users/Rehman/esp/esp-idf/components/esp_system/port/cpu_start.c:154

I (192) cpu_start: App cpu up.
I (222) cpu_start: Pro cpu start user code
I (223) cpu_start: cpu freq: 160000000 Hz
I (223) cpu_start: Application information:
I (227) cpu_start: Project name:     ethernet_basic
I (233) cpu_start: App version:      1
I (237) cpu_start: Compile time:     Dec 28 2023 10:03:51
I (243) cpu_start: ELF file SHA256:  59b7e52ed82d4e09...
I (249) cpu_start: ESP-IDF:          v5.1.1-dirty
I (255) cpu_start: Min chip rev:     v0.0
I (259) cpu_start: Max chip rev:     v3.99
I (264) cpu_start: Chip rev:         v3.0
I (269) heap_init: Initializing. RAM available for dynamic allocation:
I (276) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (282) heap_init: At 3FFB2F30 len 0002D0D0 (180 KiB): DRAM
I (288) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (295) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (301) heap_init: At 4008D558 len 00012AA8 (74 KiB): IRAM
I (309) spi_flash: detected chip: generic
I (312) spi_flash: flash io: dio
W (316) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (329) esp_apptrace_uart: UART baud rate: 115200
E (335) uart: uart_set_pin(645): tx_io_num error
E (340) esp_apptrace: Trace destination 1 not supported!

assert failed: esp_apptrace_uart_init port_uart.c:237 ((err == ESP_OK) && "Not possible to configure UART RX/TX pins. Please check and change menuconfig parameters!")

Backtrace: 0x4008189e:0x3ffe3990 0x40086ba9:0x3ffe39b0 0x4008c1dd:0x3ffe39d0 0x40084c1b:0x3ffe3af0 0x400847a5:0x3ffe3b40 0x400847db:0x3ffe3b70 0x400d1960:0x3ffe3b90 0x400d19a2:0x3ffe3bc0 0x400d1b6b:0x3ffe3bf0 0x40081576:0x3ffe3c30 0x400796cd:0x3ffe3c80 |<-CORRUPTED
0x4008189e: panic_abort at C:/Users/Rehman/esp/esp-idf/components/esp_system/panic.c:452

0x40086ba9: esp_system_abort at C:/Users/Rehman/esp/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x4008c1dd: __assert_func at C:/Users/Rehman/esp/esp-idf/components/newlib/assert.c:81

0x40084c1b: esp_apptrace_uart_init at C:/Users/Rehman/esp/esp-idf/components/app_trace/port/port_uart.c:237 (discriminator 1)

0x400847a5: esp_apptrace_init at C:/Users/Rehman/esp/esp-idf/components/app_trace/app_trace.c:69

0x400847db: __esp_system_init_fn_esp_apptrace_init at C:/Users/Rehman/esp/esp-idf/components/app_trace/app_trace.c:82

0x400d1960: do_system_init_fn at C:/Users/Rehman/esp/esp-idf/components/esp_system/startup.c:223 (discriminator 15)

0x400d19a2: do_secondary_init at C:/Users/Rehman/esp/esp-idf/components/esp_system/startup.c:406

0x400d1b6b: start_cpu0_default at C:/Users/Rehman/esp/esp-idf/components/esp_system/startup.c:467 (discriminator 3)

0x40081576: call_start_cpu0 at C:/Users/Rehman/esp/esp-idf/components/esp_system/port/cpu_start.c:689
```

nopnop2002, 2023-12-28:
```
//uart_param_config(uart, &uart_config);
//uart_set_pin(uart, TXD_PIN, RXD_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
//uart_driver_install(uart, RX_BUF_SIZE, 0, 0, NULL, 0);

uart_driver_install(uart, RX_BUF_SIZE, 0, 0, NULL, 0);
uart_param_config(uart, &uart_config);
uart_set_pin(uart, TXD_PIN, RXD_PIN, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
```

Takashi426, 2024-01-31:
solved these issue

closed 2024-01-31 by Takashi426