espressif/esp-idf #12885: Compile error if set MBEDTLS_SSL_DTLS_CONNECTION_ID=y

AxelLin, 2023-12-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2-498-gc57b352725

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Should not hit compile error.

### What is the actual behavior?

Compile error if set MBEDTLS_SSL_DTLS_CONNECTION_ID=y

/home/axel/esp/esp-idf/components/mbedtls/mbedtls/include/mbedtls/check_config.h:1184:2: error: #error "MBEDTLS_SSL_CID_PADDING_GRANULARITY was removed in Mbed TLS 3.0. See https://github.com/Mbed-TLS/mbedtls/issues/4335"
 1184 | #error "MBEDTLS_SSL_CID_PADDING_GRANULARITY was removed in Mbed TLS 3.0. See https://github.com/Mbed-TLS/mbedtls/issues/4335"

### Steps to reproduce.

Set below configuration, then I got compile error.
mbedTLS -> mbedTLS v3.x related -> DTLS-based configurations -> Support for the DTLS Connection ID extension

mahavirj, 2024-01-04:
This was fixed through https://github.com/espressif/esp-idf/pull/12177. Fix has been backported till 5.0 branch, it should appear on GH with the next sync.

AxelLin, 2024-01-06:
Fixed in v5.1: f9569bde37d8d18b004af8e61bb0215492a9e129

closed 2024-01-06 by AxelLin