espressif/esp-idf #12886: esp_ble_resolve_adv_data Trigger crash

maenkai, 2023-12-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3.1

### Espressif SoC revision.

ESP32D-WROVE

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32D-WROVE

### Power Supply used.

USB

### What is the expected behavior?

The scan adv is normal

### What is the actual behavior?

Trigger crash

### Steps to reproduce.

Run on standby for a period of time

### Debug Logs.

```plain
16:22:56.292  Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
16:22:56.293  
16:22:56.295  Core  0 register dump:
16:22:56.302  PC      : 0x400ff194  PS      : 0x00060330  A0      : 0x800f6411  A1      : 0x3ffe22b0  
16:22:56.328  A2      : 0x3fffffcc  A3      : 0x000000ff  A4      : 0x3ffe233f  A5      : 0x3fffffe1  
16:22:56.329  A6      : 0x3ffe2380  A7      : 0x3ffe2340  A8      : 0x4000002d  A9      : 0x0000004c  
16:22:56.340  A10     : 0x000000ce  A11     : 0x0000003e  A12     : 0x00060523  A13     : 0x00060523  
16:22:56.341  A14     : 0x00000001  A15     : 0x0000001a  SAR     : 0x0000001d  EXCCAUSE: 0x00000003  
16:22:56.343  EXCVADDR: 0x4000002d  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
16:22:56.366  
16:22:56.367  Backtrace:0x400ff191:0x3ffe22b0 0x400f640e:0x3ffe22d0 0x400e94b6:0x3ffe2300 0x400e9dbe:0x3ffe23a0 0x4014bf2d:0x3ffe23c0 0x4013bd2b:0x3ffe23f0 0x4013d993:0x3ffe2410
16:22:56.409  
16:22:56.409  
16:22:56.409  ELF file SHA256: ff826356d3b860d2
16:22:56.410  
16:22:56.410  Rebooting...
0x400ff194: BTM_CheckAdvData at /home/mek/esp/esp-idf-v4-3-1/components/bt/host/bluedroid/stack/btm/btm_ble_gap.c:2081
0x400f640e: esp_ble_resolve_adv_data at /home/mek/esp/esp-idf-v4-3-1/components/bt/host/bluedroid/api/esp_gap_ble_api.c:420
0x400e94b6: mz_handle_scan_result at /home/mek/Music/Lipro/common/bt/mz_ble_central.c:552
0x400e9dbe: mz_central_ble_adv_callback at /home/mek/Music/Lipro/common/bt/mz_ble_central.c:1210
 (inlined by) mz_central_ble_adv_callback at /home/mek/Music/Lipro/common/bt/mz_ble_central.c:1207
0x4014bf2d: btc_ble_mesh_ble_cb_to_app at /home/mek/esp/esp-idf-v4-3-1/components/bt/esp_ble_mesh/btc/btc_ble_mesh_ble.c:179
 (inlined by) btc_ble_mesh_ble_cb_handler at /home/mek/esp/esp-idf-v4-3-1/components/bt/esp_ble_mesh/btc/btc_ble_mesh_ble.c:195
0x4013bd2b: btc_thread_handler at /home/mek/esp/esp-idf-v4-3-1/components/bt/common/btc/core/btc_task.c:184
0x4013d993: osi_thread_run at /home/mek/esp/esp-idf-v4-3-1/components/bt/common/osi/thread.c:68
```

maenkai, 2023-12-28:
I tried to fix it this way，I wonder if this problem can be solved
```c
    while ( length && (p - p_adv <= BTM_BLE_CACHE_ADV_DATA_MAX)) {
        STREAM_TO_UINT8(adv_type, p);

        if ( adv_type == type ) {
            /* length doesn't include itself */
            *p_length = length - 1; /* minus the length of type */
            return p;
        }
        p += length - 1; /* skip the length of data */

        //mz_fix_start 
        //广播里的length字段出现错误，可能是对端发了一个错误的广播或者其它原因导致数据被污染，
        //这里判断一下防止内存越界
        if(p >= p_adv + BTM_BLE_CACHE_ADV_DATA_MAX){
          break;
        }
        //mz_fix_end

        STREAM_TO_UINT8(length, p);
    }

maenkai, 2023-12-29:
When BTM_BLE_CACHE_ADV_DATA_MAX is greater than p_length, two wrong types and lengths will be accessed illegally，
e.g：“ b6 fb ”
```c
09:42:33.475  E (13208) BT_BTM: 182 adv invalid len.
09:42:33.480  I (00:00:12.463) BT_BTM: addr = 5c:cd:7c:57:c0:61
09:42:33.491  I (00:00:12.463) BT_BTM: 02 01 06 03 03 27 18 15 16 27 18 03 ab 5c cd 7c 
09:42:33.497  I (00:00:12.470) BT_BTM: 57 c0 61 09 88 18 00 05 05 12 40 00 00 b6 fb

zhp0406, 2024-01-03:
Could you please try version 4.4 or later to see if there are any issues?

maenkai, 2024-01-03:
> Could you please try version 4.4 or later to see if there are any issues?

Can you provide relevant patches for repair? Because my product has been released based on version 4.3.1, I cannot change the version

zhp0406, 2024-01-03:
@maenkai 
ok,
how to reproduce your issue?

maenkai, 2024-01-03:
It's not duplicated in the actual test, it's code
```c
//handle
static void handle_scan_result(uint8_t *ble_data, uint16_t length, uint8_t addr[6], uint8_t addr_type, int8_t rssi, bool from_mesh) {
  uint16_t company_id = 0;
  uint32_t device_type = 0xffffffff;
  uint8_t *manuf_data = NULL;
  uint8_t manuf_data_len = 0;

  manuf_data = esp_ble_resolve_adv_data(ble_data,
                                        ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE,
                                        &manuf_data_len);
  if (manuf_data == NULL) {
    return;
  }
  //doing..............
}

//from gap evetn
    case ESP_GAP_BLE_SCAN_RESULT_EVT: {
      esp_ble_gap_cb_param_t *scan_result = (esp_ble_gap_cb_param_t *)param;
      switch (scan_result->scan_rst.search_evt) {
        case ESP_GAP_SEARCH_INQ_RES_EVT:
          mz_handle_scan_result(scan_result->scan_rst.ble_adv,
                                scan_result->scan_rst.adv_data_len,
                                scan_result->scan_rst.bda,
                                scan_result->scan_rst.ble_addr_type,
                                scan_result->scan_rst.rssi,
                                false);
          break;
        default:
          break;
      }
      break;
    }

//from mesh event
  switch (event) {
  case ESP_BLE_MESH_SCAN_BLE_ADVERTISING_PKT_EVT: {
    mz_handle_scan_result(param->scan_ble_adv_pkt.data,
                          param->scan_ble_adv_pkt.length,
                          param->scan_ble_adv_pkt.addr,
                          param->scan_ble_adv_pkt.addr_type,
                          param->scan_ble_adv_pkt.rssi,
                          true);
    break;
  }

maenkai, 2024-01-03:
> @maenkai ok, how to reproduce your issue?

It's not duplicated in the actual test, it's code
```c
//handle
static void handle_scan_result(uint8_t *ble_data, uint16_t length, uint8_t addr[6], uint8_t addr_type, int8_t rssi, bool from_mesh) {
  uint16_t company_id = 0;
  uint32_t device_type = 0xffffffff;
  uint8_t *manuf_data = NULL;
  uint8_t manuf_data_len = 0;

  manuf_data = esp_ble_resolve_adv_data(ble_data,
                                        ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE,
                                        &manuf_data_len);
  if (manuf_data == NULL) {
    return;
  }
  //doing..............
}

//from gap evetn
    case ESP_GAP_BLE_SCAN_RESULT_EVT: {
      esp_ble_gap_cb_param_t *scan_result = (esp_ble_gap_cb_param_t *)param;
      switch (scan_result->scan_rst.search_evt) {
        case ESP_GAP_SEARCH_INQ_RES_EVT:
          mz_handle_scan_result(scan_result->scan_rst.ble_adv,
                                scan_result->scan_rst.adv_data_len,
                                scan_result->scan_rst.bda,
                                scan_result->scan_rst.ble_addr_type,
                                scan_result->scan_rst.rssi,
                                false);
          break;
        default:
          break;
      }
      break;
    }

//from mesh event
  switch (event) {
  case ESP_BLE_MESH_SCAN_BLE_ADVERTISING_PKT_EVT: {
    mz_handle_scan_result(param->scan_ble_adv_pkt.data,
                          param->scan_ble_adv_pkt.length,
                          param->scan_ble_adv_pkt.addr,
                          param->scan_ble_adv_pkt.addr_type,
                          param->scan_ble_adv_pkt.rssi,
                          true);
    break;
  }

zhp0406, 2024-01-03:
Can you print the adv data before the crash?

```
    case ESP_GAP_BLE_SCAN_RESULT_EVT: {
        esp_ble_gap_cb_param_t *scan_result = (esp_ble_gap_cb_param_t *)param;
        switch (scan_result->scan_rst.search_evt) {
        case ESP_GAP_SEARCH_INQ_RES_EVT:
            esp_log_buffer_hex(GATTC_TAG, scan_result->scan_rst.bda, 6);
            ESP_LOGI(GATTC_TAG, "searched Adv Data Len %d, Scan Response Len %d", scan_result->scan_rst.adv_data_len, scan_result->scan_rst.scan_rsp_len);
            **esp_log_buffer_hex("adv_rsp", scan_result->scan_rst.ble_adv, scan_result->scan_rst.adv_data_len);**
            adv_name = esp_ble_resolve_adv_data(scan_result->scan_rst.ble_adv,
                                                ESP_BLE_AD_TYPE_NAME_CMPL, &adv_name_len);
            ESP_LOGI(GATTC_TAG, "searched Device Name Len %d", adv_name_len);
            esp_log_buffer_char(GATTC_TAG, adv_name, adv_name_len);

```

maenkai, 2024-01-03:
> Can you print the adv data before the crash? #include "esp_log.h" **esp_log_buffer_hex("adv_rsp", scan_result->scan_rst.ble_adv, scan_result->scan_rst.adv_data_len);**

OK，I need time to recreate

zhp0406, 2024-01-03:
According to your description, there might be a crash issue when encountering specific adv data. 
However, based on the logic of the code, I haven't identified any apparent problems.
I'm trying to construct specific ADV data to reproduce the issue on my end, but it hasn't occurred.such as :

```
uint8_t adv_data[] = {0x02,0x01,0x06,
0x03,0x03,0x27,0x18,
0x15,0x16,0x27,0x18,0x03,0xab,0x5c,0xcd,0x7c,0x57,0xc0,0x61,0x09,0x88,0x18,0x00,0x05,0x05,0x12,0x40,0x00,0x00,
0xb6,ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE};
adv_name = esp_ble_resolve_adv_data(adv_data,
ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE, &adv_name_len);
```

Therefore, if you can provide ADV data that may lead to a crash, it would be very helpful.

maenkai, 2024-01-03:
> According to your description, there might be a crash issue when encountering specific adv data. However, based on the logic of the code, I haven't identified any apparent problems. I'm trying to [...]

I didn't capture the broadcast that caused the crash in the real world.
But here are the data I simulated that can cause a crash
```
  uint8_t test_adv[ESP_BLE_ADV_DATA_LEN_MAX + ESP_BLE_SCAN_RSP_DATA_LEN_MAX] = {
    0x02, 0x01, 0x06,
    0x03, 0x03, 0x27, 0x18,
    0x15, 0x16, 0x27, 0x18, 0x03, 0xab, 0x5c, 0xcd, 0x7c, 0x57, 0xc0, 0x61, 0x09, 0x88, 0x18, 0x00, 0x05, 0x05, 0x12, 0x40, 0x00, 0x00,
    0xff, 0xfb
  };

  manuf_data = esp_ble_resolve_adv_data(test_adv,
                                        ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE,
                                        29);

zhp0406, 2024-01-03:
param[**out**]      length - return the length of ADV data not including type
/**
 * @brief          This function is called to get ADV data for a specific type.
 *
 * @param[in]       adv_data - pointer of ADV data which to be resolved
 * @param[in]       type   - finding ADV data type
 * @param[out]      length - return the length of ADV data not including type
 *
 * @return          pointer of ADV data
 *
 */
uint8_t *esp_ble_resolve_adv_data(uint8_t *adv_data, uint8_t type, uint8_t *length);

maenkai, 2024-01-03:
> param[**out**] length - 返回不包括类型的 ADV 数据的长度 /**
> 
> * 调用此函数以获取特定类型的 ADV 数据。
> * [in] adv_data - 要解析的 ADV 数据的指针
> * [in] type - 查找 ADV 数据类型
> * [out] length - 返回不包括类型的 ADV 数据的长度
> * ADV 数据指针
> 
> */uint8_t *esp_ble_resolve_adv_data（uint8_t *adv_data，uint8_t型，uint8_t*长度）; ![图像](https://private-user-images.githubusercontent.com/131946088/293871050-bb740ee4-54a0-430f-b29d-31d7b63720ec.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQyNzMwOTIsIm5iZiI6MTcwNDI3Mjc5MiwicGF0aCI6Ii8xMzE5NDYwODgvMjkzODcxMDUwLWJiNzQwZWU0LTU0YTAtNDMwZi1iMjlkLTMxZDdiNjM3MjBlYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwM1QwOTA2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMDE4YjIyMjEzZGM4NmUwNzRiNjFlZTgwMDkzOTQ4MjEyOGEyMWU0Yjg0MjBjNjczN2U5MzFiMWYxY2FjNzU3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.J78gU3c5Mu_IiYatQnru6RkkdcA3du-dW8VdA1x0IRo)

Sorry for writing the wrong demo, after modifying it correctly, you can see that the pointer p still accesses addresses that are out of range, which is a serious problem.
At the same time, I'm confused why doesn't it trigger a crash
```c
  /**DEMO**/
  // ESP_BLE_ADV_DATA_LEN_MAX = 31 ESP_BLE_SCAN_RSP_DATA_LEN_MAX = 31
  uint8_t test_adv[ESP_BLE_ADV_DATA_LEN_MAX + ESP_BLE_SCAN_RSP_DATA_LEN_MAX] = {
    0x02, 0x01, 0x06,
    0x03, 0x03, 0x27, 0x18,
    0x15, 0x16, 0x27, 0x18, 0x03, 0xab, 0x5c, 0xcd, 0x7c, 0x57, 0xc0, 0x61, 0x09, 0x88, 0x18, 0x00, 0x05, 0x05, 0x12, 0x40, 0x00, 0x00,
    0xff, 0xfb
  };

  uint8_t manuf_data_len = 29;

  esp_ble_resolve_adv_data(test_adv,
                           ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE,
                           &manuf_data_len);

UINT8 *BTM_CheckAdvData( UINT8 *p_adv, UINT8 type, UINT8 *p_length)
{
    UINT8 *p = p_adv;
    UINT8 length;
    UINT8 adv_type;
    BTM_TRACE_API("BTM_CheckAdvData type=0x%02X", type);

    STREAM_TO_UINT8(length, p);

    while ( length && (p - p_adv <= BTM_BLE_CACHE_ADV_DATA_MAX)) {
        STREAM_TO_UINT8(adv_type, p);

        if ( adv_type == type ) {
            /* length doesn't include itself */
            *p_length = length - 1; /* minus the length of type */
            return p;
        }
        p += length - 1; /* skip the length of data */
        printf("length is %d p = %p end = %p\r\n",length, p , p_adv + BTM_BLE_CACHE_ADV_DATA_MAX);
        STREAM_TO_UINT8(length, p);
    }

    *p_length = 0;
    return NULL;
}
```

LOG
```c
17:17:03.788  length is 2 p = 0x3ffbcdf3 end = 0x3ffbce2e
17:17:03.794  length is 3 p = 0x3ffbcdf7 end = 0x3ffbce2e
17:17:03.800  length is 21 p = 0x3ffbce0d end = 0x3ffbce2e
17:17:03.802  length is 255 p = 0x3ffbcf0d end = 0x3ffbce2e

```

maenkai, 2024-01-03:
In the mesh scan event callback, how to ensure that the data range is greater than or equal to BTM_BLE_CACHE_ADV_DATA_MAX?
Maybe that's where the problem lies
- esp code
```c
    struct {
        uint8_t  addr[6];   /*!< Device address */
        uint8_t  addr_type; /*!< Device address type */
        uint8_t  adv_type;  /*!< Advertising data type */
        uint8_t *data;      /*!< Advertising data */
        uint16_t length;    /*!< Advertising data length */
        int8_t   rssi;      /*!< RSSI of the advertising packet */
    } scan_ble_adv_pkt;     /*!< Event parameters of ESP_BLE_MESH_SCAN_BLE_ADVERTISING_PKT_EVT */
} esp_ble_mesh_ble_cb_param_t;
```

-  demo
```c
  switch (event) {
  case ESP_BLE_MESH_SCAN_BLE_ADVERTISING_PKT_EVT: {
    handle_scan_result(param->scan_ble_adv_pkt.data,
                          param->scan_ble_adv_pkt.length,
                          param->scan_ble_adv_pkt.addr,
                          param->scan_ble_adv_pkt.addr_type,
                          param->scan_ble_adv_pkt.rssi,
                          true);
    break;
  }

```

maenkai, 2024-01-03:
> param[**out**] length - return the length of ADV data not including type /**
>
> * @brief          This function is called to get ADV data for a specific type.
> * @param[in]       adv_data - pointer of [...]

Sorry for writing the wrong demo, after modifying it correctly, you can see that the pointer p still accesses addresses that are out of range, which is a serious problem.
At the same time, I'm confused why doesn't it trigger a crash

- demo
```c
  // ESP_BLE_ADV_DATA_LEN_MAX = 31 ESP_BLE_SCAN_RSP_DATA_LEN_MAX = 31
  uint8_t test_adv[ESP_BLE_ADV_DATA_LEN_MAX + ESP_BLE_SCAN_RSP_DATA_LEN_MAX] = {
    0x02, 0x01, 0x06,
    0x03, 0x03, 0x27, 0x18,
    0x15, 0x16, 0x27, 0x18, 0x03, 0xab, 0x5c, 0xcd, 0x7c, 0x57, 0xc0, 0x61, 0x09, 0x88, 0x18, 0x00, 0x05, 0x05, 0x12, 0x40, 0x00, 0x00,
    0xff, 0xfb
  };

  uint8_t manuf_data_len = 29;

  esp_ble_resolve_adv_data(test_adv,
                           ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE,
                           &manuf_data_len);

UINT8 *BTM_CheckAdvData( UINT8 *p_adv, UINT8 type, UINT8 *p_length)
{
    UINT8 *p = p_adv;
    UINT8 length;
    UINT8 adv_type;
    BTM_TRACE_API("BTM_CheckAdvData type=0x%02X", type);

    STREAM_TO_UINT8(length, p);

    while ( length && (p - p_adv <= BTM_BLE_CACHE_ADV_DATA_MAX)) {
        STREAM_TO_UINT8(adv_type, p);

        if ( adv_type == type ) {
            /* length doesn't include itself */
            *p_length = length - 1; /* minus the length of type */
            return p;
        }
        p += length - 1; /* skip the length of data */
        printf("length is %d p = %p end = %p\r\n",length, p , p_adv + BTM_BLE_CACHE_ADV_DATA_MAX);
        STREAM_TO_UINT8(length, p);
    }

    *p_length = 0;
    return NULL;
}
```

- LOG
```c
17:17:03.788  length is 2 p = 0x3ffbcdf3 end = 0x3ffbce2e
17:17:03.794  length is 3 p = 0x3ffbcdf7 end = 0x3ffbce2e
17:17:03.800  length is 21 p = 0x3ffbce0d end = 0x3ffbce2e
17:17:03.802  length is 255 p = 0x3ffbcf0d end = 0x3ffbce2e

```

zhp0406, 2024-01-03:
@maenkai 

[image]

maenkai, 2024-01-03:
> @maenkai With the added conditional statement, will there still be a possibility of a crash? ![image](https://private-user-images.githubusercontent.com/131946088/293878856-866c7870-a206-4d8b-9c85-9cab61883d08.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQyNzU2MTUsIm5iZiI6MTcwNDI3NTMxNSwicGF0aCI6Ii8xMzE5NDYwODgvMjkzODc4ODU2LTg2NmM3ODcwLWEyMDYtNGQ4Yi05Yzg1LTljYWI2MTg4M2QwOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwM1QwOTQ4MzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05MTBiYzIyMGQ5MWFmNWZhMzdjMmI5YzQ2N2ZmMGUzNjUzOTRhNjk1NTFjMTA0NDUyOGY1MzFkNzZjMjc5ODM4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.JvwS0XUPu4Vper-Ic_vVEqiZoE1pcDq_TuWtyM57SC0)

I need to test it for a long time and get back to you with the results

zhp0406, 2024-01-10:
Hi,
I believe this is a bug, and I will fix this issue in the upcoming versions.
thanks for report the issue.
commit:c66fc143

closed 2024-01-11 by maenkai