espressif/esp-idf #12888: Is it possible to stop external 40MHz crystal oscillator from running?

F1RST1, 2023-12-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The chip I am using is esp32s3. 
I've checked the idf-framework and technical reference, and I found a register named SYSTEM_SOC_CLK_SEL. SYSTEM_SOC_CLK_SEL is XTAL_CLK by default. So would external 40MHz crystal oscillator stop running if I change this register to RC_FAST_CLK or PLL_CLK? 
[image]
[image]

songruo, 2023-12-29:
Hi @F1RST1, the external XTAL won't stop running even if SYSTEM_SOC_CLK_SEL is set to PLL or RC_FAST. May I ask what are you trying to achieve, and then maybe we can provide some ideas.

F1RST1, 2023-12-29:
> Hi @F1RST1, the external XTAL won't stop running even if SYSTEM_SOC_CLK_SEL is set to PLL or RC_FAST. May I ask what are you trying to achieve, and then maybe we can provide some ideas.

Sure. The project is RGB lcd display driven by esp32s3. The hardware enginner is trying to pass the EMC test. 
Some critical reports are shown below. Any extra information could I provide to you? 
[image]
[image]
[image]

closed 2024-01-11 by F1RST1