espressif/esp-idf #12890: HA_on_off_light example not working on ESP32-H2

Liniik, 2023-12-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-H2 (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-H2-DEVKITM-1-N4

### Power Supply used.

USB

### What is the expected behavior?

Proper initialization and working device. I expect log like in the example's README like this:
```
I (901) ESP_ZB_LIGHT: Zigbee stack initialized
I (901) ESP_ZB_LIGHT: Start network steering
I (2611) ESP_ZB_LIGHT: Joined network successfully (Extended PAN ID: aa:98:48:01:a0:03:f7:84, PAN ID: 0x0e8b)
```
Then I'd expect to be able to add it to my Home Assistant instance and use it.

### What is the actual behavior?

This log - last three lines are the problem - ESP_FAIL during init and not working example.
```
SPIWP:0xee
mode:DIO, clock div:1
load:0x4083cfd0,len:0xcd4
load:0x4083efd0,len:0x2da0
load:0x408460e8,len:0x1778
entry 0x4083cfd0
I (23) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (23) boot: compile time Dec 28 2023 21:48:24
I (25) boot: chip revision: v0.1
I (26) boot.esp32h2: SPI Speed      : 64MHz
I (31) boot.esp32h2: SPI Mode       : DIO
I (35) boot.esp32h2: SPI Flash Size : 2MB
I (40) boot: Enabling RNG early entropy source...
I (45) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 000a2800
I (79) boot:  3 zb_storage       Unknown data     01 81 000b3000 00004000
I (86) boot:  4 zb_fct           Unknown data     01 81 000b7000 00000400
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=42068020 size=10d30h ( 68912) map
[... 210 lines cut ...]
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            },
            "compileCommands": "${workspaceFolder}/build/compile_commands.json"
        }
    ],
    "version": 4
}
 
--------------------------------------------------------------------------------------------------------------------------------------------
```

nopnop2002, 2024-01-13:
Before flash it to the board, it is recommended to erase NVRAM if user doesn't want to keep the previous examples or other projects stored info using ```idf.py -p PORT erase-flash```

Liniik, 2024-01-13:
Of course I did that... The result is the same. I have two devkits, erased both, tried on both, nothing.

nopnop2002, 2024-01-13:
This is my logging.
phy_version is different from yours.
```
I (410) main_task: Started on CPU0
I (410) main_task: Calling app_main()
W (420) rmt: channel resolution loss, real=10666666
I (420) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (450) phy: phy_version: 220,2, 37a29de, Dec 29 2023, 16:30:13
I (450) phy: libbtbb version: 944f18e, Dec 29 2023, 16:30:30
I (460) main_task: Returned from app_main()
I (470) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (470) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
I (470) ESP_ZB_ON_OFF_LIGHT: Start network steering
I (41150) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
I (82840) ESP_ZB_ON_OFF_LIGHT: Network steering was not successful (status: ESP_FAIL)
```

Liniik, 2024-01-13:
What is "phy"? How can I update it? I installed everything via VS Code per instructions and the esp-idf extension shows as up to date. But you still have `ESP_FAIL` at the end.

nopnop2002, 2024-01-13:
> the esp-idf extension shows as up to date. 

```
I (23) boot: ESP-IDF v5.1.2 2nd stage bootloader
```

Your esp-idf is not latest.
Latest is:
```
I (23) boot: ESP-IDF v5.1.2-602-gdb1e54a0c5-dirty 2nd stage bootloader
```

Liniik, 2024-01-13:
As you can see, in the editor I can't update it because it shows as latest version. Last update on the market is 13. 12. 2023. Is there any settings where I could update it more?
[image]

nopnop2002, 2024-01-13:
>Is there any settings where I could update it more?

git clone -b release/v5.1 --recursive https://github.com/espressif/esp-idf.git

nopnop2002, 2024-01-14:
Looks similar to this.
https://github.com/espressif/esp-idf/issues/12657

xieqinan, 2024-02-29:
@Liniik 

Would you be able to utilize two chips to run the light and switch examples provided [here](https://github.com/espressif/esp-idf/tree/master/examples/zigbee/light_sample) initially?

Alvin1Zhang, 2024-10-10:
@Liniik Thanks for reporting, would you please help share if any updates for the issue? Thanks.

Liniik, 2024-10-28:
@Alvin1Zhang Yes, but I don't know when. I completely abandoned the idea of direct development with any esp board given that no help came for an issue completely blocking whole progress. So I need to find the boards and get the environment updated and running again.

chshu, 2025-02-26:
@Liniik please reopen if any follow up questions.

closed 2025-02-26 by chshu