espressif/esp-idf #12891: Using developer mode encryption, error invalid header: 0xa3d1034b is reported

LHYYSY, 2023-12-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32 C3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-13

### Power Supply used.

USB

### What is the expected behavior?

After the encryption is successful, the operation succeeds.

### What is the actual behavior?

invalid header: 0xa3d1034b
invalid header: 0xa3d1034b
invalid header: 0xa3d1034b
invalid header: 0xa3d1034b
invalid header: 0xa3d1034b
invalid header: 0xa3d1034b

### Steps to reproduce.

[config.txt](https://github.com/espressif/esp-idf/files/13790659/config.txt)

### Debug Logs.

```plain
invalid header�ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
invalid header: 0xa3d1034b
```

AdityaHPatwardhan, 2024-01-02:
Hi @LHYYSY 
Can you share the eFuse summary for your chip?
It can be obtained through below command
`espefuse.py -p Serial_Port summary`

In the commands that you have shared,
I dont see the command that burns the SPI_BOOT_CRYPT_CNT value.

Also,
Can you please follow [these guidelines](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/security/host-based-security-workflows.html#enable-flash-encryption-externally) instead of the commands that you have shared and check once if something is missing in your case?

Thanks,
Aditya

LHYYSY, 2024-01-03:
> Hi @LHYYSY Can you share the eFuse summary for your chip? It can be obtained through below command `espefuse.py -p Serial_Port summary`
> 
> In the commands that you have shared, I dont see the command that burns the SPI_BOOT_CRYPT_CNT value.
> 
> Also, Can you please follow [these guidelines](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/security/host-based-security-workflows.html#enable-flash-encryption-externally) to enable the flash encryption externally instead of the commands that you have shared and check once if something is missing in your case?
> 
> Thanks, Aditya
Thank you very much for your reply. I have already set the encryption, but there is a warning when running the firmware. The following is the log:

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fcd5988,len:0x2bac
load:0x403cc710,len:0x968
load:0x403ce710,len:0x4aa0
entry 0x403cc710
I (37) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader
I (37) boot: compile time Jan  2 2024 13:50:59
I (37) boot: chip revision: v0.3
I (41) boot.esp32c3: SPI Speed      : 80MHz
I (46) boot.esp32c3: SPI Mode       : DOUT
I (50) boot.esp32c3: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 0000f000 00010000
I (79) boot:  1 otadata          OTA data         01 00 0001f000 00002000
I (86) boot:  2 phy_init         RF data          01 01 00021000 00001000
I (94) boot:  3 ota_0            OTA app          00 10 00030000 001c0000
I (101) boot:  4 ota_1            OTA app          00 11 001f0000 001c0000
I (109) boot:  5 fctry            WiFi data        01 02 003b0000 00010000
I (116) boot: End of partition table
I (121) esp_image: segment 0: paddr=00030020 vaddr=3c150020 size=4ad08h (306440) map
I (184) esp_image: segment 1: paddr=0007ad30 vaddr=3fc95000 size=0363ch ( 13884) load
I (187) esp_image: segment 2: paddr=0007e374 vaddr=40380000 size=01ca4h (  7332) load
I (191) esp_image: segment 3: paddr=00080020 vaddr=42000020 size=14b758h (1357656) map
I (441) esp_image: segment 4: paddr=001cb780 vaddr=40381ca4 size=131c4h ( 78276) load
I (463) boot: Loaded app from partition at offset 0x30000
I (464) boot: Checking flash encryption...
I (464) flash_encrypt: flash encryption is enabled (1 plaintext flashes left)
I (471) boot: Disabling RNG early entropy source...
I (488) cpu_start: Unicore app
I (488) cpu_start: Pro cpu up.
I (497) cpu_start: Pro cpu start user code
I (497) cpu_start: cpu freq: 160000000 Hz
I (497) cpu_start: Application information:
I (500) cpu_start: Project name:     ESP32_C3_Framework
I (506) cpu_start: App version:      c8953f5d-dirty
I (512) cpu_start: Compile time:     Jan  2 2024 13:50:51
I (518) cpu_start: ELF file SHA256:  3d83adff55525b8e...
Warning: checksum mismatch between flashed and built applications. Checksum of built application is 65b08502fec054a636dee8e4d2db683429d5a913387bc3ca30ec0464adecf8a4
I (524) cpu_start: ESP-IDF:          v5.1.1-dirty
I (529) cpu_start: Min chip rev:     v0.3
I (534) cpu_start: Max chip rev:     v0.99 
I (539) cpu_start: Chip rev:         v0.3
I (543) heap_init: Initializing. RAM available for dynamic allocation:
I (551) heap_init: At 3FCA9DE0 len 00032930 (202 KiB): DRAM
I (557) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (563) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (571) spi_flash: detected chip: generic
I (574) spi_flash: flash io: dout
W (578) flash_encrypt: Flash encryption mode is DEVELOPMENT (not secure)
I (586) sleep: Configure to isolate all GPIO pins in sleep state
I (593) sleep: Enable automatic switching of GPIO sleep configuration
I (600) coexist: coex firmware version: 80b0d89
I (605) coexist: coexist rom version 9387209
I (690) comnDevConf: nvs_open Code:1101
I (695) comnDevConf: nvs_open error unclearly.
I (700) comnDevConf: majorVerNum = 0, dev_hardRest = 1
E (706) comnDevConf: Blank Chip Rest
I (710) comnDevConf: lightPara.prjIndex=0

Warning: checksum mismatch between flashed and built applications. Checksum of built application is 65b08502fec054a636dee8e4d2db683429d5a913387bc3ca30ec0464adecf8a4，how can I avoid this warning？ And the error "E (706) comnDevConf: Blank Chip Rest"

AdityaHPatwardhan, 2024-01-03:
Hi @LHYYSY 
The reason for the checksum mismatch is explained [here](https://github.com/espressif/esp-idf/issues/12377#issuecomment-1754916745) in detail.
It seems a build command might have been run after the previous image was flashed leading to a mismatch.
You can just build the binary again and flash the same binary to avoid this warning.

Regarding the `E (706) comnDevConf: Blank Chip Rest`
It doesn't seem to be from any IDF code.
Do you know which component logs this error?

LHYYSY, 2024-01-04:
> Hi @LHYYSY The reason for the checksum mismatch is explained [here](https://github.com/espressif/esp-idf/issues/12377#issuecomment-1754916745) in detail. It seems a build command might have been run [...]
There is no warning after the rebuild, but it will also appear after a successful upgrade using ota，i don't know to avoid this problem

AdityaHPatwardhan, 2024-01-09:
@LHYYSY 
The error is related to execution of idf_monitor.py command.
Please execute the idf_monitor.py command from the directory where the elf file of firmware that is to be sent through OTA is kept. 
I am closing the issue,
Please feel free to re-open if required.

closed 2024-01-09 by AdityaHPatwardhan