espressif/esp-idf #12892: zlib compression not working for esp32

HareshPrajapati, 2023-12-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-dirty

### Espressif SoC revision.

esp32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32-devkit

### Power Supply used.

USB

### What is the expected behavior?

I am using zlib from [depereceated components](https://github.com/espressif/idf-extra-components).
I use a minimal example to compress 50 bytes of data using the  `compress(Bytef *dest, uLongf *destLen, const Bytef *source, uLong sourceLen) ` function.

expected output : should return compress ok **(Z_OK)**

### What is the actual behavior?

I am getting **Z_MEM_ERROR** (-4) for `esp32` and **Z_OK** (0) for `esp32s3` for the same project

### Steps to reproduce.

1. Create tamplate project using `Espressif IDF` extantion in `VS code`.
2. Configure **ESP-IDF** with version `v5.0-dirty`
3. Open **ESP-IDG Terminal**
4. Add **zlib** using command `idf.py add-dependency "espressif/zlib^1.3.0"`
5. In `main.c`, add the minimal code written below.
```
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <stdio.h>
#include <string.h>
#include "esp_log.h"
#include "zlib.h"

char hello[] = "hello hello helle !!!";
uint8_t compr[512];

void app_main(void) {
    int err;
    uLong len = (uLong)strlen(hello) + 1;
    uLong comprLen = sizeof(compr);
    err = compress(compr, &comprLen, (const Bytef*)hello, len);

    if(err != Z_OK) {
        ESP_LOGE("ESP","err %d",err);
    } else {
        ESP_LOGI("ESP","Compress OK");
    }

    while (1) {
        vTaskDelay(100);
    }
}
```
6. select device `esp32s3` and `port` for build and upload and see results on monitor terminal.
7. select device `esp32` and `port` for upload and see results on monitor terminal.

### Debug Logs.

```plain
For esp32s3 

--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM11 instead...
--- idf_monitor on \\.\COM11 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1684
load:0x403c9700,len:0xbe8
load:0x403cc700,len:0x2e98
entry 0x403c9904
I (25) boot: ESP-IDF v5.0-dirty 2nd stage bootloader
I (25) boot: compile time 10:42:53
I (25) boot: chip revision: v0.2
I (27) boot_comm: chip revision: 2, min. bootloader chip revision: 0     
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length  
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) boot_comm: chip revision: 2, min. application chip revision: 0
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0a334h ( 41780) map
I (114) esp_image: segment 1: paddr=0001a35c vaddr=3fc91000 size=02a90h ( 10896) load
I (117) esp_image: segment 2: paddr=0001cdf4 vaddr=40374000 size=03224h ( 12836) load
I (129) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1c900h (116992) map
I (153) esp_image: segment 4: paddr=0003c928 vaddr=40377224 size=09d84h ( 40324) load
I (162) esp_image: segment 5: paddr=000466b4 vaddr=50000000 size=00010h (    16) load
I (168) boot: Loaded app from partition at offset 0x10000
I (173) boot: Disabling RNG early entropy source...
I (182) cpu_start: Pro cpu up.
I (183) cpu_start: Starting app cpu, entry point is 0x4037518c
0x4037518c: call_start_cpu1 at E:/Installations/IDF/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (197) cpu_start: Pro cpu start user code
I (197) cpu_start: cpu freq: 160000000 Hz
I (197) cpu_start: Application information:
I (200) cpu_start: Project name:     compression_esp
I (206) cpu_start: App version:      1
I (210) cpu_start: Compile time:     Dec 29 2023 10:42:19
I (216) cpu_start: ELF file SHA256:  20a44639f2755c74...
I (222) cpu_start: ESP-IDF:          v5.0-dirty
I (228) heap_init: Initializing. RAM available for dynamic allocation:
I (235) heap_init: At 3FC946A0 len 00055070 (340 KiB): D/IRAM
I (241) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (248) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (254) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (261) spi_flash: detected chip: winbond
I (265) spi_flash: flash io: dio
W (269) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (283) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (304) ESP: Compress OK
```

For esp32
```
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM3 instead...
--- idf_monitor on \\.\COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2    
load:0x3fff0030,len:7000 
load:0x40078000,len:15452
ho 0 tail 12 room 4
load:0x40080400,len:3840
0x40080400: _init at ??:?

entry 0x4008064c
I (29) boot: ESP-IDF v5.0-dirty 2nd stage bootloader
I (29) boot: compile time 11:54:36
I (29) boot: chip revision: v1.0
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length  
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=09994h ( 39316) map
I (126) esp_image: segment 1: paddr=000199bc vaddr=3ffb0000 size=02638h (  9784) load
I (130) esp_image: segment 2: paddr=0001bffc vaddr=40080000 size=0401ch ( 16412) load
I (138) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=18f58h (102232) map
I (177) esp_image: segment 4: paddr=00038f80 vaddr=4008401c size=0791ch ( 31004) load
I (190) esp_image: segment 5: paddr=000408a4 vaddr=50000000 size=00010h (    16) load
I (196) boot: Loaded app from partition at offset 0x10000
I (196) boot: Disabling RNG early entropy source...
I (210) cpu_start: Pro cpu up.
I (210) cpu_start: Starting app cpu, entry point is 0x40081078
0x40081078: call_start_cpu1 at E:/Installations/IDF/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (224) cpu_start: Pro cpu start user code
I (224) cpu_start: cpu freq: 160000000 Hz
I (224) cpu_start: Application information:
I (229) cpu_start: Project name:     compression_esp
I (235) cpu_start: App version:      1
I (239) cpu_start: Compile time:     Dec 29 2023 11:53:59
I (245) cpu_start: ELF file SHA256:  09759252f1c71047...
I (251) cpu_start: ESP-IDF:          v5.0-dirty
I (257) heap_init: Initializing. RAM available for dynamic allocation:
I (264) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (270) heap_init: At 3FFB3120 len 0002CEE0 (179 KiB): DRAM
I (276) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (282) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (289) heap_init: At 4008B938 len 000146C8 (81 KiB): IRAM
I (296) spi_flash: detected chip: gd
I (299) spi_flash: flash io: dio
W (303) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (317) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (327) ESP: err -4
```
```

### More Information.

Thus, basically, zlib compression works on esp32s3 but not on esp32.

Why is it returning Z_MEM_ERROR (-4) for esp32?

I also try with different esp32 but same error is comming.

nopnop2002, 2024-01-13:
Add the following compile options to CMakeLists.txt.
```
idf_build_set_property(COMPILE_OPTIONS "-DMAX_MEM_LEVEL=7" APPEND)
```

HareshPrajapati, 2024-01-30:
Thanks !
It is working now.

closed 2024-01-30 by HareshPrajapati