espressif/esp-idf #12893: The default of buf size in "ble_gap_adv_set_fields" func too short to use

hibfh, 2023-12-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-beta1-263-ge49823f10c

### Espressif SoC revision.

ESP

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP

### Power Supply used.

USB

### What is the expected behavior?

I want to add  Manufacturer info in adv ,when i use mfg_data .it failed for short buf size

### What is the actual behavior?

,when i change the buf size ,it lead to BLE_HCI_MAX_ADV_DATA_LEN ,but when i change it my project can not work ,always reboot.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
const char *mfgstr = "xxxxxx"
fields.mfg_data = (uint8_t *)mfgstr;
fields.mfg_data_len = strlen(mfgstr);
   rc = ble_gap_adv_set_fields(&fields);
    if (rc != 0) {
        MODLOG_DFLT(ERROR, "error setting advertisement data; rc=%d\n", rc);
        return;
    }
log: rc=4
```

### More Information.

so how to add Manufacturer info in adv ?

hibfh, 2023-12-29:
E (1494) NimBLE: error setting advertisement data; rc=4
when it turn to maniufacturer set 
`   /*** 0xff - Manufacturer specific data. */
    if ((adv_fields->mfg_data != NULL) && (adv_fields->mfg_data_len >= 2)) {
        rc = ble_hs_adv_set_flat_mbuf(BLE_HS_ADV_TYPE_MFG_DATA,
                                      adv_fields->mfg_data_len,
                                      adv_fields->mfg_data,
                                      dst, &dst_len_local, max_len, om);
        if (rc != 0) {
            return rc;
        }
    }`
the left size  empty,so it went to BLE_HS_EMSGSIZE
`    if (*dst_len + 2 + data_len > max_len) {
        return BLE_HS_EMSGSIZE;
    }
`

hibfh, 2023-12-29:
sry for close the issues

rahult-github, 2023-12-29:
Hi @hibfh ,

 > I want to add Manufacturer info in adv ,when i use mfg_data .it failed for short buf size 
 
 Per spec, the maximum size of data that can be sent in legacy adv report is fixed. Existing example already sends details ( like name / flags ). You need to modify the complete adv data so that the final size fits as per spec recommendation.

hibfh, 2024-01-03:
ok i‘ll reduce some details

closed 2024-01-03 by hibfh