espressif/esp-idf #12896: Show a Popup in Browser

diplfranzhoepfinger, 2023-12-29:
### Is your feature request related to a problem?

there is a Function: 
`httpd_resp_send_err`

which can show such a Message in Browser e.g. Chrome:

diplfranzhoepfinger, 2023-12-29:
would be interesting Solution for https://github.com/espressif/esp-idf/issues/11220

hmalpani, 2024-01-03:
Hello @diplfranzhoepfinger 
I think this might help you: https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/include/esp_http_server.h#L1318

Please let me know if it works.

Thanks!

closed 2024-01-18 by hmalpani

franz-ms-muc, 2024-01-18:
i saw in the File Server you do it this way: 

[https://github.com/espressif/esp-idf/blob/b3f7e2c8a4d354df8ef8558ea7caddc07283a57b/examples/protocols/http_server/file_serving/main/upload_script.html#L49](https://github.com/espressif/esp-idf/blob/b3f7e2c8a4d354df8ef8558ea7caddc07283a57b/examples/protocols/http_server/file_serving/main/upload_script.html#L49)