espressif/esp-idf #12898: Hibernation (Suspend to non-volatile memory) support

huming2207, 2023-12-29:
### Is your feature request related to a problem?

I'm wondering if anyone has done any research about implementing hibernation support for ESP32?

The hibernation here means "suspend to non-volatile storage", which dumps the heap and some states/peripheral registers into some sort of non-volatile storage like Flash, ReRAM/FeRAM or SD card, and goes into deep sleep afterwards, or even turn the power completely off (de-assert CHIP_EN or even just cut off the power). Then the next time it boots up, the bootloader recognises the content from the external NV storage and loads the states from it.

This is useful for some big-ish low-power HMI display or GPS tracker projects. As we all know, unlike STM32L and STM32G series microcontrollers STOP mode, ESP32 deep sleep mode cannot retain the whole RAM, while the light sleep power consumption is too high. Therefore we need something alternative as a workaround to bring the sleep current to be less than 5uA.

### Describe the solution you'd like.

I guess this is doable by hacking the secondary-stage bootloader? I guess what I really need here are some low-level bootloader APIs to dump & read back the heap to external sources.

Some parts of the states like RF and networking are definitely not necessary to retain. This should be skipped.

### Describe alternatives you've considered.

Alternatively, since ESP chips support multiple firmware partitions, I guess we might be able to prepare a "hibernation shim" firmware as a hibernation sleep & wake manager. That means before the main firmware goes to sleep, it jumps to that hibernation shim firmware (without clearing the RAM), the hibernation shim firmware put everything to external NV storage, and power off. Then later it wakes up, the bootloader firstly boot to the hibernation shim firmware, it loads back the content from NV storage, and resume the main firmware.

### Additional context.

N/A

rkit1, 2024-07-03:
You would still have to reinitialize all the peripherals to the exact same state to be consistent with the ram you have saved.

Operating systems that implement hibernation do not save literally everything, and have systems in place to handle peripherals unexpectedly resetting, and have systems to separate user processes from drivers, interrupts and all other tricky stuff. 
It's not really that feasible with esp-idf. 

If you have some parts of your project which are so heavy to reinitialize, you might be better off figuring out how to save specifically those parts.

huming2207, 2024-07-10:
> If you have some parts of your project which are so heavy to reinitialize, you might be better off figuring out how to save specifically those parts.

Yes indeed... Maybe I need to find a way to dump a specific range of heap to flash first.