espressif/esp-idf #12899: ESP32-S3 wrong cause after wakeup from ULP

redarflap, 2023-12-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1 / v5.2.0

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Lilygo Lora T3S3 V1.2

### Power Supply used.

USB

### What is the expected behavior?

Get wakeup cause ESP_SLEEP_WAKEUP_ULP after calling ulp_riscv_wakeup_main_processor() from ULP.

### What is the actual behavior?

I keep getting ESP_SLEEP_WAKEUP_UNDEFINED no matter which example I try.

### Steps to reproduce.

Probably not an ESP-IDF bug, but I'm not sure what could be wrong.

1. Step: Create project from any ULP-RISCV example
2. Step: Build, Flash (UART) and Monitor (UART)
3. Observe output

### Debug Logs.

```plain
Entering in deep sleep

None
Waiting for the device to reconnect
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378d02
0x40378d02: esp_cpu_wait_for_intr at C:/Users/Joi/esp/esp-idf/components/esp_hw_support/cpu.c:145

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x12b4
load:0x403c9700,len:0x4
load:0x403c9704,len:0xaf0
load:0x403cc700,len:0x2bdc
entry 0x403c9898
Not a ULP-RISC-V wakeup, initializing it! 
Wakeup cause: 0
Entering in deep sleep
```

### More Information.

I see exactly the same behaviour on MacOS.

Code: Simplified GPIO wakeup example

```
extern const uint8_t ulp_main_bin_start[] asm("_binary_ulp_main_bin_start");
extern const uint8_t ulp_main_bin_end[] asm("_binary_ulp_main_bin_end");

static void init_ulp_program(void);

void app_main(void)
{

  esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();
  /* not a wakeup from ULP, load the firmware */
  if (cause != ESP_SLEEP_WAKEUP_ULP)
  {
    printf("Not a ULP-RISC-V wakeup, initializing it! \n");
    printf("Wakeup cause: %d\n", cause);
    init_ulp_program();
  }

  /* ULP Risc-V read and detected a change in GPIO_0, prints */
  if (cause == ESP_SLEEP_WAKEUP_ULP)
  {
    printf("ULP-RISC-V woke up the main CPU! \n");
  }

  /* Go back to sleep, only the ULP Risc-V will run */
  printf("Entering in deep sleep\n\n");

  /* Small delay to ensure the messages are printed */
  ESP_ERROR_CHECK(esp_sleep_enable_ulp_wakeup());
  vTaskDelay(1000);
  esp_deep_sleep_start();
}

static void init_ulp_program(void)
{
  esp_err_t err = ulp_riscv_load_binary(ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start));
  ESP_ERROR_CHECK(err);
  ulp_set_wakeup_period(0, 2000000);

  /* Start the program */
  err = ulp_riscv_run();
  ESP_ERROR_CHECK(err);
}
```

Ulp code:
```
int main(void)
{
  ulp_riscv_wakeup_main_processor();
  /* ulp_riscv_halt() is called automatically when main exits */

  return 0;
}
```

redarflap, 2024-01-01:
The problem appears to be related to a UART or JTAG monitoring session being open.
When disconnecting the monitoring session and printing the output to the OLED instead, everything works as expected.

Is there any workaround to this?

dobairoland, 2024-01-02:
Hi @redarflap. Yes, right, esp-idf-monitor should not reset the chip when reconnecting. Can you share the version of esp-idf-monitor you are using? You can find it in the header in the beginning when it is started.

peterdragun, 2024-01-02:
Hi @redarflap, thanks for reporting the issue. I was able to reproduce it using JTAG, but when I connect the chip to UART I don't see such behavior. I have used the code that you have provided on my macOS machine. The chip is communicating as expected, on the first connection it is reset by the monitor and then ULP resets the chip. See log below:
```
--- esp-idf-monitor 1.3.4 on /dev/cu.usbserial-1120 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x12b4
load:0x403c9700,len:0x4
load:0x403c9704,len:0xaf4
load:0x403cc700,len:0x2bcc
entry 0x403c9898
W (88) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
Not a ULP-RISC-V wakeup, initializing it! 
Wakeup cause: 0
Entering in deep sleep

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
pro cpu reset by JTAG
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x12b4
load:0x403c9700,len:0x4
load:0x403c9704,len:0xaf4
load:0x403cc700,len:0x2bcc
entry 0x403c9898
W (90) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
ULP-RISC-V woke up the main CPU! 
Entering in deep sleep
```

The issue with JTAG is that this port disappears and then when the monitor is trying to reconnect, it thinks that you replugged the chip. Seems like the same thing happened to you based on your log ('None' is the line that indicates that the port is not found). A possible workaround could be to use `--no-reset` cli option for the monitor, but that didn't help much in my case.

Please make sure that the chip does not work for both UART and JTAG again as I can't reproduce the UART issue. Regarding the JTAG issue, I will try to take a look at what we can do to fix the issue.

redarflap, 2024-01-02:
> Hi @redarflap. Yes, right, esp-idf-monitor should not reset the chip when reconnecting. Can you share the version of esp-idf-monitor you are using? You can find it in the header in the beginning when it is started.

According to the terminal output, I am using esp-idf-monitor 1.3.4

redarflap, 2024-01-02:
@peterdragun 

In the VSCode extension, i set the target to ESP32-S3 (UART). I am not sure how else i could determine wether JTAG is active as well or not, since i didn't find any related settings in the extensions documentation.

However setting "idf.monitorNoReset" to true, did indeed resolve the issue for me. 👍

peterdragun, 2024-01-02:
> In the VSCode extension, i set the target to ESP32-S3 (UART). I am not sure how else i could determine wether JTAG is active as well or not, since i didn't find any related settings in the extensions documentation.

Whether it is UART or native USB/JTAG is determined by how the chip is connected to PC, not by some setting. UART means there is some UART to USB converter - a chip on devkit or programmer board, e.g. CH340. Based on your board I would say that you are using the JTAG and connecting to chip directly (if you are not connecting to chip with some additional programmer board).  Some boards like our ESP32-S3-DevKitM-1 have both ports, see a [picture in docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitm-1.html#description-of-components) (notice the "USB-to-UART bridge" and "USB-to-UART port"). Other boards can have just one port based on design.
The setting that you mentioned is changing how the serial communication will be handled in SW. It can work if you select the wrong one, but some features (e.g. interactive console) might not work as expected.

> However, setting "idf.monitorNoReset" to true, did indeed resolve the issue for me. 

I am glad that it helped. We will try to fix the issue so this workaround won't be needed in the future.

hernae, 2024-01-02:
I am experiencing the same or a similar issue. 
The workaround with "--no-reset" seems to work somehow, as I can see some difference in the serial terminal resetting from time to time, but I don't get any print statements returned anymore.

peterdragun, 2024-01-03:
Hi @hernae, I had the same issue when I was trying the example above. The issue is that with JTAG there is no buffer in between in communication. So when the Monitor notices the port, the data is already written before the port gets opened. Again the issue is caused by the port disappearing when chip goes to deep sleep. In my case what helped is adding a short delay (1 sec) at the start of a program.

hernae, 2024-01-03:
> Hi @hernae, I had the same issue when I was trying the example above. The issue is that with JTAG there is no buffer in between in communication. So when the Monitor notices the port, the data is already written before the port gets opened. Again the issue is caused by the port disappearing when chip goes to deep sleep. In my case what helped is adding a short delay (1 sec) at the start of a program.
> 
> When using UART you should not have the same issues as the USB-to-UART bridge has a buffer and the port is not disconnected on deep sleep, so the buffer is not even needed in this case.

Ah perfect, I put the TaskDelay at the end instead of the beginning all the time. With your hint to put it at the beginning it works fine now. Thanks a lot!

closed 2024-01-08 by espressif-bot