espressif/esp-idf #12900: littlefs例程出现“Windows does not support LittleFS partition generation”

caidamao, 2023-12-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

测试idf5.2中的《littlefs》例程，出现了以下错误。编译是基于win10 vs code插件的。
报错内容：Windows does not support LittleFS partition generation。
我关注到了readme内提示了windows不支持，那是为什么不支持，有希望使用别的工具进行支撑吗？

caidamao, 2023-12-30:
我尝试直接屏蔽了CMakeLists.txt内关于阻止生成文件的代码段，是可以正常使用的。这样看来是否可以分成两个例程，一个只是单独展示读写存储且无需采用外部生成镜像，另一个展示调用littlefs_create_partition_image的例程，可能会好一些

adokitkat, 2024-11-05:
This shouldn't be a problem anymore since `esp_littlefs` moved from the `mklittlefs` tool to `littlefs-python` for image generation. We may need to update the example.

closed 2024-11-13 by espressif-bot