espressif/esp-idf #12902: Improve esp_lcd to allow drawing partial bitmaps

projectitis, 2023-12-30:
### Is your feature request related to a problem?

Using LVGL and the flush callback (`lv_disp_drv_t.flush_cb`), LVGL will let you know only the area of the framebuffer that has updated. Performance can be improved (over drawing the entire framebuffer) by drawing only the changed portion.

esp_lcd provides a function to draw a bitmap to the display: `esp_lcd_panel_draw_bitmap`

This function allows the drawing of a bitmap to the display. However, it expects all pixel data to be contiguous - i.e. you can only draw the entire bitmap to the display, not a portion of it.

There are no functions available to draw a partial bitmap.

In LVGL's flush callback, the pixel data is the framebuffer, and since only an area of this has changed, we'd want to draw a partial bitmap (a partial area of the fb) to the display.

[image]

### Describe the solution you'd like.

OPTION 1:

```c
esp_err_t esp_lcd_panel_draw_bitmap(
    esp_lcd_panel_handle_t panel,
    int x_start,
    int y_start,
    int x_end,
    int y_end,
    const void *src_data,
    int src_x_start = 0,
    int src_y_start = 0,
    int src_w = 0,
    int src_h = 0 );
```
Where:
- The added arguments are defaulted to 0 and the function behaves exactly as previous
- if `src_w` or `src_h` are `0` they default to `x_end - x_start` and `y_end - y_start`
- overflows are checked and adjusted for

OPTION 2:

```c
esp_err_t esp_lcd_panel_draw_partial_bitmap(
    esp_lcd_panel_handle_t panel,
    int x_start,
    int y_start,
    int x_end,
    int y_end,
    const void *src_data,
    int src_x_start,
    int src_y_start,
    int src_w,
    int src_h );
```
Where:
- New function
- Same behavious as above

### Describe alternatives you've considered.

I've tried:

1. Drawing the entire framebuffer every update (`lv_disp_drv_t.full_refresh = 1`). This results slow performance, especially if only a small area has changed.
2. Call `esp_lcd_panel_draw_bitmap` repeatedly for each horizontal line. Also very slow.

The other alternatives could be:

1. Write a custom implementation
2. Try to use a library other than esp_lcd

### Additional context.

[Here is an example of the problem](https://github.com/wireless-tag-com/ZX2D10GE01R-V4848/blob/master/main/screen.c#L32-L41). This is from an example provided by the hardware manufacturer of an [ESP32-S3-based round display](https://device.report/manual/10028569). Their example will only work if `lv_disp_drv_t.full_refresh = 1` (it took me a while to discover this bug).

suda-morris, 2024-01-23:
@projectitis Thanks for your interest in the LCD driver and so sorry for the late response.

> However, it expects all pixel data to be contiguous

For the LCD/I2C/I80 LCDs, yes, this function expects the draw buffer contains color data in `1-D` dimension (I think the buffer returned form LVGL is also contiguous, right? :thinking: ).

However, for the RGB LCD, because the framebuffer was maintained in the driver side, so we also provided a way to not do the "copy" in the `draw_bitmap` function. You can get the framebuffer from `esp_lcd_rgb_panel_get_frame_buffer`. When you passed in the framebuffer pointer to the `draw_bitmap` function, the driver will not do memory copy, but mount this framebuffer to the internal DMA, so it will get flushed in the next round. 

> There are no functions available to draw a partial bitmap.

Do you mean to copy a smaller window in the user provided draw buffer, in the "draw_bitmap" function? It could be, but I don't know how this is used together with the LVGL library?

projectitis, 2024-02-21:
> Do you mean to copy a smaller window in the user provided draw buffer, in the "draw_bitmap" function? It could be, but I don't know how this is used together with the LVGL library?

Please see my explanation, with images, at the top of this thread where I explain exactly this :) 

The LVGL render callback provides a `lv_area_t` that describes the area of the buffer that has changed for that render cycle.
However, esp_lcd does not provide a way to draw only this area to the LCD panel. It only provides a method to draw the full LVGL buffer to the LCD panel.

seijikun, 2024-02-27:
Another option: Do it like the [ArduinoGFX](https://github.com/moononournation/Arduino_GFX) library does it:

Have a copy of the `esp`-internal structure definition for [esp_rgb_panel_t](https://github.com/moononournation/Arduino_GFX/blob/master/src/databus/Arduino_ESP32RGBPanel.h#L27)

And then use that and some [casting shenanigans](https://github.com/moononournation/Arduino_GFX/blob/master/src/databus/Arduino_ESP32RGBPanel.cpp#L126) to get a pointer to the [internal framebuffer](https://github.com/moononournation/Arduino_GFX/blob/master/src/databus/Arduino_ESP32RGBPanel.cpp#L131). Then you just have a buffer you can [write into with memory access operations](https://github.com/moononournation/Arduino_GFX/blob/master/src/display/Arduino_RGB_Display.cpp#L88).

Obviously, that's quite the ugly hack.
However, that's exactly how I wished that the API works: Raw framebuffer access.
(Of course that only works, when there exists a full-sized framebuffer, instead of the bounce buffer variant).
Without direct framebuffer access, you always need at least two framebuffers - one for your drawing framework and one internal one - which seems like quite a waste.