espressif/esp-idf #12903: removing ethernet driver fails in idf v5.1.2

JB-DX, 2023-12-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

After migrating from v4.4 to v5.1.2-dirty this function to remove the ethernet section now fails at  mac->del(mac).
Is there a solution for it?
Or is there now an even better function taking care of all those little details? 
I didn't find any.
```
      ESP_ERROR_CHECK(esp_eth_del_netif_glue(eth_glue));    
      // driver should be uninstalled within 2 seconds 
      // taken from:  components/esp_eth/test/test_emac.c
      ESP_ERROR_CHECK(test_uninstall_driver(eth_handle, 2000));
      phy->del(phy);
      mac->del(mac);   // fails in v5.1

```

kostaond, 2024-01-05:
@JB-DX could you please provide more information of how the function fails? e.g. share logs. The function is used in [our target tests](https://github.com/espressif/esp-idf/blob/release/v5.1/components/esp_eth/test_apps/main/esp_eth_test_apps.c#L93) and we haven't observed any issues.

JB-DX, 2024-01-07:
I again looked into it and found that `mac` still had a nullptr. Totally puzzled about that, I found that I inadvertently created a local variable with the same name, so the global one never got assigned.
This happened when I copied some lines out of an example and did not realize that.
I should have better checked it before posting here. 
I'm sorry for wasting your time.
Thank you for your reply.

Case closed.

closed 2024-01-07 by JB-DX

kostaond, 2024-01-08:
@JB-DX no worries... Great to hear you were able to resolve the issue!