espressif/esp-idf #12907: Unable to blink the onboard LED T-Display-S3 (TOUCH)

vbguyny, 2024-01-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using the T-Display-S3 (TOUCH): https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/image/T-DISPLAY-S3-TOUCH.png
I have updated to the flash file that was uploaded on Nov 19, 2023.

I am trying to get the following example working with the board with the onboard LED (there are two, one green and the other red):
https://github.com/espressif/esp-idf/tree/master/examples/get-started/blink

I don't see any where which GPIO pin these LEDs are on. Could someone please indicate how I can blink the onboard LED using VSCode and Expressif?

Lagunaxx, 2024-01-01:
this module have no onboard led connected to any GPIO. you need connect led yourself to any output-enebled pin .
Read docs carefully:
[https://github.com/espressif/esp-idf/blob/5d8fb281e7b4ae3f3827c91aa6dc7ef969580b5a/examples/get-started/blink/main/blink_example_main.c#L19C1-L19C21]/* Use project configuration menu (idf.py menuconfig) to choose the GPIO to blink,(url)

vbguyny, 2024-01-02:
@Lagunaxx Ok, I'll connect an LED to a GPIO pin and update the menu config accordingly. Thanks!

closed 2024-01-02 by vbguyny