espressif/esp-idf #12909: 簡單的代碼保護

8888alex, 2024-01-01:
### Is your feature request related to a problem?

我想要保護我的代碼，但目前的flash加密方案太過於複雜。
加上沒有明確的設置指引且無法任意試錯(次數限制可能讓我的模組變成一塊磚)。
最後只能放棄不用。
舉個例子，用最強大的密碼規範(超複雜度的密碼+高頻率的更新)去保護一台便利店的POS電腦，結果就是一張貼在螢幕下方的密碼明文。

### Describe the solution you'd like.

1.MCU自生成讀保護密鑰(内部生成的密钥)。
2.以工程為單位，在PC端自動生成對稱密鑰(外部主机生成的密钥)，將對稱密鑰寫入falsh，類似目前的明文加密流程(就地加密)，以MCU密鑰加密保護對稱密鑰。
3.PC端編譯及以對稱密鑰保護所要寫入的代碼。
4.保護後的代碼可直接燒錄或用於OTA。

這樣做的最大理由是
1.我的代碼的價值還不足以使人願意花大價錢去費心破解，但我也不想被人隨便拿走，所以不需要太強大複雜的保護機制。
2.代碼不會被讀出竄改，所以也不需要寫入保護，想怎麼寫就怎麼寫。
3.最重要的是無需額外操作便可得到基本的代碼保護，更不用擔心設置錯誤道致設備損壞。

SMike90x, 2024-01-01:
может ты просто идешь  на хуй?

SMike90x, 2024-01-01:
прости...

mahavirj, 2024-01-19:
@8888alex 

It is possible to enable flash encryption by generating the key outside of the device. Please see this [guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/host-based-security-workflows.html#enable-flash-encryption-externally) which lists the entire procedure in step-by-step manner.

In addition, if you would like a playground before trying out the instructions on actual SoC, then please consider using ESP32 port of qemu (emulator). Please find instructions for flash encryption (self-generate key) workflow [here](https://github.com/espressif/esp-toolchain-docs/tree/main/qemu/esp32#using-flash-encryption). Qemu release artifacts can be found here https://github.com/espressif/qemu/releases.

Note: In current IDF master branch, qemu can also be launched through idf.py, please see instructions [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/qemu.html).

Hope this helps!

closed 2024-01-19 by mahavirj