espressif/esp-idf #12912: my SD/MMC no longer works from version 5.0

SybexX, 2024-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

framework-espidf@3.50002.230601

### Espressif SoC revision.

ESP32-S

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

AI Thinker ESP32-CAM

### Power Supply used.

USB

### What is the expected behavior?

older SDs without ATA-TRIM support do not work

### What is the actual behavior?

SD initialization is aborted

### Steps to reproduce.

Using an older SD card

### More Information.

As of version 5.0, FF_USE_TRIM was set from 0 to 1 and since then my card has not been recognized.
This is because it does not support ATA-TRIM^^

You have added a function esp_err_t sdmmc_can_trim(sdmmc_card_t* card) for this, but it has no effect.
I searched for the error overnight and found a solution for it.

After I modified the files as follows, everything works as it should again:

### fatfs/diskio/diskio_sdmmc.c
DRESULT ff_sdmmc_ioctl (BYTE pdrv, BYTE cmd, void* buff), at lines 106 to 110 changed from:

#if FF_USE_TRIM
        case CTRL_TRIM:
            return ff_sdmmc_trim (pdrv, *((DWORD*)buff), //start_sector
                    (*((DWORD*)buff + 1) - *((DWORD*)buff) + 1)); //sector_count
#endif //FF_USE_TRIM

to:

#if (FF_USE_TRIM)
        case CTRL_TRIM:
            if(FF_CAN_TRIM){
                return ff_sdmmc_trim (pdrv, *((DWORD*)buff), //start_sector
                    (*((DWORD*)buff + 1) - *((DWORD*)buff) + 1)); //sector_count
            }
            else{
                return RES_ERROR;
            }
#endif //FF_USE_TRIM

### fatfs/src/ff.c
added:

#include "sdmmc_cmd.h"

static FRESULT remove_chain(FFOBJID* obj, DWORD clst, DWORD pclst), at lines 1437 to 1454 changed from:

#if FF_FS_EXFAT || FF_USE_TRIM
		if (ecl + 1 == nxt) {	/* Is next cluster contiguous? */
			ecl = nxt;
		} else {				/* End of contiguous cluster block */
#if FF_FS_EXFAT
			if (fs->fs_type == FS_EXFAT) {
				res = change_bitmap(fs, scl, ecl - scl + 1, 0);	/* Mark the cluster block 'free' on the bitmap */
				if (res != FR_OK) return res;
			}
#endif
#if FF_USE_TRIM
			rt[0] = clst2sect(fs, scl);					/* Start of data area to be freed */
			rt[1] = clst2sect(fs, ecl) + fs->csize - 1;	/* End of data area to be freed */
			disk_ioctl(fs->pdrv, CTRL_TRIM, rt);		/* Inform storage device that the data in the block may be erased */
#endif
			scl = ecl = nxt;
		}
#endif

to:

#if FF_FS_EXFAT || FF_USE_TRIM
		if(FF_FS_EXFAT || FF_CAN_TRIM){
			if (ecl + 1 == nxt) {	/* Is next cluster contiguous? */
				ecl = nxt;
			} 
			else {				/* End of contiguous cluster block */
#if FF_FS_EXFAT
				if (fs->fs_type == FS_EXFAT) {
					res = change_bitmap(fs, scl, ecl - scl + 1, 0);	/* Mark the cluster block 'free' on the bitmap */
					if (res != FR_OK) return res;
				}
#endif
#if FF_USE_TRIM
				if(FF_CAN_TRIM){
					rt[0] = clst2sect(fs, scl);					/* Start of data area to be freed */
					rt[1] = clst2sect(fs, ecl) + fs->csize - 1;	/* End of data area to be freed */
					disk_ioctl(fs->pdrv, CTRL_TRIM, rt);		/* Inform storage device that the data in the block may be erased */
				}
#endif				
				scl = ecl = nxt;
			}
		}
#endif

FRESULT f_mkfs(const TCHAR* path, const MKFS_PARM* opt, void* work, UINT len), at lines 5946 to 5949 changed from:

#if FF_USE_TRIM
		lba[0] = b_vol; lba[1] = b_vol + sz_vol - 1;	/* Inform storage device that the volume area may be erased */
		disk_ioctl(pdrv, CTRL_TRIM, lba);
#endif

to:

#if FF_USE_TRIM
		if(FF_CAN_TRIM){
			lba[0] = b_vol; lba[1] = b_vol + sz_vol - 1;	/* Inform storage device that the volume area may be erased */
			disk_ioctl(pdrv, CTRL_TRIM, lba);
		}
#endif

FRESULT f_mkfs(const TCHAR* path, const MKFS_PARM* opt, void* work, UINT len), at lines 6175 to 6178 changed from:

#if FF_USE_TRIM
		lba[0] = b_vol; lba[1] = b_vol + sz_vol - 1;	/* Inform storage device that the volume area may be erased */
		disk_ioctl(pdrv, CTRL_TRIM, lba);
#endif

to:

#if FF_USE_TRIM
		if(FF_CAN_TRIM){
			lba[0] = b_vol; lba[1] = b_vol + sz_vol - 1;	/* Inform storage device that the volume area may be erased */
			disk_ioctl(pdrv, CTRL_TRIM, lba);
		}
#endif

### sdmmc/sdmmc_cmd.c
added:

int FF_CAN_TRIM = 0;

esp_err_t sdmmc_can_trim(sdmmc_card_t* card), at lines 630 to 636 changed from:

esp_err_t sdmmc_can_trim(sdmmc_card_t* card)
{
    if ((card->is_mmc) && (card->ext_csd.sec_feature & EXT_CSD_SEC_GB_CL_EN)) {
        return ESP_OK;
    }
    return ESP_FAIL;
}

to:

esp_err_t sdmmc_can_trim(sdmmc_card_t* card)
{
    if ((card->is_mmc) && (card->ext_csd.sec_feature & EXT_CSD_SEC_GB_CL_EN)) {
        FF_CAN_TRIM = 1;
        return ESP_OK;
    }
    FF_CAN_TRIM = 0;
    return ESP_FAIL;
}

### sdmmc/include/sdmmc_cmd.h

added:

extern int FF_CAN_TRIM;

SybexX, 2024-01-02:
Since this problem really affects all 5. versions, I have now created a "Issues" in espressif, let's see if something happens there.
https://github.com/espressif/esp-idf/issues/12912

haberturdeur, 2024-01-05:
Hi, thank you for the report.
Could you please try using the original code, with only this change (adding the can_trim check):

```c
DRESULT ff_sdmmc_ioctl (BYTE pdrv, BYTE cmd, void* buff)
{
    sdmmc_card_t* card = s_cards[pdrv];
    assert(card);
    switch(cmd) {
        case CTRL_SYNC:
            return RES_OK;
        case GET_SECTOR_COUNT:
            *((DWORD*) buff) = card->csd.capacity;
            return RES_OK;
        case GET_SECTOR_SIZE:
            *((WORD*) buff) = card->csd.sector_size;
            return RES_OK;
        case GET_BLOCK_SIZE:
            return RES_ERROR;
#if FF_USE_TRIM
        case CTRL_TRIM:
            if (!sdmmc_can_trim(card)) {
                return RES_PARERR;
            }
            return ff_sdmmc_trim (pdrv, *((DWORD*)buff), //start_sector
                    (*((DWORD*)buff + 1) - *((DWORD*)buff) + 1)); //sector_count
#endif //FF_USE_TRIM
    }
    return RES_ERROR;
}
```

SybexX, 2024-01-05:
After a few tests everything (initialization, reading, writing and deleting) works as it should, see picture

[image]

SybexX, 2024-01-05:
I've now tested it with two SDs that didn't work before and both work now.

SybexX, 2024-01-05:
I've now tested with a second ESP32-CAM and a third SD that didn't work before and now it works as expected ^^

pacucha42, 2024-01-08:
@SybexX - closing the task. Feel free to reopen if the issue appears again. Thank you

pacucha42, 2024-01-08:
Reopening, there is a code fix ready for merging so this task cannot yet be closed. My bad, sorry

closed 2024-01-08 by espressif-bot