espressif/esp-idf #12914: C++20 chrono::parse not supported?

wuyuanyi135, 2024-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-beta1-226-ga36d8bc742

### Espressif SoC revision.

ESP32S3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1-N16R2

### Power Supply used.

USB

### What is the expected behavior?

https://en.cppreference.com/w/cpp/chrono/parse should work in the toolchain supporting C++20

### What is the actual behavior?

chrono namespace does not contain `parse` 
[image]

### Steps to reproduce.

N/A
...

igrr, 2024-01-02:
This will be supported starting from GCC 14, see https://gcc.gnu.org/bugzilla/show_bug.cgi?id=104167. We will likely have a GCC 14.1-based toolchain a few months after the official upstream release. For now you can use one of the libraries which work as a drop-in replacement.

wuyuanyi135, 2024-01-02:
@igrr Thank you for your quick response!

teathinker, 2024-01-23:
> This will be supported starting from GCC 14, see https://gcc.gnu.org/bugzilla/show_bug.cgi?id=104167. We will likely have a GCC 14.1-based toolchain a few months after the official upstream release. For now you can use one of the libraries which work as a drop-in replacement.

make no sense to not add all in the same release

gerekon, 2025-05-19:
Fixed in the latest `esp-14.2.0_20241119` GCC toolchain release

closed 2025-05-19 by gerekon