espressif/esp-idf #12917: IDF 5.x.x global variables are not accessible from other tasks e.g. the HTTP server task

ralphde, 2024-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.x

### Espressif SoC revision.

ESP32-S2

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom

### Power Supply used.

External 5V

### What is the expected behavior?

Global variables are global and should be accessible from anywhere.

### What is the actual behavior?

panic when access trial from e.g. HTTP server component on a global variable

### Steps to reproduce.

set a global struct variable and init it in the main task. Try access it from the HTTP server task or any other task.

### Debug Logs.

```plain
non on hand right now. Any 5.x seem to be so bad that it does not make sense to investigate further. I tried work around's since two weeks putting global accessible data in the heap. It is just too much to do
```

### More Information.

I am burned out now after sacrificing Christmas and new years eve as my client expects a solution today. The whole problem started when I set up a new computer for developing with no ties to former installed IDF versions. On my old computer, where the .espressife folder contained other IDF tools it worked miraculously using IDF 5.0.2 until I started to use a clean new computer. And now I introduced the habit to delete the .espressife folder when I change the IDF version and this issue is persistent now

I am doing this since decades, Such a mess I have never seen before. see all my experienced issues regarding 5.x here on Github

ESP-Marius, 2024-01-03:
Sorry to hear that you are having a bad experience.

There shouldnt be any issues with accessing global variables from other tasks provided that you use the proper protection mechanisms (mutexes++)  to ensure thread-safety. 

Are you able to give a simple example where this doesn't work that I can reproduce? What is the error you are getting?

ralphde, 2024-01-03:
yes, indeed, it should not be a problem and if I say not accessible means
in a 'static' way, setup data, written once before any other task starts,
and read only after. And when I say accessible means I am not getting the
data as stored in the global variable. The values are random based, which
is the problem.  I sent my code already to Jakob in case #12711,  however
are you aware that 5.x has a serious problem with data cashing?  cash 0k
does not work, see  #12711. So having 8k data cashing set to ON does not
have the potential to cause the issue I am seeing?

Be aware that our code ran und does run perfectly under 4.0.6 and worked
even with 5.0.2 until around 20. Dezember. From that date on our code does
not work with any 5.x version of the IDF. Which I understand, as a failure
in the data cashing module would trigger an update of all 5.x versions
available to install. I have seen changes in the behavior of all 5.x
versions since the beginning of Dezember, because, since I am jumping
between all available 5.x versions in order to find a work around for our
issues.
Even spending many hours it was impossible. Please review my last 4 issue
posts and you may understand the weight of the problems. And I am burned
out spending about 14h a day In Dezember checking out versions and trying
work arounds for the problem. The access issue I could fix with using the
heap for global variables. The NVS problem seem to me is connected with the
same root cause of memory management.

Please have a look at our code I provided and let me know what I am doing
wrong, I need to provide a solution to my client asap.

Until then I am convinced that IDF 5.x is toxic. Sorry

Thank you

Best Regards

Mit freundlicher Empfehlung

Ralph Deffke

ESP-Marius, 2024-01-05:
>  And when I say accessible means I am not getting the
data as stored in the global variable. The values are random based, which
is the problem.

I'm sorry, but without any specific examples on what you are doing it is very hard for me to provide any useful feedback. Writing to a variable from `app_main` and then reading it in a task afterwards does indeed work when I test it (and if it didnt then I think we would see lots of issues in IDF), so I think the issue here is probably something else than "global variables doesnt work"

If you can give me a minimal example that shows this then I would be happy to take a look again.

ralphde, 2024-01-05:
well, obviously, however I would like to wait for having the issue #12711
repaired and in effect for testing again. Please keep in mind that our
application started with IDF 4.4 and works fine with 4.x versions. All the
problems started with 5.x.
Another problem I have is that I am not sure if my client is willing to pay
the expenses to prepare you an example.

Regarding your comment that you would see a lot of problems if my
experience is true, I ask you how many people you think are using data
cashing 0k addressed in #12711? (all of them having started the project
with 4.4) and I am the only one who reported this problem, is it? You
should consider that most of the users of this framework do not use the
ESP32-S2 and most of them may not have the self confidence to report such a
basic issue.

And I repeat, Jakob and the people handling #12711 do have the code showing
this issue. You may look at it. In the meantime I admit that I may have not
properly reported this issue due to my extended frustration as my client
did not cover my time to nail down these issues. I mean it keeps on and on
and on. See the issue in 5.1.2 with the SD card component #12786 or NVS
component not working in #12916, nobody even looked at it yet. AND nobody
else reported these issues yet as well. So please look at my code provided
and tell me that I am an idiot and don't know what I am doing or fix these
issues.

I am not sure if my client covers my expenses or decides to change
technology. And sorry if you find my way of expression outraged, I am just
massively frustrated.

Have a nice and paid day!

Best Regards

Mit freundlicher Empfehlung

Ralph Deffke