espressif/esp-idf #12919: ESP32-S3 GDMA and code mismatch?

terjeio, 2024-01-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Framework version is 4.43.
I have found similar issues but with no resolution, e.g. [this is one](https://forum.esp32.com/viewtopic.php?t=31016) that is for porting basically [the same code](https://github.com/grblHAL/ESP32/blob/master/main/i2s_out.c) from ESP32 to ESP32-S3. 

I am struggling to get the GDMA controller to generate EOF interrupts in custom driver code.
When trying to get to the bottom of this it seems that the TRM does not match definitions in the framework and may also be inconsistent within itself regarding the linked list descriptor. AFAIKT the code defines a descriptor entry like this :

```
typedef struct lldesc_s {
    volatile uint32_t size  : 12,
             length: 12,
             offset: 5, /* h/w reserved 5bit, s/w use it as offset in buffer */
             sosf  : 1, /* start of sub-frame */
             eof   : 1, /* end of frame */
             owner : 1; /* hw or sw */
    volatile uint8_t *buf;       /* point to buffer data */
    union {
        volatile uint32_t empty;
        STAILQ_ENTRY(lldesc_s) qe;  /* pointing to the next desc */
    };
} lldesc_t;
```
and the PRM like this in section 3.4.1, figure 3.3:

[image]

and then it goes on to describe the fields in figure 3.3.

In section 3.4.5 there is a reference to a EOF bit in the descriptor, which is _not_ shown in figure 3.3:

> Naively, it would seem to be possible to do this by clearing the EOF bit of the final descriptor in the existing list and setting
its next descriptor address pointer field (DW2) to the first descriptor of the to-be-added list.

Am I missing something when trying to make sense of the above?

In the code there are five decriptors set up based on the `lldesc_t` type and they are linked in a circular fashion with the `eof` field set. The DMA controller outputs the data continously as it should but does not generate the EOF interrupt nor sets the related raw interrupt or the interrupt status flag when a frame has been transmitted. The interrupt enable flag and the eof mode flag in the conf0 register are both set so for me it looks like a hardware bug. Or perhaps I am again overlooking something?

suda-morris, 2024-01-03:
[this](https://github.com/espressif/esp-idf/blob/release/v4.4/components/hal/include/hal/dma_types.h#L40) DMA descriptor definition is more consistent with the TRM.

Can you show how you set up your DMA descriptors?

terjeio, 2024-01-03:
> Can you show how you set up your DMA descriptors?

```
    for (int buf_idx = 0; buf_idx < I2S_OUT_DMABUF_COUNT; buf_idx++) {
        // Initialize DMA descriptor
        i2s_sr.dma.desc[buf_idx]->owner        = 1;
        i2s_sr.dma.desc[buf_idx]->eof          = 1;  // set to 1 will trigger the interrupt
        i2s_sr.dma.desc[buf_idx]->sosf         = 0;
        i2s_sr.dma.desc[buf_idx]->length       = I2S_OUT_DMABUF_LEN;
        i2s_sr.dma.desc[buf_idx]->size         = I2S_OUT_DMABUF_LEN;
        i2s_sr.dma.desc[buf_idx]->buf          = (uint8_t *)i2s_sr.dma.buffers[buf_idx];
        i2s_sr.dma.desc[buf_idx]->offset       = 0;
        i2s_sr.dma.desc[buf_idx]->qe.stqe_next = (lldesc_t *)((buf_idx < (I2S_OUT_DMABUF_COUNT - 1)) ? (i2s_sr.dma.desc[buf_idx + 1]) : i2s_sr.dma.desc[0]);

        i2s_clear_dma_buffer(i2s_sr.dma.desc[buf_idx], port_data);
    }
```

I have changed it (and updated related code) to:

    for (int buf_idx = 0; buf_idx < I2S_OUT_DMABUF_COUNT; buf_idx++) {

        memset(i2s_sr.dma.desc[buf_idx], 0, sizeof(dma_descriptor_t));
        // Initialize DMA descriptor
        i2s_sr.dma.desc[buf_idx]->dw0.owner = 1;
        i2s_sr.dma.desc[buf_idx]->dw0.length = I2S_OUT_DMABUF_LEN;
        i2s_sr.dma.desc[buf_idx]->dw0.size = I2S_OUT_DMABUF_LEN;
        i2s_sr.dma.desc[buf_idx]->buffer = i2s_sr.dma.buffers[buf_idx];
        i2s_sr.dma.desc[buf_idx]->next = (dma_descriptor_t *)((buf_idx < (I2S_OUT_DMABUF_COUNT - 1)) ? (i2s_sr.dma.desc[buf_idx + 1]) : i2s_sr.dma.desc[0]);

        i2s_clear_dma_buffer(i2s_sr.dma.desc[buf_idx], port_data);
    }

and the code behaves exatly the same - the data in the buffers are output and no interrupt is generated.

I do not set `suc_eof` since the list is circular, if I do the buffers are output and transmission stops, and no `GDMA_OUT_TOTAL_EOF_CHn_INT` is asserted when enabled.
I do not set `err_eof` since this is only used when receiving according to the TRM (section 3.4.1).

Section 3.4.7 states:

> The GDMA controller uses EOF (end of frame) flags to indicate the end of data frame or packet
transmission.

"flags" suggests  there should be at least two flags but I cannot find one related to end of frame (except the `GDMA_OUT_EOF_MODE_CHn` bit in the configuration register). Is the statement incorrect?

A side note: if I search for [dma_descriptor_t](https://github.com/search?q=repo%3Aespressif%2Fesp-idf%20dma_descriptor_t&type=code) and [lldesc_t](https://github.com/search?q=repo%3Aespressif%2Fesp-idf%20lldesc_t&type=code) in the framework code (on github) the I see that `lldesc_t` is still used extensively, even in code not specific to a processor variant, and `dma_descriptor_t` seemingly only in code shared by all. I find this confusing since the `dma_descriptor_t` type was introduced in v4.3, I assume to support newer processor variants, and thus not available in [previous versions](https://github.com/espressif/esp-idf/blob/release/v4.2/components/hal/include/hal/dma_types.h).

suda-morris, 2024-01-03:
I think you still needs to set the EOF flag in the descriptor for the TX path. You can refer to how the i2s driver configures the DMA descriptor from [here](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_i2s/i2s_common.c#L423-L441)

terjeio, 2024-01-03:
> I think you still needs to set the EOF flag in the descriptor for the TX path. 

This is a circular argument, first you tell me to use `dma_descriptor_t` which do not have a `eof` flag and now we are back to `lldesc_t` [which has](https://github.com/espressif/esp-idf/blob/ece73357caa6c766770136b82639964870e340ba/components/esp_driver_i2s/i2s_common.c#L428). This means the TRM is inconsistent/incorrect and do not match the processor?

suda-morris, 2024-01-04:
I2S uses this eof flag: https://github.com/espressif/esp-idf/blob/release/v4.4/components/hal/include/hal/dma_types.h#L35

terjeio, 2024-01-04:
From the TRM:

> suc_eof (DW0) [30]: Specifies whether this descriptor is the last descriptor in the list.

My list is _circular_, there is no last descriptor. And I want an interrupt _on each completed frame_ like the ESP32 can. If this is not possible please tell me so and also update the documentation to reflect that.

BTW there are two typos in the include, `reversed24_27` is misspelled and `/*!< Whether the descriptor is the last one in the link */ `makes no sense to me and does not match the TRM in section 3.4.1, according to it should be `/*!< Whether the descriptor is the last one in the list */`. And according to you neither of the two is correct?

suda-morris, 2024-01-04:
> Specifies whether this descriptor is the last descriptor in the list.

That doesn't mean "you can't set the EOF flag for a circular descriptor list". It is for sure possible. You can get EOF notification every time that descriptor being handled by the DMA.

closed 2025-01-17 by suda-morris