espressif/esp-idf #12920: Crash when calling `esp_http_client_init()`

gamename, 2024-01-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-796-g30870c819f

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-WROOM-32D

### Power Supply used.

USB

### What is the expected behavior?

Sending an HTTPS POST message

### What is the actual behavior?

intermittent crash when reading the `esp_http_client_config_t` client config

### Steps to reproduce.

1. Flash the image
2. Try to send a POST to and HTTPS address
3. About 3 out of 5 times, you'll get a crash

The code is [here](https://github.com/mailbox-notifier/esp32-mailbox-reed-switch-sensor/blob/edf23b111917d1dce51e2afa79c1d7b0ea51f5af/components/http_manager/http_manager.c#L153), so I suspect something is not correct in my config.

### Debug Logs.

```plain
Executing action: flash
Running cmake in directory /Users/tennis/src/esp32-mailbox-reed-switch-sensor/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/Users/tennis/.espressif/python_env/idf5.3_py3.11_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/tennis/src/esp32-mailbox-reed-switch-sensor"...
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /Users/tennis/src/esp32-mailbox-reed-switch-sensor/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- Found Python3: /Users/tennis/.espressif/python_env/idf5.3_py3.11_env/bin/python (found version "3.11.6") found components: Interpreter
-- App "esp32-mailbox-reed-switch-senso" version: edf23b1
-- Adding linker script /Users/tennis/src/esp32-mailbox-reed-switch-sensor/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/tennis/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /Users/tennis/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /Users/tennis/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /Users/tennis/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Users/tennis/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /Users/tennis/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /Users/tennis/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /Users/tennis/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
[... 613 lines cut ...]
I (860) wifi:state: auth -> assoc (0)
I (870) wifi:state: assoc -> run (10)
W (880) wifi:[ADDBA]rx delba, code:39, delete tid:0
W (880) wifi:[ADDBA]rx delba, code:39, delete tid:0
W (880) wifi:[ADDBA]rx delba, code:39, delete tid:2
I (900) wifi:connected with disconnected, aid = 2, channel 6, BW20, bssid = 40:ed:00:42:7e:ea
I (900) wifi:security: WPA2-PSK, phy: bgn, rssi: -56
I (910) wifi:pm start, type: 1

I (910) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (930) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (950) wifi:<ba-add>idx:0 (ifx:0, 40:ed:00:42:7e:ea), tid:0, ssn:327, winSize:64
I (1920) esp_netif_handlers: sta ip: 192.168.0.182, mask: 255.255.255.0, gw: 192.168.0.1
I (1920) WIFI_MANAGER: Successfully connected to SSID:disconnected
I (1920) WIFI_MANAGER: Connected to WiFi
I (1930) TIME_MANAGER: Initializing SNTP
I (1930) TIME_MANAGER: Waiting for system time to be set... (1/10)
I (3940) TIME_MANAGER: Waiting for system time to be set... (2/10)

Done
```

### More Information.

This is new code, so its possible I am mis-configuring something.

gamename, 2024-01-06:
@hmalpani I've done some refactoring in hopes of getting around the error.  How the crash happens consistently at [this](https://github.com/mailbox-notifier/esp32-mailbox-reed-switch-sensor/blob/ad10fd6bac2659b1d22485e92c46192377e34a1f/components/http_manager/http_manager.c#L177) point. 

I will attache a coredump.

gamename, 2024-01-06:
[esp_core_dump.txt](https://github.com/espressif/esp-idf/files/13850879/esp_core_dump.txt)

closed 2024-01-09 by gamename

yaghmr, 2024-02-22:
I have the same problem you had. How did you solve it?

gamename, 2024-02-22:
@yaghmr I wrote an http wrapper and it finally worked.  Here is the zip file for it
[http_manager-main.zip](https://github.com/espressif/esp-idf/files/14374187/http_manager-main.zip)